nodemcu/nodemcu-firmware #3176: RFC: Tidy up node.flashindex interface

TerryE, 2020-06-23:
### Background
-  The 5.1 LFS interface loads a compiled Lua index(name) function which is essentially the following, and with the `lua_lfsindex()` function a wrapper around this.
```Lua
  local name = ...
  if name == 'mod1' then return mod1;
  if name == 'mod2' then return mod2;
  if name == 'mod3' then return mod3;
  -- ... 
  return {'mod1', 'mod2', 'mod3' --[[, ... ]]}
```
-  The 5.3 includes a ROTable which is of the form `{'mod1' = mod1 , 'mod2' = mod2,  --[[, ... ]]}`with the `lua_lfsindex()` function essentially going a `lua_rawget()` from this ROTable.

The switch in the underlying implementation was largely driven by feedback from a few users who has a _lot_ of functions in LFS and were having scaling issues with the Lua access function.  This switch is done now and I am happy with the new approach.  

The original `node.flashindex()` overloading of the function lookup and configuration fetch was a bit of a botch IMO, which is why I added the LFS pseudo-table in the `_init` example.

### Proposed tweak to this interface

We I would like to do is to tweak this in a way that tidies up the implementation but without introducing a break for standard implementations:

- `node.flashindex(name)` returns the function LFS.name if present and _`nil` otherwise_ 
- `node.flashindex(nil)` returns an extended: build timestamp,  the LFS flash and mapped addresses, the LFS length, and an array of the valid module names in the LFS. 

The difference is that `node.flashindex(invalidname)` would return `nil` instead of the extended info.

The reason for this is that in 5.3, the array of the valid module names has for be done at runtime on each call by effectively doing `for k in pairs(LFSnametable) do res[#res+1] = k end` so we should only do this when the programmer really wants the list, and hence requiring a `nil` parameter is a sensible choice here.

### Suggested change

At the moment the LFS size is LFS size _used_ if the LFS is loaded and the partition size if LFS is not.  Hence there isn't a simple way to see a used / free combination.  I think that adding the extra field so that programmers can easily do this calc would be a good idea ?

Doe anyone else have any suggestions whilst we are at this?

nwf, 2020-06-23:
For the return-`nil` change, :+1:, we'll just have to chase down places where we use the 2nd return from `node.flashindex()` instead of checking whether the first is `nil` (as in the example `_init.lua`).

For the use/free change, maybe we should be returning configuration state in a table a la `node.info()`?  In fact, maybe `node.flashindex()` should *just* do the lookup aspect and `node.info("lfs")` should be added to report the LFS configuration?

TerryE, 2020-06-23:
> For the use/free change, maybe we should be returning configuration state in a table a la `node.info()`? In fact, maybe `node.flashindex()` should _just_ do the lookup aspect and `node.info("lfs")` should be added to report the LFS configuration?

My :+1: to that.  I suspect that most devs use the `LFS._config` return if at all. A `node.info()` option would be a better fit.

TerryE, 2020-06-24:
So I think that we are suggesting that:
-  `node.flashindex(name)` returns the type `function` value of the module "name" in LFS if it exists and `nil` otherwise.
-  `node.flashindex(nil)` behaviour of returns and of the sorted array module names in LFS  if it exists and `nil` otherwise.
-  We add `node.info('lfs')` which returns the keyed array containing the LFS partition size and base address, LFS partition used, the DTS.
-  We preserve the current LFS array behaviour across both versions and tweak `_init.lua` accordingly.

My implementation suggestion is that I move the new hard-coded LFS implementation from lua core to `node` to expose the pseudo array `node,LFS` and hence the `_init.lua` implementation is simply `_G.LFS = node.LFS`, but again individual users are free to vary this.

Note that:
1.  `node'info'.lfs_size` is **wrong** if the partition has been resized after build.  We should either remove this or use the partition size as loaded.
2.  My reason for leaving out the module list from the LFS info return is that generating this list is quite expensive (we sort the array), so I would rather do this only when we need to.

nwf, 2020-06-24:
> * `node.flashindex(name)` returns the type `function` value of the module "name" in LFS if it exists and `nil` otherwise.

Yes.

> * `node.flashindex(nil)` returns the sorted array module names in LFS  if it exists and `nil` otherwise.

I'm not sure whether this should be `node.flashindex(nil)` or `node.info("lfstoc")` or somesuch.  I'm just imagining surprise if someone computes the argument to `node.flashindex` and it happens to end up `nil`.

> * We add `node.info('lfs')` which returns the keyed array containing the LFS partition size and base address, LFS partition used, the DTS.

Yes, mostly; see below.

> * We preserve the current LFS array behaviour across both versions and tweak `_init.lua` accordingly.
>
> My implementation suggestion is that I move the new hard-coded LFS implementation from lua core [...]

:+1:

> Note that:
> 
> 1. `node'info'.lfs_size` is **wrong** if the partition has been resized after build.  We should either remove this or use the partition size as loaded.

Let's get rid of it in favor of having people look at the partition table itself with `node.getpartitiontable()`?

> 2. My reason for leaving out the module list from the LFS info return is that generating this list is quite expensive (we sort the array), so I would rather do this only when we need to.

:+1:

TerryE, 2020-06-24:
> I'm not sure whether this should be `node.flashindex(nil)` ...

Happy with this suggestion, though I'll keep the underyling `lua.h` API call `lua_lfsindex()` doing both jobs.   

> Let's get rid of [the `lfs_size` parameter] in favour of having

Maybe not so much this as a the startup banner reports the as-built size rather than as as-startup size and this second is more relevant, IMO, but let's leave this as an aside for now.

HHHartmann, 2020-06-24:
I completely go with your suggestions here.

I also always wondered about the naming of the methods:

``` Lua
node.flashid()
node.flashindex()
node.flashreload()
node.flashsize()
```

Shouldn't `node.flashindex()` and `node.flashreload()` rather contain LFS in their names instead of flash?

Might be a good opportunity to change the names while also changing the interface.
Leaving the flash* methods in for some (maybe extended) period of time for everyone to pick up the change.

TerryE, 2020-06-24:
OK, `node.LFSindex()` and `node.LFSreload()` with the old names as deprecated versions for now.  "Flash" in this context is a bit misleading.

TerryE, 2020-06-24:
Incidentally, there all sorts of undocumented conventions involved with the Lua source file structure.  For example:
-  All C are either classified as as core or library (set by defining `LUA_CORE` _or_ `LUA_LIB` respectfully, but not both).
-  The Lua core source files implement the core Lua functions and runtime.  The access other core functions by including the appropriate `lxxxx.h` header files.  Whilst core functions implement the public API, they rarely call these API functions internally and so _don't_  include the `lauxlib.h` header.  Note that almost API functions take a `lua_lock()` around the function to make reentrency safe, and in fact the Lua test configuration will barf if locks are nested. 
-  The Lua library source files only use the public API and therefore only include the `lua.h` and `lauxlib.h` headers, so no backdoors into Lua core files and data structures.
-  This is the reason for the `lstring.c`+`lstrlib.c` and  `ltable.c`+`ltablib.c`, etc. split. One does the core stuff and the other implements the Lua-callable library functions. 

The standard Lua code sticks to this split without exception, however some of the NodeMCU changes and enhancements don't obey these rules, though since I've got to grips with the Lua internals and the valid reasons for this approach, I have been backing out our abuses in the `app/lua` code.  The 'app/lua53` implementation is a lot cleaner in this regard, since I controlled all of the changes from standard.  

Doing this makes library support for both platforms a lot easier as the migration path is only defined for the public interface and there are no guarantees in respect of the core implementation.

In terms of the implementation of pure core NodeMCU functionality such as LFS, `app/lua/lnodemcu.c` implements the additions to the core API and `app/modules/node.c` provides the library components.  And as I discuss in #3097, this why this module doesn't belong in `modules` folder. 

Some of the tweaks in this PR will take us further along this clean-up journey.

TerryE, 2020-06-27:
Picking up the musings / discussions in #3180, the Lua library functions for LFS handling will now all be in `node.c` and therefore common to both 5.1 and 5.3.  

We will now have a `node.LFS` table that mirrors the global LFS table as previously set up my _int etc.  This implements "_name_" lookup returning the function variable for module "name" or `nil` if it isn't loaded.  It also returns the  `_time`,`_config` and `_list` properties.

`node.flashindex(name)` will be a deprecated synonym for `node.LFS.name` to avoid existing application breaks.

The `_init` module only needs to include `_G.LFS = node.LFS` (this works for both versions) to maintain existing functionality.  Note that existing applications running the old `_init` module except that  the  `_time`,`_config` and `_list` properties will all return `nil` until it is updated.

On reflection whilst I do see an advantage in keeping (albeit deprecated) the `flashindex` function to avoid breaking existing applications, I see no advantage in adding an `lfsindex` synonym as the correct method of referencing modules is as above. Ditto for the "_" properties.

At a C API layer `lua.h` will include extra `lua_lfsXXX` calls that the `node.c` code will call to interface with LFS. The API entries will be the same for both 5.1 and 5.3, though the implementations will be different because the LFS implementations are different.

@nwf, @HHHartmann, etc. what are your feelings on this?  Should we move the reload function into `node.LFS._reload()` or keep it as a direct node function, say `lfsreload`?  If is worth keeping a deprecated synonym `flashreload` to avoid breaking exiting applications that do application based reload.

nwf, 2020-06-27:
I think I agree that `flashindex` (with a deprecation warning) should be kept around a new `lfsindex` is superfluous.

I know I'm being fickle in terms of API, but... I wonder if we should reserve `node.LFS` exclusively for the contents of LFS and push all the metadata lookup somewhere else (e.g., `node.info("lfs")`?).  That is, I think `node.LFS._time` is probably not a great name, because it would clash with a module in LFS named `_time`; `node.info("lfs").time` might be better?  Maaaybe `node.flashindex()` should continue to expose `_time`, `_config`, and `_list` for backwards compatibility?

By the same token, I think `node.lfsreload()`, is the right idea, rather than `node.LFS._reload()`.

TerryE, 2020-06-27:
> I know I'm being fickle in terms of API ...

What I am trying to do here is (i) to unify LFS handling from a Lua perspective across 5.1 and 5.3, and (ii) only break existing LFS-based apps if there is a compelling reason.  If we were designing the interface afresh now, then I suspect that we might well have encapsulated it differently, but the original `flashindex()`  was rather driven by the simplicity of implementation rather than user friendliness -- which was the original driver for the `LFS` wrapper.

My understanding for various comments on issues,  most developers using LFS just execute `node.flashindex('_init')()` in their `init.lua` and then use the LFS API, plus the overlaying of `dofile`, `loadfile` and `require` rather than getting into the extra fields returned by `node.flashindex(nil)` and so use `LFS._config`, etc. -- if at all.  So I doubt that any LFS user has a module `_time` as `LFS._time` already means the timestamp (effectively a unique ID) for that image.

Because we don't want to break existing app code, and allow a simple `_G.LFS=node.LFS` then we need to preserve `node.LFS._time` as the LFS image timestamp.  Ditto for the `_config` and `_list` properties, but there is nothing to stop this also being accessible through `node.info("lfs").time` and therefore also `node.LFS._config.time`, as this is a non-breaking change.

BTW, I am not saying that we mustn't change the convention, but more that I feel that we need a wider consensus of LFS users before we should consider this option.

HHHartmann, 2020-06-28:
In the end I think we should remove the _* from node.LFS as they restrict the namespace for modules as @nwf points out. (but reading the code an overwrite is possible here, as the LFS is checked first.)

And since anyone wanting to use node.LFS directly has to touch their code anyways, they could as well change the usage of the _* parts.

For all others the _init.lua has to be adapted anyways (add the _G = node.LFS, change the require stuff), so for backwards compatibility we could introduce aconstruct similar to what we have now, just using nodeLFS instead of node.flashindex.

The only disadvantage here is, that the LFS functionality is split over several functions here which are not clearly identifiable to belong together.
``` Lua
node.LFS
node.LFSreload()
node.info("LFS")
node.getpartitiontable()
```
Hmm maybe it's not as bad as it looked at first.
We also need to define what exactly `node.info("LFS")` should return.
And I am not sure where the module list is to come from but I am sure you cofered this somewhere.

Whatever you do, please remove the fs_mapped and fs_size keys from _config, as they really don't belong there.
And please use uppercase LFS in `node.LFSreload()`

TerryE, 2020-06-28:
I am happy to use LFS (upcase) in the node method names.

After the assignment LFS=node.LFS, then `LFS`, `node.LFS` (and `ROM.node.LFS` for that matter) all point to the same ROTable.

I don't think that the LFS functionality is split as such.  It's more that LFS parameters are reported though different access paths.  

If you do have a module called `_list` for example then this function is masked by the `LFS._list` property, but in fact still accessible via `node.flashindex'_list'` and the `dofile`, `loadfile` and `require` behaviour.  We aren't changing this.  Do you want to introduce this break?

nwf, 2020-06-28:
I had proposed that the `dofile` and `loadfile` and loaders backing `require` could all just use `node.LFS[key]` syntax in discussion for #3168.  It seemed simpler to me, but I could be convinced otherwise.

TerryE, 2020-06-28:
The existing LFS naming rules are the base-name part of a source file must conform to Lua identifier syntax, that is any string of letters, digits, and underscores, not beginning with a digit and not being a reserved word.  Your recommendation here is that we reserve this full namespace available to developers for naming their modules.  This isn't really a problem in terms of my coding this so long as we exclude the [metamethod names](https://www.lua.org/manual/5.3/manual.html#2.4), and in practice it would be best to avoid any beginning with "`__`" .

I realise that we can still access `node.info"LFS"` but I would still prefer access this through the LFS table and to be able to enumerate the list of modules, and get the DTS.  If we've are excluding all valid identifiers, then how do we do this.  Of course there is nothing to stop us using invalid identifier field names such as `LFS["?list"]` and `LFS["?dts"]`.

(Note that I can't use the `__pairs` metamethod as this only works with Lua 5.3 and not Lua 5.1).

So guys, I will follow the consensus, but you are going to have to make a choice.

HHHartmann, 2020-06-28:
I don't fully understand why node.LFS is not enumerable but I don't think that an enumerable list should be accessed by node.info("LFS") especially not together with the other values.

I think that keeping the `dofile` and `loadfile` and loaders backing `require` helps writing more readable and more portable code.
Users know what these do even when not knowing anything about LFS.
Also code can be run against LFS or without. No changes needed.

TerryE, 2020-06-28:
`node.LFS` is not enumerable because it is not real table. You can't have a ROTable of {name1 = luaFunc1, ... } because Lua closures are runtime constructs and can't be stored in the `text` segment.  Yes, we could generate a real LFS table in RAM at startup, but this would have a heavy RAM footprint for larger LFS images, so this is not a good idea.  

In the same way, the built-in libraries weren't enumerable (until I put them all into the `ROM` ROTable).  

The `node.LFS` ROTable is empty but has a single metamethod, `__index`, which points to a lookup function that does this JiT.  The internal implementation in Lua 5.1 and 5.3 are different.  In the case of 5.1, a Lua index function is compiled into the LFS image, but a RTOable is used in Lua 5.3 because this scales better.

We could add a `__pairs` metamethod for 5.3 so `pairs(node.LFS)` would work as expected, _however_, 5.1 does not.  support this.

So now the list of modules is generated if and when needed (the two implementations are different, but they yield the same result).   The LuaGC will collect it when the application is done with it.

nwf, 2020-06-28:
I don't know what the right answer is; we've encouraged the use of `_`-prefixed modules at least with `_init`, but I admit that others may be so rare as to not merit thinking about.  I think I'd prefer all LFS metatada to be via `node.info("lfs")` and `node.LFS` to be the LFS itself, but I admit that this is at least a little strange given that we also like nested ROtables for our own commands elsewhere (`tls.cert.*`, `node.task.*`, ...).

nwf, 2020-06-28:
I think `__pairs`, while it might be nice to have, can absolutely wait a release or two, because it's 5.3-only and we have another mechanism for getting at those data right now.

TerryE, 2020-06-29:
THe doing is the easy bit.  It's getting good agreement on doing _what_ that takes the effort.  :smile:

HHHartmann, 2020-06-29:
Reading of `tls.cert.*` makes me think of `node.LFS.name-list`, `node.LFS.functions` and the like. Just pushing the functionality down one level and having it divided cleanly yet all in one place.
The names are not good, I know, so change them as needed if you like the idea.

TerryE, 2020-06-29:
@HHHartmann @nwf, I've been pondering this and came to the same conclusion as Gregor suggests, and that is to push the methods down one level so we reserve _one_ name. After all Lua itself reserves 22 (`and` through `while`)
I thing `get` is a good choice,  e.g. `node.LFS.get.list`,  but I can swap this with any preferred pick.

nwf, 2020-06-29:
Hm; I read @HHHartmann's proposal differently, as (preserving names)

* `node.LFS.names` is the array of modules in LFS and
* `node.LFS.functions[x]` is the module `x` in LFS.

HHHartmann, 2020-06-29:
@nwf that't exactly what I was trying to say.

TerryE, 2020-06-29:
Using `functions` to return a single function seems counter-intuitive to me and we can't use the reserved word `function`, so why not `load`?. Hence `node.LFS.load("_init")` for example. We would also have the properties `list`, `config` and `timestamp`.
My other suggestion is that for any other property of `node. LFS`, this is a synonym for `node.LFS.load()`, and that way `LFS  = node. LFS` acts as a none-breaking change for developers.

nwf, 2020-06-29:
Ah ha, an escape hatch for overloaded names.  :+1: for

* `node.LFS` as a `rotable` with...
* `node.LFS.load` a `function` yielding the module named by its sole argument from LFS or `nil` if not present,
* `node.LFS.list` (`rotable`), `node.LFS.config` (`rotable`) and `node.LFS.timestamp` (`integer`) existing with special semantics, and
* `node.LFS.__index` behaving as `node.LFS.load` unless its argument is one of the overrides just enumerated.

We should keep `node.flashindex()` for backwards compatibility, but add a deprecation warning on use.
Given the above, `node.info("lfs")` now seems superfluous, so I propose it not be added.  Similarly, no point in `node.lfsindex()`.

Questions:

* Do we still want `node.lfsflash()` or should we add `node.LFS.flash` (a `function` taking a filename) as another override?
* Should all our overrides begin with `_` so that they are obviously "somewhat special"?

TerryE, 2020-06-29:
@HHHartmann @nwf, I was wondering about this issue of namespaces in LFS.  It would be quite easy to add a facility to add this type of name separation where we can say `fred=node.LFS.namespace("FRED") then refer to `fred.sub1` in your code which would map to the module `FRED-sub1` in LFS, and `require('FRED.sub1")` would work as expected.

Is it worth the extra few dozen lines of code to support this?

`node.LFSindex` (as per Gregor's request or `node.lfsflash()` or whatever you want to call it) is really just a synonym for `node.LFS.get()`.  No extra code, so we may as well keep it.

I personally dislike the `_` prefix.  Not my suggestion.  Given that we have an alternative access function, I suggest that we drop this.  Also remember that we can't used Lua reserved words in this notation anyway so `LFS.for` generates a compile error.  (You need to write `LFS['for']`.)

nwf, 2020-06-29:
The use of namespaces sounds like it could be interesting, but probably not worth holding up integration of this RFC.  If it's easy, I don't mean to stop you, but I'm aware of how much you have on your plate and the risk of "feeping creatureism".

OK re: `node.LFSindex()` and no `_` prefix.

HHHartmann, 2020-06-30:
I personaly dislike too many entries for the same thing. So I don't need LFSindex. But we should keep both flashindex and flashreload with a deprecation warning for now. At least for the 5.1 branch (I know it's not a branch)
The 5.3 is free to define its own interface, but there should be _one_ common way to access it.
 I also don't like the _stuff. It was used before as namespaceng but is not needed anymore as we now have a deeper hierarchy anyways.

TerryE, 2020-07-03:
This code all seems to be working.  I will raise the PR tomorrow.

TerryE, 2020-07-06:
I just want to summarise what the I believe the behaviour as implemented in #3193 is:
-  `node.flashindex(n)` has a synomyn `node.LFSindex(n)` and returns a function type of module `n` if this exists in LFS _and `nil` otherwise_.
-  `node.LFS` implements in C the previous `_init.lua` table `LFS`.
    -  the config, list and time properties are before excepting the leading underscore has been dropped.
    -  the function `node.LFS.get(n)` is a  synomyn for `node.LFSindex(n)`
    -  `node.LFS.xxxx` is a synonym for `node.LFSindex('xxxx')` so long as the name `xxxx` doesn't conflict with one of the previous properties or function.
-  This behaviour is identical for both Lua 5.1 and 5.3 

The user is free to hoist this to a local or global, e.g.  `local LFS=node.LFS`, and this will allow expressions of the form `LFS.func1(args)`.

HHHartmann, 2020-07-07:
I had a slightly different layout in mind.

> I just want to summarise what the I believe the behaviour as implemented in #3193 is:
> 
> * `node.flashindex(n)` has a synomyn `node.LFSindex(n)` and returns a function type of module `n` if this exists in LFS _and `nil` otherwise_.

`node.flashindex(n)` stays as is now for backward compatibility and will be deprecated. No need for `node.LFSindex(n)`

I thought of `node.LFSreload() `as synonym of `node.flashreload()` which is to be deprecated.

> * `node.LFS` implements in C the previous `_init.lua` table `LFS`.

same same but different

>   * the config, list and time properties are before excepting the leading underscore has been dropped.
>   * the function `node.LFS.get(n)` is a  synomyn for `node.LFSindex(n)`

it should return a function type of module `n` if this exists in exists in LFS _and `nil` otherwise_ as you suggested above but would change if my suggestion would be implemented.

>   * `node.LFS.xxxx` is a synonym for `node.LFSindex('xxxx')` so long as the name `xxxx` doesn't conflict with one of the previous properties or function.

I would like to have either `node.LFS.xxxx`  or `node.LFS.get.xxxx` tendency to `node.LFS.get.xxxx` as it does not limit the namespace. But reserved words are also ok in the end as long as there is an error if user tries to add it to LFS.

> * This behaviour is identical for both Lua 5.1 and 5.3

Perfect

> 
> The user is free to hoist this to a local or global, e.g. `local LFS=node.LFS`, and this will allow expressions of the form `LFS.func1(args)`.

also good.

TerryE, 2020-07-07:
@HHHartmann Gregor, can I again suggest that you take the lead on coordinating and agreeing the specification on this.

If I am talking purely about my _personal_ preference here as input into this exercise, I would:
-  keep `node.flashindex()` and `node.flashreload()` but deprecate them.
-  Do everything else through a `node.LFS` table, so that all LFS methods and properties operate from this
-  We have the methods `reload()` and `list()`
-  We have two reserved properties `config` and `time` with the former a synonym for `net.info"lfs"`
-  The getter could be exposed as a function, e.g. `node.LFS.get('sub1') or a pseudo-array property `node.LFS.get.sub1` /  `node.LFS.get[s]`. 
-  I still prefer the convenience of `node.LFS.somefunc` but so long as we use the pseudo-array property above then it is trivial to have `local LFSfn = node.LFS.get` then `LFSfn.sub1` will work fine for me.
There are arguments both ways for function vs property, but runtime performance differences are negligible, and the implementation differences are tiny -  about 2 orders of magnitude less than the hassle of having extended debate, so you do the coordination and I will implement your spec.

HHHartmann, 2020-07-12:
@TerryE sorry for not returning to this earlier. I was finishing up everything before going on vacation, so not that much time here. But now I am on vacation and hope to have some more time free.

> * keep `node.flashindex()` and `node.flashreload()` but deprecate them.
> * Do everything else through a `node.LFS` table, so that all LFS methods and properties operate from this
> * We have the methods `reload()` and `list()`

Agreed

> * We have two reserved properties `config` and `time` with the former a synonym for `node.info"lfs"`

`node.info` currently contains only static values. and filesystem infos are also found in the file modules, so I would drop `node.info"lfs"`. It also makes things unnecessarily complicated if there are several access paths'.
It was also suggested by @nwf above to drop `node.info"lfs"`.

Taking the discussion from elsewhere (can't find it fast enough) where we found, that an LFS image should have some identifier not related to the timestamp in the header, `time` should be renamed to something more general, like ID or LFSID or whatever you come up with.

In general, why is `time` not included in `config` would safe one reserved word.

> * The getter could be exposed as a function, e.g. `node.LFS.get('sub1')` or a pseudo-array property `node.LFS.get.sub1` /  `node.LFS.get[s]`.
> * I still prefer the convenience of `node.LFS.somefunc` [...]

I think we here should also only have one way to access things. There seems to be a preference by you two to have a getter directly in node.LFS. Thinking about this it is also OK for me. As you stated Terry "Lua also has reserved words", so why not add some here. As stated above I then would like to have an error when one of these reserved words (`config` and `time`) are being added to LFS.

So drop `node.LFS.get('sub1')` or a pseudo-array property `node.LFS.get.sub1` /  `node.LFS.get[s]`.

For the LFS I would prefer the pseudo-array method, as doing node.LFS.somefunc is just better readable.

So to sum it up:

* Keep `node.flashindex()` and `node.flashreload()` but deprecate them.
 * Do everything else through a `node.LFS` table, so that all LFS methods and properties operate from this
 * We have the methods `reload()` and `list()`
 * We have two reserved properties `config` and `ID` or `LFSID` or whatever.
 	* possibly move `time` or its new name inside `config`.
 	* we will not have `net.info"lfs"`
 * No getter
 * The functions will be accessible through `node.LFS.somefunc` .
 	* An error will be thrown when one of these reserved words (`config` and `ID` or whatever) are being added to LFS. And what about `reload` and `list`? they also need to be excluded.

I hope this meets everybody's preferences and needs.

TerryE, 2020-07-12:
@HHHartmann, note that I am proposing that we add a CRC-32 checksum to all code sump files so this can also act is a reasonable id.

TerryE, 2020-07-12:
>     * An error will be thrown when one of these reserved words (`config` and `ID` or whatever) are being added to LFS. And what about `reload` and `list`? they also need to be excluded.

I would personally stick with allowing a getter and allow normal blinding rules to apply.  I'd hate to have to add a `if funcName == "config" then error (...) end` (or the C equiv) into dump logic.  Other than this, I will proceed with what is suggested. Thanks

HHHartmann, 2020-07-13:
> 
> I would personally stick with allowing a getter and allow normal blinding rules to apply. I'd hate to have to add a `if funcName == "config" then error (...) end` (or the C equiv) into dump logic. Other than this, I will proceed with what is suggested. Thanks

@TerryE that sounds fine to. I felt somewhat uncomfortable with the explicit exclusion list too.