nodemcu/nodemcu-firmware #3179: Why is there no sleep() in esp32?

f4grx, 2020-06-24:
### Missing feature
ability to use esp32 light sleep mode is missing

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html?highlight=sleep

Only deep sleep is available.

Not sleeping at all consumes a lot of energy.

### Justification
Deep sleep can only be exited with a reboot. it is basically a shutdown with only the RTC working. Wifi connections are not maintained. This is cumbersome. Suspending code execution with simple resume would be much easier to use,

### Workarounds
I do not know any workaround. maybe the freertos backend is already using light sleep when doing nothing?

Thank you for reading this.

f4grx, 2020-06-24:
I have started implementing this, including uart wakeup and associated documentation.

f4grx, 2020-06-25:
Not gonna happen for a while since it seems that the esp toolchains are not buildable on a pi :/

tomsci, 2021-01-09:
I plan to start looking at this shortly.

tomsci, 2021-01-09:
I've made a start on this here https://github.com/tomsci/nodemcu-firmware/commit/9a45d99c7839b6a0bc63d86df6a47abf57c981c6 which is on an out of date weird branch of mine. The documented functionality works (at least, timers and UART wakeup do) although I haven't tested it hugely thoroughly yet.

One thing I haven't done yet is support EXT1-format GPIO wakeup, it might be nice to support that so that you could have `sleep()` and `dsleep()` behave similarly, but I'm undecided on that. It also depends on what we want to do with `bootreason()` because my original impl of that converted touchpad id to GPIO number, so you'd get consistent numbers back whether the trigger was normal GPIO or the GPIO in touch mode, but I haven't done that here with the return value of `sleep()`.

Any comments before I rebase this properly and open a PR?

tomsci, 2021-01-10:
After some more testing of GPIO wakeup I'm less bothered about not supporting wakeup via EXT1 (or indeed EXT0), I can't see any real reason why you'd want to use it instead of GPIO wakeup for light sleep. `gpio.trig()` callbacks all occur as you'd expect so the one tangible extra feature you get from EXT1 wakeup (being able to query which GPIO(s) triggered the wake) is pretty much irrelevant.

There might be some marginal power benefit to using EXT1 - I haven't investigated whether the lightsleep code will gate more power domains if not using GPIO wakeup. But even if there is I don't imagine it'd be at all significant. Unfortunately I can't do any power testing right now, and my equipment probably wouldn't be sensitive enough to pick up any difference anyway.

marcelstoer, 2021-01-10:
Can't say anything about the code. It's been mostly you and @jmattsson who worked on the node & uart modules. However, I love the care you show for documentation 👍 🥇

jmattsson, 2021-01-11:
I haven't had the time to work on ESP stuff for a looong time now, sadly. If I recall correctly, back when I first ported NodeMCU to the ESP32, there was some issue with the light sleep breaking WiFi somehow (assuming light sleep is the "modem sleep" stuff), so I decided to not expose it at that time. Hopefully that issue has been long since fixed, but as I said, I haven't had the time to work in this space for a very long time now.

tomsci, 2021-01-11:
The underlying API `esp_light_sleep_start()` does not support preserving WiFi or BT, the sleep docs say:

> WiFi and BT connections will not be maintained in deep sleep or light sleep

So it look like that issue is still there. The docs go on to say:

> If WiFi connection needs to be maintained, enable WiFi modem sleep, and enable automatic light sleep feature (see `Power Management APIs`). This will allow the system to wake up from sleep automatically when required by WiFi driver, thereby maintaining connection to the AP.

but I'm not sure how that would interact with the NodeMCU bits or indeed works at all. For the moment I've just replicated the warning about wifi and BT not being supported in the docs for `node.sleep()`. For my own personal use, the times when I really need the sleep and dsleep, it's a project that isn't using wifi anyway, so I think there's value in having the API even with that limitation.

tomsci, 2021-01-24:
@marcelstoer can this issue be resolved now that https://github.com/nodemcu/nodemcu-firmware/pull/3370 is merged?

marcelstoer, 2021-01-24:
Thanks, forgot to close it.

PR #3370 (2021-01-17): Add node.sleep() to ESP32 API
Supporting light sleep wakeup from (non-EXT1) GPIO, UART, ULP, timer and touch

Fixes #3179

- [x] This PR is for the `dev-esp32` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well.
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

As noted in the linked issue, the esp32 branch has not until now exposed the light sleep API to Lua code, and this PR addresses  [... cut]
closed 2021-01-24 by marcelstoer