nodemcu/nodemcu-firmware #3185: Compiling the luac.cross on mingw error

wnoliveira, 2020-06-28:
In the mingw makefile, more specific at the line https://github.com/nodemcu/nodemcu-firmware/blob/master/app/lua/luac_cross/mingw32-Makefile.mak#L39 , the compiler expects for the file "lrotable.c". But this file isn't needed by the gnu make version of the makefile.

I was not able to compile at Windows ultil I remove the "lrotable.c" from this list. After that, the luac.cross compilation works fine.

TerryE, 2020-06-28:
@wnoliveira, thanks for this.  Note: self: need to update keep the MinGW and MSC makefiles in sync with the Linux on.

closed 2021-07-21 by stale[bot]

TerryE, 2021-09-26:
TBH, my 10 year old laptop that used to dual boot into Win7 has died so I am not sure how I can support this. 😒