nodemcu/nodemcu-firmware #3188: NET_PING_ENABLE comment contradicts the code

532910, 2020-06-29:
https://github.com/nodemcu/nodemcu-firmware/blob/8d091c476edf6ae2977a5f2a74bf5824d07d6183/app/include/user_config.h#L139-L141

marcelstoer, 2020-06-30:
I see no issue.  "Uncomment to enable...", yes, it's enabled by default.

closed 2020-06-30 by marcelstoer

532910, 2020-06-30:
To be able to `uncomment` it must be commented before.

532910, 2020-07-01:
It should be:
```H
// The net module optionally offers net info functionality. Uncomment the following 
// to enable the functionality. 
// #define NET_PING_ENABLE
```
or
```
// The net module optionally offers net info functionality. Comment out the following 
// to disable the functionality. 
#define NET_PING_ENABLE
```

As the current text is confusing.