nodemcu/nodemcu-firmware #3190: Add support for BMP280 sensor

hugokerstens, 2020-06-30:
### Missing feature
There is no module for reading out the [BMP280 sensor](https://www.bosch-sensortec.com/products/environmental-sensors/pressure-sensors/pressure-sensors-bmp280-1.html).

### Justification
The BMP280 is a common sensor that other people might also want use with nodemcu.

### Workarounds
The `bmp085` module reports some measurements when used with the BMP280 sensor, but the reported values are incorrect.

### Implementation
I will work create a PR for this myself. Bosch has an available API in C for the sensor at [BoschSensortec/BMP280_driver](https://github.com/BoschSensortec/BMP280_driver), which I will integrate into a module.

KT819GM, 2020-06-30:
Are you sure [BME280](https://nodemcu.readthedocs.io/en/master/modules/bme280/) does not support BMP280? Not sure, but some time ago I’ve used them with this module.

hugokerstens, 2020-06-30:
You're right, I can confirm the BME280 module also supports the BMP280. Don't know how I missed this, thanks for the quick reply!

closed 2020-06-30 by hugokerstens