nodemcu/nodemcu-firmware #3191: ftpserver mutated without adapting  documentation - breaking change

HHHartmann, 2020-07-02:
### Expected behavior
code works as documented

### Actual behavior
code fails with errormessage:

```
node.flashindex('ftpserver')().createServer("test","12345",true)
Lua error: 	...itHub\nodemcu-firmware\lua_modules\ftp\ftpserver.lua:70: attempt to index a string value
stack traceback:
	...itHub\nodemcu-firmware\lua_modules\ftp\ftpserver.lua:70: in function 'createServer'
	stdin:1: in main chunk
	[C]: ?
	[C]: ?
```

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU startup banner
You MUST include the firmware startup banner to describe the version you are using. We reserve the right to immediately close any bug that doesn't.

### Hardware
Describe which ESP8266/ESP32 device you use and document any special hardware setup
required to reproduce the problem.

TerryE, 2020-07-03:
My bad.  Yes, this is a minor breaking change that needs documenting.  The old ftp server could only be run out of LFS which is an impediment for beginners.  I've made some tweaks so the file can be split into dynamically loaded separate LC files for top level methods and this runs fine out of RAM or the #3117 fast loader, _but_ this required you to pass the ftp table in as an arg hence the need for the object notation.

I will fix the documentation.

HHHartmann, 2020-07-03:
The bad thing is that this is on Master already. 
Do we have a policy for fixing issues on Master? 

Btw how can it be split?

TerryE, 2020-07-03:
We are talking about a few line change to the MD file. If @marcelstoer is OK with this, I'll just patch the master documentation. 

The split stuff is discussed in #3117 and my gists.  Sorry about this small gremlin, but the whole Lua 5.1 to 5.3 program is complex and a lot of tidying is needed and we should realistically expect the odd bump.

Thanks for picking this up.

marcelstoer, 2020-07-03:
> If @marcelstoer is OK with this, I'll just patch the master documentation.

On master or on release? And what about RTD?

If you commit this to master you'll probably have to merge it back to dev. Otherwise we'll have conflict next time we drop to master.

TerryE, 2020-07-03:
I'll just make the same change to Dev

HHHartmann, 2020-07-03:
Should we also mention this in the Breaking Change  section of the release?

nwf, 2020-07-03:
Since we're hot-patching `master`, we should do criss-crossing merges to `dev` when we do the next release.

TerryE, 2020-07-03:
Your the expert here, not me :rofl:

marcelstoer, 2020-07-03:
> I'll just make the same change to Dev.

That's what will cause conflicts.

> we should do criss-crossing merges to `dev` when we do the next release.

👍 except that I propose this happens immediately after the master patch
- no unpleasant surprises while cutting a release
- memory of what the change is all about is still fresh
- no need to manually weave the same change into dev

nwf, 2020-07-03:
Seems OK by me.  Commit to `master`, merge the new `master` into `dev`, push new `master` and `dev` to github.  Everybody's repos should be happy with that sequence.  I think I agree with @marcelstoer that we should make this the policy for every commit on `master`/`release` that isn't just a re-tagging of `dev`.

marcelstoer, 2020-08-29:
2 months have passed and we're approaching another master drop. IMO the only thing still relevant at this point is to fix it before the _next_ master drop.

@HHHartmann just to be sure, only https://github.com/nodemcu/nodemcu-firmware/blob/dev/docs/lua-modules/ftpserver.md is affected?

HHHartmann, 2020-08-29:
Yes

marcelstoer, 2020-08-30:
Ok, I was confused since you mentioned the HTTP server

> Documentation says:
httpserver.createServer()

TerryE, 2020-08-30:
Gregor is correct. It is my bad. I changed the FTP slightly so it can be loaded into non-LFS configs. I'll do a quick tidy  and push the fix PR. Sorry about this.

HHHartmann, 2020-08-31:
> Ok, I was confused since you mentioned the HTTP server
> 
> > Documentation says:
> > httpserver.createServer()

Oh. Didn't see that one. But yes it is the ftpserver only

PR #3322 (2020-11-04): Fix ftpserver documentation and some formating in httpserver.md
Fixes #3191.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Fix documentation
I also sneaked in some formatting fixes for `httpserver.md` which I had sitting here for quite some time.

Hope that's ok.
closed 2020-11-04 by marcelstoer