nodemcu/nodemcu-firmware #3198: luac.cross -e does not work correctly

HHHartmann, 2020-07-05:
### Expected behavior
Execute Lua code containing `print` and `type` (and others)
as it does with e.g. a "3.0-master_20190907" luac.cross.

### Actual behavior
`print` and `type` are unknown
Errormessage:
`attempt to call global 'print' (a nil value)`

### Test code
Put the program in a file and call `luac.cross -e <filename>`
```Lua
print("Hello World")
```

### NodeMCU startup banner
Irrelevant as it runs on the host

### Hardware

TerryE, 2020-07-05:
Thanks. I will take a look.

TerryE, 2020-07-05:
Somehow accidentally deleted an "`_`" in `linit.c`
```Diff
@@ -61,7 +61,7 @@ extern LROT_TABLE(math);
 #if defined(LUA_CROSS_COMPILER)
 extern LROT_TABLE(base_func);
 LROT_BEGIN(rotables_meta, NULL, LROT_MASK_INDEX)
-  LROT_TABENTRY( _index, base_func)
+  LROT_TABENTRY( __index, base_func)
 LROT_END(rotables_meta, NULL, LROT_MASK_INDEX)
 
 extern LROT_TABLE(oslib);
```
You can patch your local version as a workaround.  I've already added this to my working branch and will update #3196 on my next commit.  Sorry.

closed 2020-07-05 by TerryE

HHHartmann, 2020-07-06:
@TerryE I was starting on a diff of that file but couldn't see it. I am using the old version for now, so no need to patch it for me right now. 
I am using this for the gambiarra self testing right  now. It really saves a lot of time not having to transfer it to the ESP every time.

TerryE, 2020-07-06:
One has two `_` chars in front of index and the other has one. Meta fields are prefixed by two underscores not one.

HHHartmann, 2020-07-06:
Thanks for the above edit which explains how modules are linked in. 
I found it by chance as I usually only read the mails which don't send an update on edits.

TerryE, 2020-07-06:
> I found it by chance as I usually only read the mails which don't send an update on edits.

I quite often spot typos or wording that needs tidying and do this as an edit.  On occasion I also decide on reflection to add another point which doesn't really merit a separate post, so again I do an edit.

I guess that I tend to use the email as a prompt and always go to the webpage to review the thread, though I do know that edits aren't sent out, but just the original draft of the comment.

marcelstoer, 2020-07-08:
> ...and will update #3196 on my next commit

I think you meant to reference a different PR?

TerryE, 2020-07-08:
> > ...and will update #3196 on my next commit
> 
> I think you meant to reference a different PR?

Yup #3193. Fixed original reference.