nodemcu/nodemcu-firmware #3204: 8266 not recognized on iPhone

jwothke, 2020-07-09:
Hello,

I have searched high and low but cant seem to find a solution or even mention of this problem.

I have been using a 8266 thru my windows 10 AND Samsung Note 9 phone with no issues.  As soon as I plug the 8266 in it lights right up.  I created the design on the Windows machine.  

When I plug the same 8266 into my iPhone it does noting.  The settings -> wifi     does not recognize the 8266 at all even with correct name and PW.

When I say the iPhone does not recognize the 8266 is the blue light does not light when plugging it in to iPhone etc

What I have tried:
1) tried 4 different cables from 3 diff manufactures on iPhone and nothing.
2)burned another 8266 and tested on Samsung phone and Windows with both working fine.  As soon as I plug in the 8266 on either of those 2 devices, the blue LED lits up.
3)crawled all over web for any mention of such an issue and find none....very surprised about this. Makes me believe I am doing something silly. (yes I have turned the iPhone ON  :-)

If anyone has any suggestions that would be great.  Seems like I should be able to just plug 8266 in and get at least a blue led to light a bit.

Thanks

KT819GM, 2020-07-09:
You wan’t to power ESP8266 (NodeMCU dev board?) trough USB-OTG on iPhone? If you get no light blink it’s highly possible that you have no power feed trough that cable because iPhone does not support USB-OTG.

jwothke, 2020-07-09:
Yes I want to power thru 'lightning' cable.  seems like phone should support cable as i use it to download images to my PC....but i am guessing.

KT819GM, 2020-07-09:
Any dev developing something on microcontrollers should have and be able to use a multimeter. Buy one cheap if you don’t owe any and check voltage on your board. It will save much guessing time :)

jwothke, 2020-07-10:
no voltage on any pins.....  the most i read was .12 on a few Data pins.  Nothing on 3.3 or Vin etc. 
do you  happen to know if the iPhone has any type of enable for the lightning I/O?

KT819GM, 2020-07-10:
This is not in the scope of this project,  sorry. Issue should be closed.

closed 2020-07-10 by marcelstoer