nodemcu/nodemcu-firmware #3205: RFC: Coding of internal and external document links in our documentation

TerryE, 2020-07-11:
### Internal Document References

Both Github and ReadTheDocs adopt the same convention for internal anchors.
-  Any section head generates an HTML anchor
-  The algo for these anchors is to take the text of the anchor:
    -  covert to lowercase
    -  replace any space characters by "-"
    -  ignore any non-word characters
-  Hence for example:
    -  `nodemcu-uploader.py` becomes `nodemcu-uploaderpy` and this reference can be encoded
`[nodemcu-uploader.py](#nodemcu-uploaderpy)`
    -  `What is this FAQ for?` becomes `what-is-this-faq-for` and this reference can be encoded
`[What is this FAQ for?](#what-is-this-faq-for)`
-  URI resolution assumes the current document so everything before the anchor marker (the "`#`" character) should be omitted.

### External Document References

These should just use the standard markdown syntax `[text name of URI](Fully qualified URI)`

### Relative Document References

An example here might be one document linking to a companion document in the same directory.  Yes we could simply use external document references as above, but this can have problems for example we have:
-  15 references that embed `/master/` in their URI
-  10 references that embed `/dev/` in their URI.
-  3 references that link to the fully qualified (`/dev/` version) nodemcu.readthedocs.io URI.

It can be very confusing when you are browsing the `dev` documentation, then click on a link and then notice that you are now in the `master` documentation.  And of course we have this whole `master` issue to mitigate soon.

_My recommendation is that we always use relative URIs for relative document references._

_However_, there is an issue to note when doing this and that is ReadTheDocs "friendly-ifies" URIs, converting 
-  `https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/getting-started.md` becomes
-  `https://nodemcu.readthedocs.io/en/dev/getting-started/` 

So if you want to include a reference to an anchor in a sister document say "Build Options in `build.md` do you use
-  `[check the documentation](build.md#build-options)` (the file relative format) or
-  `[check the documentation](../build/#build-options)` (the psedo-directory relative format)

_The answer here is **always use file relative** format_ because ReadTheDocs also "friendly-ifies" these URIs. 

It is also smart enough that if the relative URI is outside the docs tree but links across to the code, for example ` [telnet.lua](../../lua_modules/telnet/telnet.lua)` then it detects this and substitute a correct github NodeMCU project link rather than an invalid readthedocs.io one.

### So where do we document this?

Working this out took a little effort and is worth documenting somewhere, but where?  And @marcelstoer, you are the guru here.  Any comments?

TerryE, 2020-07-11:
Having done a `grep '\]\(' -RIPn docs  | vi -` and looked at the hits, we've got maybe a few hundred links that need cleaning up in the doc tree and also reflecting lua module moves from `lua_examples` to `lua_modules`.  This is pretty straight forward exercise in script-foo, and one that needs tacking as its own clean-up PR.

marcelstoer, 2020-07-11:
>  is worth documenting somewhere, but where? And @marcelstoer, you are the guru here. Any comments?

Sir Terry just out-gurued me but I suggest the existing wiki page https://github.com/nodemcu/nodemcu-firmware/wiki/Notes-about-writing-docs

closed 2021-07-30 by stale[bot]

TerryE, 2021-09-27:
It would be nice to have a volunteer to do this cleanup. 😃