nodemcu/nodemcu-firmware #3207: Can no longer build firmware with latest SDK

pjsg, 2020-07-13:
### Expected behavior

The firmware builds

### Actual behavior

```
[philip@neptune nodemcu-firmware]$ ./dobuild
make: Entering directory '/opt/nodemcu-firmware'
Files /tmp/buildinfo.h and app/include/buildinfo.h differ
make[1]: Entering directory '/opt/nodemcu-firmware/app'
make[2]: Entering directory '/opt/nodemcu-firmware/app/user'
DEPEND: CC app/user/user_main.c
FATAL: kernel too old
Aborted (core dumped)
CC app/user/user_main.c
FATAL: kernel too old
FATAL: kernel too old
Aborted (core dumped)
../../Makefile:422: recipe for target '.output/eagle/debug/obj/user_main.o' failed
make[2]: *** [.output/eagle/debug/obj/user_main.o] Error 134
make[2]: Leaving directory '/opt/nodemcu-firmware/app/user'
../Makefile:380: recipe for target '.subdirs' failed
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory '/opt/nodemcu-firmware/app'
Makefile:380: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/opt/nodemcu-firmware'

```

It appears that the new SDK has been built for a more recent version of linux than the version that I am running 

```
Linux neptune.gladstonefamily.net 2.6.32-754.30.2.el6.x86_64 #1 SMP Wed Jun 10 11:14:37 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

This is the current supported kernel from Centos. 

If I change the TOOLCHAIN_VERSION in the Makefile back to 20181106.0, then it builds fine.

NicolSpies, 2020-07-13:
I have done successful dev build yesterday using Docker on Windows 10.

pjsg, 2020-07-14:
Did we build the SDK or is it a version that comes directly from espressif?

marcelstoer, 2020-07-15:
> If I change the TOOLCHAIN_VERSION in the Makefile back to 20181106.0, then it builds fine.

Does this suggest it's actually an issue with https://github.com/jmattsson/esp-toolchains rather than with the SDK?

marcelstoer, 2020-09-18:
@pjsg can we close this? Your EUS enhancements indicate your build chain is actually working well 😜

pjsg, 2020-09-18:
Yeah -- you can close this. I just use the old version of the toolchain for
my development....

closed 2020-09-18 by marcelstoer