nodemcu/nodemcu-firmware #3208: node.restart() vs Hard Reset

chathurangawijetunge, 2020-07-13:
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: dev
	commit: 2fa63a1303493aa66831af5466311436d11cf827
	release: 
	release DTS: 202007071335
	SSL: false
	build type: integer
	LFS: 0x40000 bytes total capacity
	modules: file,gpio,mqtt,net,node,rtctime,sntp,tmr,uart,wifi
 build 2020-07-08 00:54 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

I am using above build....
some times not allays in long run wifi gets disconnected and does not reconnect automatically. even if i do a soft reset using  node.restart() but if i hard reset or toggle power supply it reconnects.
had this issue with 4 esp-07 devices .
not sure if it is a bug or not......

KT819GM, 2020-07-13:
I've noticed same issue, when in some cases (did not found exact ones, so did not reported yet). Wi-Fi monitor will report ```201``` when connection lost and not restored. It will not attempt to reconnect till power cycle. Also strange behavior is with ```#define WIFI_STA_HOSTNAME``` as after firmware write you have to do power cycle for changes to be reflected. I've checked only 5.3 version only. Any suggestions how to catch fault?

chathurangawijetunge, 2020-07-13:
There is definitely something wrong with the wifi sta modul.... wifi gets disconnected after some time and reports 201 and never reconnect until power cycle..
I think this is a bug

KT819GM, 2020-07-14:
@chathurangawijetunge Maybe you got any steps to reproduce this constantly? On dev version, `LUA5.3` normal reconnection on normal circumstances works okay. I can't remember now what was exact reason when module just got stuck on `201` and will not reconnect till power cycled.

chathurangawijetunge, 2020-07-14:
It happens with both master and dev lua5.3.
201 means AP not found

chathurangawijetunge, 2020-07-21:
any updates with regards to above situation....?

nwf, 2020-08-01:
Is this a regression?  If so, can you please identify a previous version that worked?

nwf, 2020-08-01:
Glancing at the logs, I don't see anything interesting happening to wifi between `3.0-master_20190907` (i.e. 310faf7fcc9130a296f7f17021d48c6d717f5fb6) and 2fa63a1303493aa66831af5466311436d11cf827, but you might have a go with `git bisect` to see what happens.

chathurangawijetunge, 2020-08-03:
Yes 
3.0-master_20190907 work's fine but the issue is with new dev

chathurangawijetunge, 2020-08-07:
Any one having this issue....?
It happens in long run >24 hours

nwf, 2020-08-08:
It may well be that nobody but you is experiencing this problem (yet); perhaps all our long-running esp8266es are still back on `master`.  Because you are able to reliably see it, please `git bisect` and tell us when the problem first appeared.  Otherwise, I'm afraid you'll just be waiting for someone else to notice, and that might not happen.

chathurangawijetunge, 2020-08-10:
> It may well be that nobody but you is experiencing this problem (yet); perhaps all our long-running esp8266es are still back on `master`. Because you are able to reliably see it, please `git bisect` [...]

i don't know how to do git bisect exactly... 
but 
3.0-master_20190907 works fine
and the issue is with
3.0-master_20200610 and in (dev)

KT819GM, 2020-08-10:
I will leave 2 devices with #995114b LUA53 (I've obsoleted 5.1 in my head already) with weak wifi connection (<-85 dBm) and will respond after few days about results. As I mentioned before I've also had same issue with few boards but somehow I could not identify problem and now it for me it works stable (on the first glance). 
Config will be:
```Lua
station_cfg = {}
wifi.sta.sethostname("WiFitest")
wifi.sta.autoconnect(1)
station_cfg.ssid = "ssid"
station_cfg.pwd = "password"
station_cfg.save = true
wifi.sta.config(station_cfg)
```
With two `eventmon`
```Lua
wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, function(T)
    print("\n\tSTA - CONNECTED" .. "\n\tSSID: " .. T.SSID .. "\n\tBSSID: " ..
              T.BSSID .. "\n\tChannel: " .. T.channel)
end)
wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T)
    print("\n\tSTA - DISCONNECTED" .. "\n\tSSID: " .. T.SSID .. "\n\tBSSID: " ..
              T.BSSID .. "\n\treason: " .. T.reason)
    connectedToMqtt = false
end)
```
Because it will use mqtt, I will add additional check:
```Lua
function MyMqtt.watch_mqtt()
tmr.create():alarm(10000, tmr.ALARM_AUTO, function()
    if not connectedToMqtt and wifi.sta.getip() ~= nil and wifi.eventmon.STA_CONNECTED == 0 then
        m:close() print('Reconnecting to Mqtt!') collectgarbage()
        tmr.create():alarm(1000, tmr.ALARM_SINGLE, function()
        MyMqtt.Connect()
            end)
        elseif not connectedToMqtt and wifi.sta.getip() == nil then
            wifi.sta.config(station_cfg)
        end
    end)
end
```
Also mqtt will be as indicator of lost and not restored connection  if it will happen.

chathurangawijetunge, 2020-08-11:
> I will leave 2 devices with #995114b LUA53 (I've obsoleted 5.1 in my head already) with weak wifi connection (<-85 dBm) and will respond after few days about results. As I mentioned before I've also [...]

At what point does MyMqtt.watch_mqtt() is called..?

nwf, 2020-08-11:
@KT819GM: `wifi.eventmon.STA_CONNECTED == 0` is a comparison between two constants; I do not think it means what you think it means?

chathurangawijetunge, 2020-08-11:
> @KT819GM: `wifi.eventmon.STA_CONNECTED == 0` is a comparison between two constants; I do not think it means what you think it means?

At what point does MyMqtt.watch_mqtt() is called..?
do we need to call wifi.sta.autoconnect(1) separately as in wifi.sta.config() defaults is true ...?
and do we have to reconfigure wifi.sta.config(station_cfg) if connection gets lost as wifi.sta.autoconnect is enabled in the beginning...?
and wifi.eventmon.STA_CONNECTED is all was 0

nwf, 2020-08-11:
@chathurangawijetunge Please either edit your earlier comments or merely refrain from making duplicate comments like that.  They are, like the duplicate issues, not conducive to conversation.

KT819GM, 2020-08-11:
> @KT819GM: `wifi.eventmon.STA_CONNECTED == 0` is a comparison between two constants; I do not think it means what you think it means?

Yeah, it bit of brain fart, got stuck experimenting with `wifi.NULLMODE` so did even more stupidness like asking question about it on gitter, and it's the reason why I don't like to provide code examples being not a programmer.

> At what point does MyMqtt.watch_mqtt() is called..?
> do we need to call `wifi.sta.autoconnect(1)` separately as in `wifi.sta.config()` defaults is true ...?
> and do we have to reconfigure [...]

```MyMqtt.watch_mqtt()``` is started after mqtt connected. I've did not posted full code for reasons I've said bit higher. 
```wifi.sta.autoconnect(1)``` I've usually declare what I use, yes, it defaults to true anyways.
```wifi.sta.config(station_cfg)``` this is what does reconnection when ```wifi.sta.autoconnect(1)``` fails, as an example:
Connect to Wifi, push ```Lua wifi.NULLMODE``` then put it back to ```Lua wifi.STATION``` and ```wifi.sta.autoconnect(1)``` will not be active, only ```wifi.sta.config(station_cfg)``` will put it back online. This is my dirty workaround for more stable wifi. 

p.s. both units online, one at constant -86 / -92 dBm

pjsg, 2020-08-11:
There is definitely something weird going on on the dev branch. Today, the node that I'm working on got into a state where it wouldn't connect to the AP. It kept on given eventmon reason 23 (a type of auth fail) and the AP also reported 
```
Aug 11 22:21:43.273 | Debugging | Station 2c3a.e835.f4eb Authentication failed
```
I tried redoing the `wifi.sta.config({ssid="correct ssid", pwd="correct pw"})` and it didn't help. I tried changing modes to `wifi.NULLMODE` and back to `wifi.STATION` -- no help. I tried power cycling the board. No help.

I tried disabling the AP that it was trying to connect to so that it would switch to a different AP. No help.

I switched to `wifi.sta.config({ssid="correct ssid", pwd="incorrect pw"})` and then back to correct. No help.

I switched to `wifi.sta.config({ssid="incorrect ssid", pwd="correct pw"})` and then back to correct. *This worked*.

The eventmon data showed the correct ssid.

This started after I loaded a new LFS image. I have no idea whether this is related -- I include it for completeness.

chathurangawijetunge, 2020-08-15:
Not just with dev this happens in 3.0-master_20200610
too.....

pjsg, 2020-08-23:
I'm adding some code so that I can read out the last 12k of flash (where the wifi setup informaion is stored) and see when/if it changes unexpectedly.

chathurangawijetunge, 2020-08-29:
> I'm adding some code so that I can read out the last 12k of flash (where the wifi setup informaion is stored) and see when/if it changes unexpectedly.

Any luck in finding the bug....,?

pjsg, 2020-08-29:
I managed to reproduce it today. It turns out that (I think) SPIFFS writes into the flash area at the end of the flash chip and overwrites the wifi settings. This is *ugly*. 

Normally the last 12k doesn't change -- even on a reboot. however, sometimes it does -- it *could* be to do with reloading the LFS region -- that was when it happened. However, the LFS partition also got corrupted at that time, so I don't know whether I can really blame it. It was SPIFFS data that was found in the last 12k. I suppose that I ought to check that the spiffs partition doesn't overlap the end of the flash.....

TerryE, 2020-08-30:
@pjsg,  This shouldn't make any difference, because the SDK is supposed to use the PT now.  See my comments in #3260.  If there some  bit of the code in our current SDK that are still writing to the old locations then we have wider issues that we need to scope and understand.  We are currently running an old 3.0 SDK version.  My first instinct would be to rebaseline to a current version and see if that fixes the problem before abandoning use of the Partition Table.

TerryE, 2020-08-30:
We currently use SDK 3.0.1. 

From what you say saving the default  `wifi.sta.config` ssid + password does not use the PT partition but writes to this old SDK 2.x area. We can just erase a chip and set the SPIFFS at 0x100000 then the 5 page reqion should be left at `FF`.  I will try to see if saving SSID credentials corrupts this.  I'll also rebaseline our SDK to 3.0.4 and see if this repeats the issue.

pjsg, 2020-08-30:
This is my partition table:
```
[{"size":4096,"address":45056,"type":4},{"size":4096,"address":49152,"type":5},{"size":12288,"address":53248,"type":6},{"size":45056,"address":0,"type":101},{"size":507904,"address":65536,"type":102},{"size":131072,"address":573440,"type":103},{"size":3469312,"address":704512,"type":106}]
> 
```

I just iterated through all the partition types and these were the values that were returned. This looks plausible, but nevertheless, when you do wifi.sta.config, it does overwrite the last 12k of flash (I have a 4MB flash chip).

TerryE, 2020-08-30:
This is as below and this looks pretty typical.  It is worth moving SPIFFS to 1M for 1M, say, so you can see exactly what is writing to the forbidden region.  Let me have a play. 

type|address|size
----|---|----
SYSTEM_PARTITION_RF_CAL|0xB000|0x1000
SYSTEM_PARTITION_PHY_DATA|0xC000|0x1000
SYSTEM_PARTITION_SYSTEM_PARAMETER|0xD000|0x3000
NODEMCU_PARTITION_EAGLEROM|0x0000|0xB000
NODEMCU_PARTITION_IROM0TEXT|0x10000|0x7C000
NODEMCU_PARTITION_LFS0|0x8C000|0x20000
NODEMCU_PARTITION_SPIFFS|0xAC000|0x34F000

TerryE, 2020-08-30:
I've just tried various combinations of `setmode()`, `sta.config()` getconfig, connect and autoconnect including valid and invalid SSID / password combos.  All works as expected and it only seems to be the valid system partitions that are getting updated.  Every dump of 0XFB000 is 20Kb × 0xFF, so if the SDK is writing to this, it's going to be on some weird error path.

pjsg, 2020-08-30:
I just did a test on my system, and these are all the offsets where my SSID is found. However, I suspect that a lot of these are left over from previous attempts. When I program a new SSID, it shows up in *one* place in the flash at f4c8e (well, it did once). On another attempt (with a different ssid) it was at f528e. [Actually those addresses are within the ssid field]. However, these addresses are within the SPIFFS partition and not within the system parameter area.

Address    cbfb5
Address    cc2ee
Address    cc457
Address    ce30a
Address    ce543
Address    ce5a7
Address    e08b5
Address    e0aee
Address    e0c57
Address    e2b0a
Address    e2d43
Address    e2da7
Address    eb3ee
Address    eb557
Address    ed40a
Address    ed643
Address    ed6a7
Address    f377f
Address    f397f
Address    f3b8e
Address   3fd010
Address   3fd144
Address   3fd1a8
Address   3fe010
Address   3fe144
Address   3fe1a8

pjsg, 2020-08-30:
I added 
```
static int node_peek(lua_State *L) {
  luaL_Buffer b;
  luaL_buffinit( L, &b );
  char *p = luaL_prepbuffer(&b);

  uint32_t addr = luaL_checkint(L, 1);
  uint32_t len = luaL_checkint(L, 2);

  if (len > LUAL_BUFFERSIZE) {
    len = LUAL_BUFFERSIZE;
  }

  int size = platform_s_flash_read(p, addr, len);

  luaL_addsize(&b, size);
  luaL_pushresult( &b );
  return 1;
}
```

to node.c and added 
```
  LROT_FUNCENTRY( peek, node_peek )
```

And then ran:

```
local last = 0
for addr = 0, 4 * 1024 * 1024 - 256, 128 do
  local data = node.peek(addr, 256)
  local _start, _end = string.find(data, "test1234")
  if _start then
    if last ~= addr + _start - 1 then
        last = addr + _start - 1
        print (string.format("Address %8x", last))
    end
  end
  tmr.wdclr()
end   
```

to search for the `test1234` ssid

TerryE, 2020-08-30:
Bugger.  My D1 Pros have 16Mb and the pages at 0xFD000 and 0FE000 and 0xFF000 are getting written to as well as the PT partition copies.  Clearly some code uses the PT, some ignores it and uses the fixed locations. I have tried upping the SDK version to the latest V3.0.4 and it makes no difference to this issue.  (Here is the Makefile patch to do this:
```diff
@@ -20,12 +20,10 @@ else ifeq ("$(RELEASE)","master")
   SDK_ZIP_ROOT   := ESP8266_NONOS_SDK-$(SDK_VER)
   SDK_FILE_VER   := $(SDK_VER)
 else
-# SDK_VER        := 3.0
-# SDK_FILE_VER   := v$(SDK_VER)
-  SDK_FILE_VER   := e4434aa730e78c63040ace360493aef420ec267c
-  SDK_VER        := 3.0-e4434aa
-  SDK_FILE_SHA1  := ac6528a6a206d3d4c220e4035ced423eb314cfbf
-  SDK_ZIP_ROOT   := ESP8266_NONOS_SDK-$(SDK_FILE_VER)
+  SDK_VER        := 3.0.4
+  SDK_FILE_VER   := v$(SDK_VER)
+  SDK_FILE_SHA1  := 24c702348ee5b9ae3ae640c1aead60c97768c881
+  SDK_ZIP_ROOT   := ESP8266_NONOS_SDK-$(SDK_VER)
 endif
 SDK_REL_DIR      := sdk/esp_iot_sdk_v$(SDK_VER)
 SDK_DIR          := $(TOP_DIR)/$(SDK_REL_DIR)
```
If you unpack the SDK from the cache to get the `examples/IoT_demo/user/user_main.c` which is what the documentation refers you to:  the RFcall, PhysData and SP partitions are assumed to be the _last_ 1+1+3 pages of physical flash memory.  We need to update our user_main code to reflect this.  I will do this tomorrow. Time for bed for me.

TerryE, 2020-08-31:
PS.  Double checking, the RFcall, PhysData partitions are based on the locations in the PT.  It is only ignored for the SP.

TerryE, 2020-08-31:
PS. Philip, you really need to do a flash erase before reflashing to make sure that you've got rid of legacy stuff and use nodemcu-partition to set the spiffs out of the way. e.g. 1Mb for ½Mb

chathurangawijetunge, 2020-09-05:
Even with
 
NodeMCU 3.0.0.0 
	branch: dev
	commit: a0d27059bcb8663bb9c8a9367142dbdc8aa7ad9d
	release: 3.0-master_20200610 +25
	release DTS: 202009031150
	SSL: false
	build type: integer
	LFS: 0x40000 bytes total capacity
	modules: file,gpio,mqtt,net,node,rtctime,sntp,tmr,uart,wifi
 build 2020-09-04 17:10 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

I am getting the same error..... wifi will not auto reconnect.
I am using esp-12 with 4M and since im on windows i use PyFlasher with the option 'yes,wipe all data' enable to flash the freeware. but no luck.
am i using the correct way to 'flash erase'....?
and if the issue is still there why is this issue is closed?????

KT819GM, 2020-09-05:
I'm not sure now, but it's possible that I got that behavior when I've compiled version without limiting spiffs (I always do `#define SPIFFS_MAX_FILESYSTEM_SIZE` and #define SPIFFS_SIZE_1M_BOUNDARY` on ESP-07S 4M devices). So maybe because of that I do not get this problem anymore (I have approx 15 units now living their life on MQTT).

p.s. one of devices I left 26 days ago is still alive and has active connection. Sadly second one had to go, to do some other work 😄

chathurangawijetunge, 2020-09-06:
> I'm not sure now, but it's possible that I got that behavior when I've compiled version without limiting spiffs (I always do `#define SPIFFS_MAX_FILESYSTEM_SIZE` and `#define [...]

OK I will do the same let u guys know....

chathurangawijetunge, 2020-09-07:
Finally i have manage to run over 24 hours  after nearly about 3 month without connection issues by enabling define SPIFFS_SIZE_1M_BOUNDARY.
i really don't know what is really happening by enabling SPIFFS_SIZE_1M_BOUNDARY 
so i thought of sharing my findings with u guys....

KT819GM, 2020-09-07:
So seems it has something to do with Philip observations. Terry will have to look at this again (I hope).

TerryE, 2020-09-07:
> So seems it has something to do with Philip observations. Terry will have to look at this again (I hope).

We have 9 maintainers on the project and at least 4 are active and C developers. I am the one who focuses on the Lua core, and that is where we are doing most of the development ATM.  I usually leave networking to the others.  If as @chathurangawijetunge says, lifting SPIFFS above the 1M boundary removes this failure mode then this could imply that some code is accessing the high 5 pages in the first 1Mb.  Something for Nathaniel or Philip to check?

chathurangawijetunge, 2020-09-09:
Same Error Pop after about 48 hours..... 
now I'm Trying with both 
#define SPIFFS_SIZE_1M_BOUNDARY
&
#define SPIFFS_MAX_FILESYSTEM_SIZE 0x20000

will update the findings......

chathurangawijetunge, 2020-09-09:
Nop even with 

#define SPIFFS_SIZE_1M_BOUNDARY
&
#define SPIFFS_MAX_FILESYSTEM_SIZE 0x20000

Error is still there after about 18 hours both devices got disconnected and no reconnection.....

TerryE, 2020-09-09:
@chathurangawijetunge if your SPIFFS is 1Mb for 2Mb then nothing is writing to the system parameter area, AFAIK other than the SDK.  Have you considered that this isn't a NodeMCU bug at all but something to do with your network config / router?   If this was a NodeMCU issue then I'd expect more users reporting this or being able to reproduce it.  For example you have a short default DHCP lease and this is expiring so your router if ignoring your ESPs?

chathurangawijetunge, 2020-09-10:
My  router DHCP lease time is 259200 sec.
the thing is when i flash 3.0-master_20190907 with the same esp-8266 it works fine.
it happens with 3.0-master_20200610 and dev only and I am using node.setonerror() function.. code is below 

node.setonerror(function(e)         
      print("'Panic Error: '",e)  
      file.open("log", "a+")
      file.write(e)
      file.close()
      node.restart()
end)

will remove node.setonerror() and try..

KT819GM, 2020-09-10:
Ok, so now seems I see exact scenario for this error. As @chathurangawijetunge is using `SPIFFS` for error logging it's definitely has to do something with it. I've just recreated same scenario with ~122 Kb file - I was uploading `*.gz` file to spiffs with `esplorer`. Write failed  failed at some point (~40 %) and at the same moment
 ```
	STA - DISCONNECTED
	SSID: ssid
	BSSID: 00:00:00:00:00:00
	reason: 201
```
was fired. Now, in that state ESP does not reconnect event after restart. Manual `wifi.sta.config(station_cfg)` does not help either. **After power cycle module reconnected.**
I've observed such behavior only with spiffs. All modules which are working OK does not use it, everything is in LFS. Who would have time to help on gitter / whatever with GDB, I'm thinking that if I physically short circuit HW pins on flash I could imitate flash write failure and catch exact error. Or maybe @pjsg could add his 2 cents how to catch that error.

chathurangawijetunge, 2020-09-10:
> Ok, so now seems I see exact scenario for this error. As @chathurangawijetunge is using `SPIFFS` for error logging it's definitely has to do something with it. I've just recreated same scenario with [...]

Correct..... all my code is in LFS.

KT819GM, 2020-09-10:
I've played with it for an ~hour. On heavily used MCU (TLS MQTT running) I could easily force it to disconnect at the time of file writing but in most cases it returns to connected state. Still, 2 times got reproduced fatal disconnection where only power cycle helped. Now I think where I did mistake - WiFi credentials were stored in LFS. Will try again tomorrow with `enduser setup`.

pjsg, 2020-09-10:
Are you doing the wifi.sta.config() on every startup? Or do you rely on the
fact that they are saved in the flash between boots?

KT819GM, 2020-09-10:
On this unit `wifi.sta.config()` was running on every reboot and it had `ssid` and `pass` saved on `LFS`. That's why possibly after power cycle it reconnected. But - it did not reconnected on software restart, as `wifi.sta.config()` was launched also.

TerryE, 2020-09-10:
We are currently running on 3.0.1 and the current is 3.0.4. We'll rebaseline the SDK immediately after the next master drop. This might help.

chathurangawijetunge, 2020-09-11:
I have following code ruing on 3 modules

local Status_led = 4
for i=0, 8, 1 do gpio.mode(i, gpio.OUTPUT) gpio.write(i,0) end
wifi.setmode(wifi.STATION)
wifi.sta.clearconfig()
wifi.sta.config({ssid="SSID" ,pwd="PWD"})
--~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
local Status_led_map={0,2,3,4,0,1} -- 1=Connected 2=Connecting 3=password 4=AP not found
tmr.create():alarm(5000, 1, function()
      tmr.softwd(30)
      local WiFi_Status=Status_led_map[wifi.sta.status()+1]
      if WiFi_Status==0 then
         gpio.write(Status_led,0) 
      else   
         gpio.serout(Status_led,gpio.LOW,{180000,180000},WiFi_Status,nil)
      end        
      print(wifi.sta.status())  
end)
--~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

after overnight all 3 went offline error '201 AP not found' this is in my init.lua not in LFS (now)

KT819GM, 2020-09-11:
I've bumped SDK to `3.0.4` also and left one device alive with ```wifi.sta.config()``` inactive. Attached it to ```broker.hivemq.com``` on channel: ```9741000/#```. It send data every 20 sec, with content: ```{"Serial": "9741000", "Boot reason": "1", "Heap": "32776", "rssi": "-84", "Live": 269, "powered by Lua 5.3.5 on SDK 3.0.4(9532ceb)"}``` where Boot reason indicates `1` - power cycle, `2` - software restart, `Live` is `tmr.time()` counter in seconds.

chathurangawijetunge, 2020-09-11:
> I've bumped SDK to `3.0.4` also and left one device alive with `wifi.sta.config()` inactive. Attached it to `broker.hivemq.com` on channel: `9741000/#`. It send data every 20 sec, with content: [...]

It seems that device have WTD restarted...
2020-09-11 18:38:08Topic: 9741000/dataQos: 0{"Serial": "9741000", "Boot reason": "4", "Heap": "32512", "rssi": "-81", "Live": 6968, "powered by Lua 5.3.5 on SDK 3.0.4(9532ceb)"}

KT819GM, 2020-09-11:
> It seems that device have WTD restarted...
Yeah, because "somebody" have done mqtt publish without checking if mqtt is available at all 😄. I've left it on battery and sadly can't fix that now. Still, if `wifi` will fail it should not reconnect even after restart.

chathurangawijetunge, 2020-09-12:
> > It seems that device have WTD restarted...
> 
> Yeah, because "somebody" have done mqtt publish without checking if mqtt is available at all 😄. I've left it on battery and sadly can't fix that now. Still, if `wifi` will fail it should not reconnect even after restart.

True... but to my experience this error happens only after about >6-12hors so if the device reboots in between it might not pop

chathurangawijetunge, 2020-09-12:
not sure if this is related to this issue, but i have notice that by `wifi.sta.clearconfig()` wan't clear the MAC address of previously connected router.
and once i get the error error '201 AP not found' 
```
function listap(t)
    for k,v in pairs(t) do
        print(k.." : "..v)
    end
end
wifi.sta.getap(listap)
```
will do nothing...

chathurangawijetunge, 2020-09-12:
> > It seems that device have WTD restarted...
> 
> Yeah, because "somebody" have done mqtt publish without checking if mqtt is available at all 😄. I've left it on battery and sadly can't fix that now. Still, if `wifi` will fail it should not reconnect even after restart.

Your device is re starting.... not running continuously

chathurangawijetunge, 2020-09-12:
Even with `LUA5.3` wifi issue is still there. I guess I have to stick with 3.0-master_20190907 for my projects 🙁

TerryE, 2020-09-14:
@chathurangawijetunge, are you saying that 3.0-master_20190907 doesn't manifest this issue but 3.0-master_20200610 does?  If so this piece of data will help to work out any underlying failure.

KT819GM, 2020-09-14:
> Your device is re starting.... not running continuously

To be honest I see watchdog restart for the first time, and I think it came from the code part I added from your example ```tmr.softwd(30)``` Still, let's keep thinking test was unsuccessful and go further.

@TerryE Please consider checking WiFi when ```UART``` writing is in process - I can get disconnections reliably when simply transferring files to ```SPIFFS```. Reconnecting most times but sometimes it's not.

chathurangawijetunge, 2020-09-15:
> @chathurangawijetunge, are you saying that 3.0-master_20190907 doesn't manifest this issue but 3.0-master_20200610 does? If so this piece of data will help to work out any underlying failure.

Yes @TerryE I have devices running on 3.0-master_20190907 over 6 months with out any issue.
But 3.0-master_20200610 gets wifi disconnect. Sometimes in about 2 hours and in some 24 hours 
Yesterday I even bought a new ESP-12 and tested it got discounted after abot 14 hours with error 201 AP not gound...
**(3.0-master_20190907 working perfectly)**

pjsg, 2020-09-16:
I'm going to try the example above and see if it does anything strange. I'm using a regular nodemcu board with nothing attached. 

However, I've been running a node off the dev branch for a while and after fixing the issue with spiffs overwriting the config, it has been rock solid.

[15 more replies]

PR #3260 (2020-09-01): Don't allocate flash to SPIFFS that is reserved by the system.
Fixes #3208

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This PR just subtracts the 20k reserved memory from the end of the flash before setting up the SPIFFS partition. This also explains why the contents of SPIFFS would get corrupted on the dev branch.

The downside of this change is that [... cut]
closed 2020-10-05 by nwf