nodemcu/nodemcu-firmware #3226: node.flashreload should not return if it worked.

pjsg, 2020-07-19:
### Expected behavior

node.flashreload("valid.img") 

should not return.

### Actual behavior

If the image is valid, then it returns `true` and then the platform restarts very shortly thereafter.

### Test code

```
print(node.flashreload("foo.img"))
```

and observe that it prints true. This is with the lua53 build.

nwf, 2020-07-20:
We could say, going forward, that `node.LFS.reload()` returns whether or not a LFS reload is required, in analogy with `adc.force_init_mode()`.

TerryE, 2020-07-20:
If it's returning, it's a bug. It's not safe to return.

pjsg, 2020-07-20:
That was my thought. The reload code ends up calling `system_restart`. The Espressif docs say
[image]

TerryE, 2020-07-20:
Yup.  My bad.  The problem is that the running code links into the old LFS and LFS references but this is long gone.

HHHartmann, 2020-07-20:
@TerryE Might this also be related to your new panic handler?

TerryE, 2020-07-20:
No. I used to force time out using the infinite loop method, but then replaced this by a jump to the h/w reset entry point. Then had a mindfart and just called reset.  You can't just throw an error the app might have done a protected call.  No, you need to leave Lua entirely.  Easy to fix.

mk-pmb, 2020-09-01:
Mine has even stranger behavior, it [fails pcall and then reboots](https://gist.github.com/mk-pmb/dc4c3c10c716d8973e3cdbe0a963934b). I'm using [this firmware](https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/actions/runs/233185495) built with `make LUA=53 clean all` from [this branch](https://github.com/mk-pmb/nodemcu-firmware/tree/lua53-dev), tested on a LoLin v3 ESP8266.

TerryE, 2020-09-01:
> Mine has even stranger behaviour ...

Not strange.  The SDK function `system_restart()` doesn't restart the CPU immediately.  It schedules a high priority close-down task which does a clean shutdown and CPU restart.  However this is an SDK task and it only be scheduled _after_ the current running task has been completed.  Hence the code schedules a system restart and throws a "system restarting" error to terminate the task.  You pcall wrapper isn't needed; just let the API call throw.

The SDK doesn't offer an immediate shutdown function. The only alternatives are to do a `while (1) {}` which will will hang the CPU for a couple of seconds before the H/W watchdog resets the CPU, or to call the undocumented ROM function `_ResetHandler()` which does an immediate reset.  Clean shutdown is best, I think.

TerryE, 2020-09-01:
Maybe we could add this to the documentation but it would probably just confuse most programmers.

II think that this one can be closed.

mk-pmb, 2020-09-01:
@TerryE then we should somehow adjust the docs saying

> the ESP is then immediately rebooted, so control is not returned to the calling Lua application in the case of a successful reload.

because for me that sounded like on success, the pcall should never have returned any value and thus print could never be invoked.
However, if it was successful as expected, why the error after reboot? Shouldn't it have thrown another error earlier, and then *not* reboot, so I can transmit a better version of the LFS image?

TerryE, 2020-09-01:
I am about to finish the on_ESP LFS changes.  I will tidy this up as part of the documentation rewrite for this.

closed 2020-09-01 by TerryE