nodemcu/nodemcu-firmware #3227: ESP32:I2S Enhancements

prtomlinson, 2020-07-21:
First, thanks to the devTeam for adding I2S support to the ESP32 version of Nodemcu. I have a working setup for standard wav(stereo 16bit signed PCM) data streams. This requires the input byte stream to be 32 bits/sample coded and as far as I can tell each sample is Left{LSB,MSB} then Right(LSB,MSB}. This is OK as the mono sound power IC I'm using supports this mode and  the ability to mix Left/Right. However, the current files I'm playing are mono and require I add a null right channel. Too muck work!

As far as I can tell, the current modules code does not support the options discussed in the docs option table. I'm not a professional code guy and new to freeRtos. I'd like to add code that plays mono files or mixes two mono files. I do not ask your group to do the work. I will write and test the code, but I need some support understanding the code structure. In return for the team's help I will be happy to submit my final code plus docs for inclusion  into the project.

BTW, sram space is not an issue for me. Current I'm using 2 dma buffers of 1024 bytes. Current available heap is 185KB. My current firmware build size is 884KB.

Thanks for your review,
 tekguy49@gmail.com

marcelstoer, 2020-08-03:
The initial code came from @zelll and was added by @devsaurus - neither of them has been active around here for quite some time. I'm not sure if anyone else here is familiar with I2S enough to assess your extensions.

prtomlinson, 2020-08-03:
Marcel,
 Thanks for the feedback. I've been using the i2s code under nodemcu(ESP32
dev) and it seems quite stable. FYI, the esp32 version of the code is
stable and I've converted some code from the Esp8285 to the ESP32 with no
issues.

Congrats to the development team.
Regards,
 Paul Tomlinson

closed 2021-09-07 by stale[bot]