nodemcu/nodemcu-firmware #3228: RFC: Extensions to the Lua API for load / save of functions

TerryE, 2020-07-22:
With #3206 I am adding the ability to create LFS images on-ESP.  This require two enhancements to the compiled code save/store functionality:
1.  The binary formats for loading functions into RAM and into LFS are now the same.
2.  You can store multiple top level functions in the same compiled file.

The standard [`string.dump`](https://www.lua.org/manual/5.3/manual.html#pdf-string.dump) (and [`lua_dump`](https://www.lua.org/manual/5.3/manual.html#lua_dump), etc.) currently take a Lua function as parameter.  Our enhancement will _also_ allow an array parameter where the values are functions and the dump stream / file will now contain multiple functions.  By default the function name is taken from the `Proto->source` field, but this can be renamed using a `name=func` entry in the array.   So this is an upwards compatible enhancement.

What I want to discuss and agree is how we extend the `load` / `loadfile` / `dofile` APIs.
-  In Lua a source file can only compile to a single top level function (TLF) 
-  These APIs should all work as now when referencing a Lua code file the only contains one TLF.
-  `dofile` does not do a protected call and throws any errors (so for example doing an invalid source file will throw a syntax error). `dofile` can only meaningfully execute a single TLF code file, so it should likewise throw a "multiple functions found" error.

IMO we have two feasible approaches for the load APIs:
1.  The existing call work as now and throw an error if attempting to load a multiple TPF input.  We have separate `loadmultiple`, etc. functions which return an array of functions instead of a singleton.
2.  We overload the current functions, so that they return a function if the input contains one TLF and an array if it contains multiple TLFs

Either way will work fine without introducing code breaks so long as LC files only contain a single TLF.  **_Issues will only arise if Lua developer creates a multiple TLF code file and then tries to reload this using legacy code._**  For (1) the load will throw a  "multiple functions found" error, and for (2) the load will succeed and return an array, so trying to execute this as a function will throw a "attempt to call a table value" error, so in most circumstances the error will get thrown a line or two later.

My own vote is for (2).

pjsg, 2020-07-22:
As I understand it, the only way of creating a multiple TLF file is to use one of the `dump` methods and passing it an array. When the load happens, a table is returned with entries of funcname=function in it. If so, then I'm +1 on option 2.

jmattsson, 2020-07-22:
I have nothing against (2) here. It's a clean extension, and it's not something I can see anyone having sanely depended on the non-extended behaviour.

HHHartmann, 2020-07-22:
2 is fine for me too. Especially when loading a file of unknown content multiple/single TLF it can be checked by is Type in the next line instead of errorhandling.

TerryE, 2020-07-22:
The real reason for my supporting multiple TLFs in an LC file is that the LC file format and the LFS image file format are the same.  You need to be able to have multiple TLFs in an LFS image, and most of the code is shared, so it makes sense to tolerate multiple TLFs for at least one Lua load function to make it easy to "make" LFS images on ESP.

If you think about this, it also gives you a ZIP-like facility to bundle a set of modules in a single LC file if you want. Loading a multi-TLF returns an array of `{method=function}`, pretty much the same as `require(someModule)` does. 

Lastly the main scaling issue with on-ESP building of LFS images is the RAM limit for compiling source files. Once we've stabilised the compile file format, there is nothing to stop us making versions of our Lua modules in Lua compiled format, or offering a cloud service to bundle these -- as I already do, BTW.

TerryE, 2020-09-01:
Back to work on this tomorrow :smile:

TerryE, 2020-09-07:
A related aspect of this issue is the handling of debug stripping.  See [`node.stripdebug()`](https://nodemcu.readthedocs.io/en/dev/modules/node/#nodestripdebug) and the configuration setting [`LUAI_OPTIMIZE_DEBUG`](/nodemcu/nodemcu-firmware/blob/dev/app/include/user_config.h#L255).  In the days of RAM-only loading of code, stripping out the debug info might save perhaps 30-40% of the RAM footprint of functions in memory so losing meaningful error diagnostics might be a worthwhile trade-off when it could mean the difference between your application fitting in memory and it not doing so.  We support three levels of error information:
1. All debug metadata are retained
2. Local and Upvalue metadata are discarded but line numbering is retained.  
3. All debug metadata are discarded.

IMO, the best setting  is level 1 for LFS code, and level 2 and functions loaded into RAM. The current default is 2, but this can be updated by setting `node.stripdebug(1)`, for example.  Currently `node.compile()` forces strip=2 which is also the default.
  
What I want to do is for _**the dump, load and compile functions to honour the current strip default**_; this can be changed by `stripdebug()`) rather than the static as-built `LUAI_OPTIMIZE_DEBUG` setting.   Note that if  `stripdebug()`) isn't called then the as-built setting still applies, and since most developers don't even know about this option, let alone call it, the behaviour is unchanged for them.  This would allow on-ESP operations such as
```Lua
node.stripdebug(1)
node.compile('fred.lua')
node.compile('jim.lua')
a={} for k in pairs(node.LFS.list()) do 
  if k ~= 'fred' and k~= 'jim' then a{#a+1} = node.LFS[k] end
end
node.dumpfile('lfs.lc', a)
node.LFS.reload('lfs.lc', 'fred.lc','jim.lc'}
```
or even have this sort of update as a canned `update` script in LFS:
```Lua
node.LFS.update{'fred','jim'}
```

jmattsson, 2020-09-08:
Sounds good to me!

closed 2022-04-16 by stale[bot]