nodemcu/nodemcu-firmware #3229: Add modules to luac.cross which do not have hardware dependencies

HHHartmann, 2020-07-22:
### Missing feature
Use non HW dependent modules in luac.cross

Modules are at least:
struct.c bit.c color_utils.c sjson.c   // work without problems
bloom.c crypto.c encoder.c           // would need some more work

### Justification
help creating tests and allow usage of known functionality on the host.

### Workarounds
none

TerryE, 2020-07-23:
The `luac.cross` environment already supports `os` and `io`.  I did look at adding `file` but decided that given we already support `io`, I could not see the point of doing this dev work.

`struct` since this is already implemented in the Lua 5.3 `string` library (see [String pack and unpack](https://www.lua.org/manual/5.3/manual.html#6.4.2)), so it is really redundant in a 5.3 world.

Ditto for deprecating `bit` as these bitwise operations are natively supported in Lua 5.3.

`sjson` and possibly `crypto.c` `encoder.c` might make sense, but I see the main potential advantage if we have some form of `LFS` object so that we can use the `-e` option to generate Protos for saving in a multi-TLF compiled file.  This way developers could create resource functions.  But let me get on-ESP LFS update and two LFS regions first, eh?  Because this is going to rework luac.cross quite a lot.

HHHartmann, 2020-07-24:
> `file` could be added without much difficulty.

Hmm file depends on vfs which operates on SPIFFS or FAT. SPIFFS uses platform_flash_read/write to access the storage. So some work to add another blockdevice would be necessary.

TerryE, 2020-07-24:
@HHHartmann Grogor, as I said we already have `io` for host filesystem I/O.  The one potential use for `file` that could be of benefit is a wrapper around SPIFFS the same way that `spiffsimg` does.  This way you could use Lua scripts to provision a SPIFFS on host.  A well-scoped and entirely doable little project.

HHHartmann, 2022-05-01:
Still would like the associated PR to get in.
Is anybody willing to back this?