nodemcu/nodemcu-firmware #3232: Wifi connection lost

chathurangawijetunge, 2020-07-29:
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
branch: dev
commit: 2fa63a1
release:
release DTS: 202007071335
SSL: false
build type: integer
LFS: 0x40000 bytes total capacity
modules: file,gpio,mqtt,net,node,rtctime,sntp,tmr,uart,wifi
build 2020-07-08 00:54 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

Wi-Fi connection lost in long run... will not auto reconnect 
Till power cycle....
Earlier version works fine...

KT819GM, 2020-07-29:
Duplicate of #3208. I have difficulties to reproduce issue in repeatable manner. Do you have a way for devs to get consistent issue? Can you post your WiFi code part? I'm also using esp07 modules, but with current dev (compiled natively) I have stable WiFi, not as I've posted in #3208

nwf, 2020-08-01:
This appears to be a complete duplicate of #3208.  Please do not spuriously open redundant issues.

closed 2020-08-01 by nwf