nodemcu/nodemcu-firmware #3237: Can you please check the SPI module.

NeilSayok, 2020-08-06:
I was trying to read data from an SD Card and use a shift register SN74HC595 using the spi module. Even though I have built the Firmware with FatFS and SPI module activated they are returning a nil value. I have tried mounting the SD Card multiple times even using loops with delays but all results the same.

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)
vol = file.mount("/SD0", 8) 
print(vol)
```
### NodeMCU startup banner
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: dev
	commit: bf15299cea9b1d544499f99f2a2559b422ac0507
	release: 
	release DTS: 202008031858
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: file,gpio,http,i2c,mdns,net,node,rtctime,sjson,sntp,spi,tmr,uart,websocket,wifi
 build 2020-08-05 19:05 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
NodeMcu Lolin V3

galjonsfigur, 2020-08-28:
I've just tested it and everything seems to work. I used WeMos D1 Mini and SD Card Shield (with added jumper to pin D8 instead of D4 for convenience). I used similar code taken straight from [the](https://nodemcu.readthedocs.io/en/dev/sdcard/) [documentation](https://nodemcu.readthedocs.io/en/dev/modules/file/):
```lua
spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)

-- initialize other spi slaves

-- then mount the sd
-- note: the card initialization process during `file.mount()` will set spi divider temporarily to 200 (400 kHz)
-- it's reverted back to the current user setting before `file.mount()` finishes
vol = file.mount("/SD0", 8)   -- 2nd parameter is optional for non-standard SS/CS pin
if not vol then
  print("retry mounting")
  vol = file.mount("/SD0", 8)
  if not vol then
    error("mount failed")
  end
end

file.chdir("/SD0")
l = file.list();
for k,v in pairs(l) do
  print("name:"..k..", size:"..v)
end
-- Prints file list from my SD card as it should
```

It seems like SD card isn't properly connected to the ESP8266. What happens if you remove SN74HC595? Maybe there is problem with chip select or the other lines - it's hard to guess without any schematic.

marcelstoer, 2020-08-29:
@galjonsfigur thanks for the confirmation that this doesn't appear to be a firmware issue. In accordance with #1010 I am closing this.

closed 2020-08-29 by marcelstoer