nodemcu/nodemcu-firmware #3245: Compiling lastest release with Lua 5.3 and TLS + SSL enabled breaks tls.c module

Edduaardo, 2020-08-21:
### Expected behavior
Building NodeMCU with Lua 5.3, TLS and SSL enabled with no problems

### Actual behavior
Compiling [lastest release](https://github.com/nodemcu/nodemcu-firmware/releases/tag/3.0-master_20200610) with Lua 5.3 (make LUA=53), TLS and SSL enabled breaks tls.c module whereas building with Lua 5.1 (just the make command) builds fine.

### Make output
```
CC app/modules/tls.c
tls.c: In function 'append_pem_blob':
tls.c:388:7: error: implicit declaration of function 'isspace' [-Werror=implicit-function-declaration]
       if (isspace(*(uint8_t*) pem)) {
       ^
tls.c: In function 'tls_cert_auth':
tls.c:466:5: error: implicit declaration of function 'lua_unref' [-Werror=implicit-function-declaration]
     lua_unref(L, ssl_client_options.cert_auth_callback);
     ^
tls.c:470:29: error: 'LUA_TLIGHTFUNCTION' undeclared (first use in this function)
       || (lua_type(L, 1) == LUA_TLIGHTFUNCTION)) {
                             ^
tls.c:470:29: note: each undeclared identifier is reported only once for each function it appears in
tls.c:471:5: error: implicit declaration of function 'lua_ref' [-Werror=implicit-function-declaration]
     ssl_client_options.cert_auth_callback = lua_ref(L, 1);
     ^
tls.c: In function 'tls_cert_verify':
tls.c:525:29: error: 'LUA_TLIGHTFUNCTION' undeclared (first use in this function)
       || (lua_type(L, 1) == LUA_TLIGHTFUNCTION)) {
                             ^
cc1: all warnings being treated as errors
```

### Modules enabled
user_modules.h
```C
#define LUA_USE_MODULES_ADC
#define LUA_USE_MODULES_BIT
#define LUA_USE_MODULES_DHT
#define LUA_USE_MODULES_FILE
#define LUA_USE_MODULES_GPIO
#define LUA_USE_MODULES_HTTP
#define LUA_USE_MODULES_I2C
#define LUA_USE_MODULES_NET
#define LUA_USE_MODULES_NODE
#define LUA_USE_MODULES_OW
#define LUA_USE_MODULES_SJSON
#define LUA_USE_MODULES_SPI
#define LUA_USE_MODULES_TLS
#define LUA_USE_MODULES_TMR
#define LUA_USE_MODULES_UART
#define LUA_USE_MODULES_WIFI
```

user_config.h
```C
#define CLIENT_SSL_ENABLE
```
I don't know if this is a problem of the TLS module being incompatible with Lua 5.3 or NodeMCU, but since we are moving to Lua 5.3 soon, I thought to report this here.

HHHartmann, 2020-08-21:
Just to make sure: did you do a `make clean` in between the builds?

Edduaardo, 2020-08-21:
Yes, I always do that between builds

KT819GM, 2020-08-21:
Just did:
```C
make clean
make -j8 LUA=53
```
with current `dev` branch. Everything compiled successfully, could you please check if it's compiling for you on `dev`.

galjonsfigur, 2020-08-21:
I remember that error - it happens when you switch from Lua 5.1 to 5.3 (or vice-versa - not sure) but without doing proper `clean`. Try doing 
```
make LUA=51 clean
make LUA=53 clean
```

nwf, 2020-10-01:
I believe this fixed, now that the release has happened.

closed 2020-10-01 by nwf