nodemcu/nodemcu-firmware #3256: Make user_config.h easier to update: Split out the demons part

mk-pmb, 2020-08-29:
Can we please split the `// *** Heareafter, there be demons ***` part of `user_config.h` into a separate file? That would make it a lot easier for me to rebase, and to upgrade outside of git.

TerryE, 2020-08-29:
@mk-pmb, I would hold off on this.  We've got the unification of the ESP8266 and ESP32 branches on the roadmap.  In this scenario, `user_config.h` get binned / auto-generated.  Take an example out of the scripts and just add your own override script using some `sed` commands as part of your build wrapper.  That way you can to a git fetch / pull and remake.

mk-pmb, 2020-08-29:
Nice! Do you have an estimate when it will land?

mk-pmb, 2020-08-29:
Meanwhile I found that the split seems rather feasible after all, the required adjustments to the Makefile are minimal and are a good change for style reasons, too.

TerryE, 2020-08-29:
Lot's of balls to juggle and very few hands on.  This sort of tweak can have side effects, so I tend to apply a high threshold for integrating these.

marcelstoer, 2020-08-29:
My vote is for closing this as "won't do". Too many side effects.

mk-pmb, 2020-08-29:
> My vote is for closing this as "won't do". Too many side effects.

I thought the same at first, but so far I haven't found any. My major concern were the greps on gcc output, but gcc still detects all defines correctly. However, I paused my research half way through due to Terry's reply.
Would you mind sharing the details of some of the side effects you found?

closed 2020-08-29 by mk-pmb