nodemcu/nodemcu-firmware #3261: tmr.create():alarm() Broken on New (dev)

chathurangawijetunge, 2020-08-30:
**Device :** ESP8266

NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: dev
	commit: d72ea91ed01518ef0779d7aa998c5c3a9c269d71
	release: 
	release DTS: 202008291943
	SSL: false
	build type: integer
	LFS: 0x40000 bytes total capacity
	modules: file,gpio,mqtt,net,node,si7021,sntp,tmr,uart,wifi
 build 2020-08-30 03:07 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
cannot open init.lua:

**Code:-**
if not tmr.create():alarm(5000, tmr.ALARM_SINGLE, function()
  print("hey there")
end)
then
  print("whoopsie")
end

**Error:-**
Lua error: 	stdin:1: bad argument #1 to 'alarm' (boolean expected)
stack traceback:
	[C]: in function 'alarm'
	stdin:1: in main chunk
	[C]: ?
	[C]: ?

galjonsfigur, 2020-08-30:
Can confirm - commit d72ea91ed01518ef0779d7aa998c5c3a9c269d71 is causing this issue and it has to do with how `alarm` function is done in C code. It looks like becuse of `tmr:start` now accepts arguments, function `tmr_alarm` would have to manipulate the stack somehow before calling `tmr_start` - I could be wrong as I'm not an Lua C API expert.   @vsky279 could you look at this issue?

marcelstoer, 2020-08-30:
We definitely need to fix this before the next master drop 😞

vsky279, 2020-08-30:
I'll have a look. Sorry about that.

mk-pmb, 2020-08-31:
Let's discuss strategies for automated hardware testing on Gitter.

vsky279, 2020-08-31:
I think the fix is ready to be merged.

PR #3263 (2020-09-05): Fixes broken `tmr.alarm`
Fixes #3261.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Fixes broken `tmr.alarm` following the commit [d72ea91](https://gi [... cut]
closed 2020-09-05 by marcelstoer