nodemcu/nodemcu-firmware #3265: String formatting: plus sign ignored for %+g

mk-pmb, 2020-08-31:
### Expected behavior
My number has a plus sign, as on Ubuntu:
```sh
$ lua <<<'print(("%+g"):format(123), _VERSION)'
+123    Lua 5.2
```

### Actual behavior
No sign in the REPL on my ESP8266 (using [this firmware](https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/actions/runs/233185495) built with `LUA=53` from [current dev branch + tmr fix](https://github.com/mk-pmb/nodemcu-firmware/tree/lua53-dev).

### Test code
```lua
> print(("%+g"):format(123), _VERSION)
123     Lua 5.3
```

### NodeMCU startup banner
```text
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
        branch: lua53-dev
        commit: 9ff8d9f3e99b5c815b90cc40cb93b0102f80c2dd
        release: 3.0-master_20190907 +104
        release DTS: 202008311952
        SSL: false
        build type: float
        LFS: 0x10000 bytes total capacity
        modules: adc,bit,coap,color_utils,cron,crypto,encoder,file,gpio,gpio_pulse,http,i2c,mdns,mqtt,net,node,ow,pcm,perf,pipe,pwm,pwm2,rfswitch,rotary,rtcfifo,rtcmem,rtctime,sigma_delta,sjson,sntp,spi,struct,tls,tmr,uart,websocket,wifi,wifi_monitor
 build 2020-08-31 19:55 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
cannot open init.lua: 
```

### Hardware
ESP8266 LoLin v3

vsky279, 2020-09-06:
@mk-pmb Can you please check the fix in the PR #3270?

mk-pmb, 2020-09-06:
I tested it with [this firmware](https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/actions/runs/242110725) and [this program](https://gist.github.com/mk-pmb/be928b5508c8cb0558ae50e07fa82210). The [results](https://gist.github.com/mk-pmb/2e8138484e9b4d1144df6f079e342b30) show that now the + is the same on NodeMCU as on Ubuntu's LUA, thanks!

However, at the same time they indicate they have a different interpretation of the number of spaces when the sign is omitted. Also you might wonder why I used 3.1410 as the number, not 3.1415… that's because they also have different opinions about rounding.
Should we extend the issue and PR to tackle that?

vsky279, 2020-09-07:
Thanks for testing. I'll try to have a look at both issues. If there's easy way to fix it it can be done.

nwf, 2020-09-11:
This didn't get picked up as fixed automagically.  If it should remain open, please feel free to reopen it!

mk-pmb, 2020-09-12:
I tested with [this firmware](https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/actions/runs/250690930) built from commit 64bbf006898109b936fcc09478cbae9d099885a8 = the current dev branch. The results are as from my previous test above.

@vsky279 Since the MR is merged now, I guess it would be confusing to continue the discussion there. Will you make a new MR for the spacing and/or rounding bugs? Shall I open issues for them?

vsky279, 2020-09-12:
You were too fast guys. I suggest reopening this issue rather than opening another one.

I have almost ready the PR for testing for the `' '` flag (e.g. `% 8.3f`).

As for the rounding - I think it is not something to be corrected. I think the difference @mk-pmb observes between Ubuntu Lua 5.2 and NodeMCU Lua 5.3 is due to the fact that we are using 32bit float while Ubuntu is using 64bit double.

3.1415 does not has precise representation in 32bit float. It is in fact stored as 3.141499996185303 (`0x40490e56 =  
2.00000000 * 1.5707499980926514`) so when rounded to 3 decimals we are still at 3.141. However I don't know how C manage to print 3.1415 instead of 3.141499...
Maybe some more experience programmer can elaborate on this.

mk-pmb, 2020-09-12:
(If anyone else wonders: He's using big endian notation.) Maybe they multiply by 10\*\*precision first, or calculate the next decimal digit and decide based on that.

vsky279, 2020-09-14:
@mk-pmb please test the PR #3283.

mk-pmb, 2020-09-15:
Will do, probably tomorrow. Meanwhile: For changing the rounding I think we should ensure a wide consensus, do we have that?

vsky279, 2020-09-15:
Definitely there needs to be wide agreement. This is also why I have two separate commits.

mk-pmb, 2020-09-15:
Done for ESP8266. Meanwhile I got an ESP-32 as well, but it may take another few hours until I can test on that.

pjsg, 2020-10-04:
I think that this should be solved with the merging of the snprintf implementation. Can you verify?

vsky279, 2020-10-04:
I tested this already. Both flags '+' & ' ' work fine with the new snprintf implementation.

The double rounding - it would be nice to have it implemented. Is it possible to submit a PR on source files provided by a submodule?

pjsg, 2020-10-04:
I think that you should submit the PR to the original module....

vsky279, 2020-11-07:
I think that with 64-bit doubles (PR #3225) there's no need to have the double rounding implemented.
If someone needed precise rounding with doubles it is achieved (`"%.3f"%3.1415` gives `3.142` with `#define LUA_NUMBER_64BITS`). So think we can close this issue.

nwf, 2020-11-17:
Closing by request of @vsky279.

PR #3270 (2020-09-09): Adding '+' flag to string.format
Fixes #3265.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

It seems that adding the "+" formatting flag is not that complex.
I've done several tests but further independent testing is needed before it's merged.
closed 2020-11-17 by nwf