nodemcu/nodemcu-firmware #3268: On i686 GNU/Linux, luac.cross sneakily builds invalid LFS images that fool even node.LFS.reload

mk-pmb, 2020-09-01:
### Expected behavior
* `cd app/lua/luac_cross && make LUA=53 clean all` either builds a useful executable or refuses with a link to this issue to explain why we don't support i686.
** I'd accept "nobody uses 32 bit CPUs anyway" if someone is bold enough to make that claim.
* `node.LFS.reload()` detects _before_ reboot that the reboot would fail, and thus, would not reboot at all.

### Actual behavior
When I create an LFS image with luac.cross compiled from commit b4c148e (current dev branch) with LUA=53 on Ubuntu on an x86_64 machine, it works as expected.

When I try the same on an i686 machine, it goes through all the motions with no sign that anything is wrong, and not even the pass 1 check of node.LFS.reload complains. Only after the MCU reboot it fails with `(null): invalid header in precompiled chunk`.

* [The LUA source and both resulting LFS images](http://web.archive.org/web/20200901202306if_/http://l.proggr.de/misc/proggr-dl/20/75g5la0/hello.zip), maybe it can help determine what went wrong.
* [REPL session log](https://gist.github.com/mk-pmb/97dff32db09244c7507ed54d837c42dc)

### Test code
```lua
-- -*- coding: UTF-8, tab-width: 2 -*-
return function () print('Hello.') end
```

```bash
( cd app/lua/luac_cross && make LUA=53 clean all
) && sudo mv -v luac.cross /usr/local/bin/luac-for-nodemcu
luac-for-nodemcu -f -o hello.i686.lfs hello.lua
```

### NodeMCU startup banner
See REPL log.

### Hardware
LoLin v3 ESP8266

TerryE, 2020-09-02:
If you look at the code, then you will see that we support both 32-bit and 64-bit hosts.  I understand your report that reload isn't working for an LFS image built on a 32-bit host, but at worst this is a bug and not some sort of deep-state conspiracy, so please avoid this type of wording.   You are free to use sarcasm when you are paying for my time; it really doesn't help when I am giving it pro bono.

TerryE, 2020-09-02:
OK, I have just tried an image build on my i5 laptop and an RPi4 running 32-bit rasbian compiling _init.lua into an LFS image.  They are identical, so we haven't got fundamental error here.  You will need to fix the link before I can do more.  I might have some old 32-bit VM images that I can spin up, but I can't see any reason for the output from a 32-bit ARM compiled version and a 32-bit x86 compiled version to be different.

TerryE, 2020-09-02:
On reflection saying as there isn't an error in RPi variants and I hope to have the next feature release within the week, I will check that for 32 x86 working.

mk-pmb, 2020-09-03:
Sorry if this came accross the wrong way, it wasn't meant against you or anyone in person. The wording "sneakily" was intended to be funny, I guess it wasn't after all. Sorry again.
The 32bit remark was to clarify that it's not overly important to me to continue the 32bit support. Lots of projects (e.g. Ubuntu) have chosen to drop 32bit versions lately, so considering the scarcity of C devs here, I considered the possibility that, if this bug could be too cumbersome, we might follow them. (Not saying we should though.)

No idea why the Wayback Machine deleted the file. I fixed the first link to point to [the live version](http://l.proggr.de/misc/proggr-dl/20/75g5la0/hello.zip) instead.

mk-pmb, 2020-09-03:
By accident I discovered that there is a luac in the lua53 directory as well, its subdirectory called "host". That one works perfectly. So I guess the correct fix for the first part is to make the 5.1 luac Makefile refuse to build if the "LUA" variable is not empty, with a hint where the 5.3 luac is. I'll prepare a PR for that.

TerryE, 2020-09-06:
The makes are a bit cludgy when you swap versions.  you really need to `make LUA=51 clear` before doing the make LUA=53" and v.v.

mk-pmb, 2021-09-08:
I still want this.

TerryE, 2021-09-26:
OK, let's not close this.