nodemcu/nodemcu-firmware #3276: The SNTP PLL is bit aggressive

pjsg, 2020-09-10:
### Expected behavior

The PLL that controls the clock should settle down and track the crystal in a fairly stable fashion.

### Actual behavior

It is a bit sensitive to estimates of the current time. This is especially a problem when the path from the nodemcu system to the actual clock is long and not very stable.

nwf, 2020-09-10:
I spent a while tweaking the PI controller in `sntppkt`; I wonder if that behaves better in your environment?

closed 2022-04-16 by stale[bot]

pjsg, 2022-04-17:
With the old PLL, this is the typically performance that I get. 

[image]