nodemcu/nodemcu-firmware #3277: Error with file.read() if filesize between  1024 and 2048

bazooka07, 2020-09-12:
### Expected behavior
I have on the ESP8266 a file "script.js" with a filesize of  1132 bytes. I am reading the content of the file chunk by chunk. The first part has the length of  1024 bytes. But the second part is "nil".

The file is on github at https://gist.github.com/bazooka07/9b7df0832cf6947ff5ae64bf4168b669

### Test code
I'm a Linux user. I make this test with picocom via the usb port ( _picocom -b 115200 /dev/ttyUSB0_ )
```Lua
> =file.list()["script.js"] -- says 1152
> file.open("script.js")
> chunk=file.read()
> =#chunk -- says 1024. Right !
> chunk=file.read()
> =#chunk -- gives an error: stdin:1: attempt to get length of global 'chunk' (a nil value)
> =chunk -- says nil !!!!
```
### NodeMCU startup banner
NodeMCU 3.0.0.0 
	branch: perso
	commit: f41df18444331a7ad62461bf7eff356cbefde50a
	release: 3.0-master_20200910 +1
	release DTS: 202009120653
	SSL: false
	build type: float
	LFS: 0x32000 bytes total capacity
	modules: adc,am2320,bit,bme280,crypto,dht,encoder,file,gpio,http,i2c,mdns,mqtt,net,node,rotary,rtcmem,rtctime,sjson,sntp,spi,struct,tmr,uart,wifi
 build 2020-09-12 08:58 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
cannot open init.lua: 

I compile the firmware with "Ubuntu 20.04.1 LTS"

### Hardware
The board is an ESP8266 Wemos D1 with 4MB of flash

KT819GM, 2020-09-12:
Is there any particular reason not to use `file.getcontents()`? Like:
```Lua
print(file.getcontents('script.js')) 
```

bazooka07, 2020-09-12:
Because I want to use with a websocket server :
https://github.com/whendonkiesfly/esp8266-websocket-server

KT819GM, 2020-09-12:
Not sure why it behaves like that on your side, just tried:
```Lua
> Uploading to ESP file script.js...Success

> =file.list()["script.js"]
1151
> file.open("script.js")
> chunk=file.read()
> =#chunk
1024
> chunk=file.read()
> =#chunk
127
> =chunk
ent)

		this.send('status')
	}

	ws.onclose = function(event) {
		console.log('Websocket closed')
		console.log(event)
	}

})()
```
I'm on 5.3 though. Let's wait for `dev's` to take a look at it.

bazooka07, 2020-09-12:
Sorry, but it is not working with LUA 5.3. I make "esptool erase_flash;  LUA=53 make; make flash4m"
Here is my banner on the ESP8266 board :
```
NodeMCU 3.0.0.0 
	branch: perso
	commit: f41df18444331a7ad62461bf7eff356cbefde50a
	release: 3.0-master_20200910 +1
	release DTS: 202009120653
	SSL: false
	build type: float
	LFS: 0x32000 bytes total capacity
	modules: adc,am2320,bit,bme280,crypto,dht,encoder,file,gpio,http,i2c,mdns,mqtt,net,node,rotary,rtcmem,rtctime,sjson,sntp,spi,struct,tmr,uart,wifi
 build 2020-09-12 11:17 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
cannot open init.lua: 
```
Notice that I'm using LFS.
Can you give me your banner with your modules in use ?

KT819GM, 2020-09-12:
Yes, I'm using `LFS` also, but your test file was on `SPIFFS`, about banner - sure:
```Lua
TestDev 3.0.4.0 
	branch: dev
	commit: 64bbf006898109b936fcc09478cbae9d099885a8
	release: 3.0-master_20200910
	release DTS: 202009090323
	SSL: true
	build type: float
	LFS: 0x40000 bytes total capacity
	modules: adc,bit,bme280,crypto,dht,encoder,enduser_setup,file,gpio,gpio_pulse,http,i2c,mdns,mqtt,net,node,ow,pipe,sjson,softuart,spi,tls,tmr,uart,u8g2,wifi
 build 2020-09-11 21:07 powered by Lua 5.3.5 on SDK 3.0.4(9532ceb)
```

KT819GM, 2020-09-12:
Don't forget to `make clean` before building `Lua5.3`:
```C
make LUA=53 clean && make -j8 LUA=53
```

galjonsfigur, 2020-09-12:
I've looked into it and it's a really interesting bug. I can reproduce it on both Lua 5.3 and Lua 5.1. @KT819GM It seems that you cannot reproduce it because your `script.js` file is one byte too short. If the file is exactly 1152 (1024 + 128) the it fails to get those last 128 bytes. I'm now looking into the code. @bazooka07 Can you try adding a semicolon at the end (or any other character somewhere in the js code) and see it it works?

KT819GM, 2020-09-12:
>@KT819GM It seems that you cannot reproduce it because your `script.js` file is one byte too short. If the file is exactly 1152 (1024 + 128) the it fails to get those last 128 bytes.

My bad, apologies, by being lazy I've created file from clipboard thus removing the `whitespace` at the end of original file. With original file I can reproduce bug, symbol (replacement of whitespace) does not make difference (if file size stay 1152). If size 1153 - no bug.

galjonsfigur, 2020-09-12:
To be honest it's good you did that - thanks to it I was able to determine that the exact file length was key to reproduce this error :smile:

bazooka07, 2020-09-12:
@galjonsfigur 

I have try with different filesizes : 1151, 1152, 1153, 1154. Only 1152 makes an issue (1152 = 1024 + 128).

pjsg, 2020-09-12:
It looks to me (from code inspection) that reading a big chunk where the file has a multiple of 128 bytes left will fail

```
  for (j = 0; j < n; j += sizeof(p)) {
    int nwanted = (n - j >= sizeof(p)) ? sizeof(p) : n - j;
    int nread   = vfs_read(fd, p, nwanted);

    if (nread == VFS_RES_ERR || nread == 0) {
      // I suspect that we should break if j > 0 instead
      lua_pushnil(L);
      return 1;
    }

    for (i = 0; i < nread; ++i) {
      luaL_addchar(&b, p[i]);
      if (p[i] == end_char) {
        vfs_lseek(fd, -nread + j + i + 1, VFS_SEEK_CUR); //reposition after end char found
        nread = 0;   // force break on outer loop
        break;
      }
    }

    if (nread < nwanted)
      break;
  }
```

pjsg, 2020-09-13:
Fixed in the dev branch

PR #3280 (2020-09-13): Fix problem reading files a multiple of 128 bytes long
Fixes #3277.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [N/A ] The code changes are reflected in the documentation at `docs/*`.

If we have successfully read some [... cut]
closed 2020-09-13 by pjsg