nodemcu/nodemcu-firmware #3278: dofile() want excite files in Flash with LFS

chathurangawijetunge, 2020-09-12:
I have `_init.lua` in my `LFS` if i `dofile("abc.lua")` where `abc.lua` is in my flash not in `LFS` I get a error
``attempt to call a nil value
in function 'dofile'
``

vsky279, 2020-09-15:
I can reproduce the error. It seem that the return parameters of  `node.flashindex` have changed (no second variable is returned).

I think the simplest implementation of both function is:
```Lua
local lf, df = loadfile, dofile
G.loadfile = function(n)
  local mod, ext = n:match("(.*)%.(l[uc]a?)");
  local fn = index(mod)
  if fn then return fn else return lf(n) end
end

G.dofile = function(n)
  local mod, ext = n:match("(.*)%.(l[uc]a?)");
  local fn = index(mod)
  if fn then return fn() else return df(n) end
end
```
We don't need to check the `ext` value for 'lua|lc' instead we can call `df(n)` directly and let the user face directly the error.

@TerryE Terry, could you please have look?

bazooka07, 2020-09-16:
I have had the same issue. 

I adjust _init.lua as follow :
```lua
package.loaders[3] = function(module) -- loader_flash
  return lfs_t[module]
end

local lf, df = loadfile, dofile
G.loadfile = function(n)
  local mod, ext = n:match("(.*)%.(l[uc]a?)");
  if ext ~= 'lc' and ext ~= 'lua' then return lf(n) end
  local fn   = index(mod)
  if fn then return fn else return lf(n) end
end

G.dofile = function(n)
  local mod, ext = n:match("(.*)%.(l[uc]a?)");
  if ext ~= 'lc' and ext ~= 'lua' then return df(n) end
  local fn = index(mod)
  if fn then return fn else return df(n) end
end
```
Notice that "local index = node.flashindex" is deprecated with the last release, but _init.lua is not updated. So, replace as follows :
```lua
local index = node.LFS.get
```
I'm using LUA 5.1. Not testing with LUA 5.3

nwf, 2020-10-23:
I believe this was just fixed in #3290 .  Please scream if that isn't true.

PR #3290 (2020-10-23): Further adaptation of lfs/_init.lua
See https://github.com/nodemcu/nodemcu-firmware/issues/3278 .  Thanks to
@vsky279 and @bazooka07 for suggestions.

Fixes #3278.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.
closed 2020-10-23 by nwf