nodemcu/nodemcu-firmware #3279: LUA 5.3 node.egc.setmode() error

chathurangawijetunge, 2020-09-12:
NodeMCU 3.0.0.0 
	branch: dev
	commit: 64bbf006898109b936fcc09478cbae9d099885a8
	release: 3.0-master_20200610 +32
	release DTS: 202009090323
	SSL: false
	build type: integer
	LFS: 0x40000 bytes total capacity
	modules: file,gpio,mqtt,net,node,rtctime,sntp,tmr,uart,wifi
 build 2020-09-12 15:45 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
> 

```
node.egc.setmode(node.egc.ON_MEM_LIMIT, 30720)
Lua error: 	stdin:1: attempt to index a nil value (field 'egc')
stack traceback:
	stdin:1: in main chunk
	[C]: in ?
	[C]: in ?
```

TerryE, 2020-09-12:
Not an error. `node.egc` was dropped as no longer relevant in 5.3.

closed 2020-09-13 by chathurangawijetunge

marcelstoer, 2020-09-13:
@chathurangawijetunge Also, we asked you many times to stick to our issue template like everybody else - it's there for a reason. In the future we might close non-compliant issues without any comments.