nodemcu/nodemcu-firmware #3285: Wifi.eventmon scheduling

TerryE, 2020-09-19:
This is really a quick placeholder that should be replaced by a proper bugrep when I have time to do this, but in responding to  @KT819GM's post on [#3208](3208#issuecomment-695178385). I note that the `wifi.eventmon` code does not comply with SDK scheduling rules and hence its extended use could possibly cause the WiFi stack to fail.  Specifically [line 96](https://github.com/nodemcu/nodemcu-firmware/blob/release/app/modules/wifi_eventmon.c#L96) uses a `task_post_low()` to reschedule itself.  This means that once monitoring is initiated, there will always be at least one instance of event checker scheduled or running, thus starving out any lower priority tasks.

@dnc40085 is the maintainer of this code, but hasn't been active on the project since May 2018.  If he isn't available to look at this then this is going to be a TODO for one of the active maintainers, I think. if necessary, I can investigate further once I've merge my current in ESP LFS enhancement. 

### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU startup banner
You MUST include the firmware startup banner to describe the version you are using. We reserve the right to immediately close any bug that doesn't.

### Hardware
Describe which ESP8266/ESP32 device you use and document any special hardware setup
required to reproduce the problem.

pjsg, 2020-09-19:
I'm not sure that I follow. To my reading of the code, when the callback is fired by the OS, it adds an entry to the queue and then schedules a low priority task to start cleaning the queue. The queue cleaning task then takes an element from the queue and calls the lua callback (if any). Also, if there are more entries in the queue, it reschedules itself. 

This seems to be a reasonable approach -- what am I missing?

I guess that it is possible, if there is a flood of events, that they will take some time to be processed, but I don't think that that is the case. Also, it isn't clear what you would do about that.

pjsg, 2020-09-19:
I would hope that:
```
function task()
   node.task.post(node.task.LOW_PRIORITY, task)
end
```
would not take down the system. Can we run our low priority stuff at the lowest conceivable priority? Otherwise, I think these sorts of issues will continue to arise...

TerryE, 2020-09-19:
> I would hope that:
> 
> ```
> function task()
>    node.task.post(node.task.LOW_PRIORITY, task)
> end
> ```
> 
> would not take down the system.

That's my point. You can get away with it for a period, say, maybe 60 seconds, but if you recursively post your own task then you are are blocking the scheduler from executing any lower priority tasks, as there is always at least one task ready to run at an application priority, so it will always schedule this task and not a lower priority task. This blocked SDK task might be something like wi-fi stack housekeeping.

TerryE, 2020-09-19:
Besides that, this sort of practice will leave the CPU 100% compute bound. Never a good practice. If the app is only occasionally using the stack then it seems far more sensible to me to do the status check just before you do the the network I/O sequence rather than continually hitting the network stack with validation / check calls.

pjsg, 2020-09-19:
The event monitor does not (as far as I can see) do a hard CPU loop. It can consume all the CPU time if the callback code is time consuming and there is a high rate of events.

Interstingly, the arduino port for the ESP8266 does use the lowest priority for running its code which is effectively in a hard loop. I think that we ought to do the same for our low priority stuff.

TerryE, 2020-09-19:
BTW, it look like you are are correct.  The issue that causes confusion for me is that the `wifi_` "namespace" is used by both the `app/modules/wifi*.c` routines _and_ the SDK sec 3.5 wifi API.  I feel that this is bad practice.  It would have been better to use a different prefix in our code, say, `wifimod_`  so you can tell at a glance what calls are local to the modules and what are SDK API calls.

dnc40085, 2022-04-16:
My apologies for the long absence, I just recently started working on my project again, now I'm trying to get my dev environment working again.
It has been a while since I wrote that code but I'm pretty sure what stood out was the section...
```c 
    if(queue_len == 0){ //if queue was empty, post task
      EVENT_DBG("Posting task");
      task_post_low(wifi_event_monitor_task_id, false);
    }
```
While it does indeed look like it just repeatedly posts the task, the section is just written out of order. The queue length is actually checked before the newest event is added, this is to check for any pre-existing events in the queue that still have yet to be processed. At this certain spot the task can only be posted when the queue is fully emptied and when the SDK calls the `wifi_event_monitor_handle_event_cb` function.

The only place the task is reposted is in the function [`wifi_event_monitor_process_event_queue`](https://github.com/nodemcu/nodemcu-firmware/blob/f25dc56d3c6213b8ac7ce46d1293466137746eae/app/modules/wifi_eventmon.c#L133) but this only happens if the queue is not empty after the event is processed.

Also, I agree... I've ran into the issue with the `wifi_` prefix far too many times when messing around with the NodeMCU source.

PR #3293 (2020-09-27): Fix platform_post by adding ICACHE_RAM_ATTR
Fixes #3285 (maybe).

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.

The ICACHE_RAM_ATTR is missing from the platform_post function and this could cause all sorts of failures under high event loads.