nodemcu/nodemcu-firmware #3287: Some numbers do not roundtrip through sjson correctly.

pjsg, 2020-09-20:
### Expected behavior

If `a` is a lua number then
```
a == sjson.decode(sjson.encode({a}))[1]
```

Specific example:
```
a = -4392806464851163000000000

rt_a = sjson.decode(sjson.encode({a}))[1]

if a ~= rt_a then
   print ("Different", a, rt_a, a-rt_a)
end
```

### Actual behavior

The code above prints:
```
Different	-4.392806e+24	-4.392806e+24	-5.764608e+17
```

### NodeMCU startup banner
```
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
	branch: float-test
	commit: 9d74cd5aa0960c10a9d8b04196d049023281e908
	release: 3.0-master_20200910 +4
	release DTS: 202009182147
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: adc,bit,dht,enduser_setup,file,gpio,i2c,mqtt,net,node,ow,sjson,spi,tmr,uart,wifi
 build 2020-09-20 13:08 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
```

I had to implement a couple of extra methods to be able to create uniformly distributed floating point values. I may well do a PR for those (simple) functions (which are under `#ifdef DEVELOPMENT_TOOLS`)

pjsg, 2020-09-20:
The actual differences are all (it seems) in the last 3 bits of the floating point representation. This suggests that adding one more decimal place to number format that `sjson` uses might fix the problem.

pjsg, 2020-09-20:
Actually, it turns out that you need two more decimal places...

pjsg, 2020-09-21:
And worse, you can actually crash the interpreter when formatting very large numbers with a %f format. The root cause of this turns out to be that we don't have an snprintf implementation, and so the length of the buffer is ignored(!). Thus formatting a number like 1e297 with %f leads to a huge buffer overflow. I suspect that the same thing happens in lua5.1, but I haven't actually tried it.

nwf, 2020-09-21:
:O Is `ets_snprintf` that badly broken?  Does Espressif know?

pjsg, 2020-09-21:
We don't use ets_snprintf as it doesn't have support for floats
(apparently). We have our own sprintf implementation..... I did some
looking for other open source implementations and found
https://github.com/weiss/c99-snprintf which looks plausible.
We just #define l_sprintf to be sprintf and throw away the length
argument.....

nwf, 2020-09-22:
Yikes!

pjsg, 2020-09-22:
I have now got the firmware to build with the new snprintf
implementation -- testing to follow....

pjsg, 2020-10-04:
I'm going to close this ticket out -- it now round trips correctly with floats, though there are some issues with doubles.....

PR #3291 (2020-09-27): Add snprintf and remove old sprintf
Fixes #3287.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This replaces the in-tree implementation of sprintf with snprintf (from https://github.com/weiss/c99-snprintf ). This is added as a submodule.

There are still places that call sprintf that ought to be calling snprintf.
closed 2020-10-04 by pjsg