nodemcu/nodemcu-firmware #3301: Bug in `file.readline` when line length exceeds 128 chars

vsky279, 2020-09-28:
### Expected behavior
`file.readline` is expected to read correctly lines longer than 128 chars.

### Actual behavior
When a file has lines with length exceeding 128 chars the line following such a line is read incorrectly.

Actually the bug is present in `file.read(end_char)` called with `end_char` as `file.realine` is just `file.read("\n")`.

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
h=file.open("test.txt", "w");
for i=65,75 do
    for j=1,i-55 do
        h:write(("%s%09d"):format(string.char(i), j))
    end
    h:write("\n")
end
h:close()

h=file.open("test.txt");
l=""; 
while l do 
  print(l:sub(1,-2))
  l=h:readline()
end
h:close()
```
Expected output:
```
A000000001A000000002A000000003A000000004A000000005A000000006A000000007A000000008A000000009A000000010
B000000001B000000002B000000003B000000004B000000005B000000006B000000007B000000008B000000009B000000010B000000011
C000000001C000000002C000000003C000000004C000000005C000000006C000000007C000000008C000000009C000000010C000000011C000000012
D000000001D000000002D000000003D000000004D000000005D000000006D000000007D000000008D000000009D000000010D000000011D000000012D000000013
E000000001E000000002E000000003E000000004E000000005E000000006E000000007E000000008E000000009E000000010E000000011E000000012E000000013E000000014
F000000001F000000002F000000003F000000004F000000005F000000006F000000007F000000008F000000009F000000010F000000011F000000012F000000013F000000014F000000015
G000000001G000000002G000000003G000000004G000000005G000000006G000000007G000000008G000000009G000000010G000000011G000000012G000000013G000000014G000000015G000000016
H000000001H000000002H000000003H000000004H000000005H000000006H000000007H000000008H000000009H000000010H000000011H000000012H000000013H000000014H000000015H000000016H000000017
I000000001I000000002I000000003I000000004I000000005I000000006I000000007I000000008I000000009I000000010I000000011I000000012I000000013I000000014I000000015I000000016I000000017I000000018
J000000001J000000002J000000003J000000004J000000005J000000006J000000007J000000008J000000009J000000010J000000011J000000012J000000013J000000014J000000015J000000016J000000017J000000018J000000019
K000000001K000000002K000000003K000000004K000000005K000000006K000000007K000000008K000000009K000000010K000000011K000000012K000000013K000000014K000000015K000000016K000000017K000000018K000000019K000000020
```

Actual output:
```
A000000001A000000002A000000003A000000004A000000005A000000006A000000007A000000008A000000009A000000010
B000000001B000000002B000000003B000000004B000000005B000000006B000000007B000000008B000000009B000000010B000000011
C000000001C000000002C000000003C000000004C000000005C000000006C000000007C000000008C000000009C000000010C000000011C000000012
D000000001D000000002D000000003D000000004D000000005D000000006D000000007D000000008D000000009D000000010D000000011D000000012D000000013
13E000000014
F000000001F000000002F000000003F000000004F000000005F000000006F000000007F000000008F000000009F000000010F000000011F000000012F000000013F000000014F000000015
13G000000014G000000015G000000016
H000000001H000000002H000000003H000000004H000000005H000000006H000000007H000000008H000000009H000000010H000000011H000000012H000000013H000000014H000000015H000000016H000000017
13I000000014I000000015I000000016I000000017I000000018
J000000001J000000002J000000003J000000004J000000005J000000006J000000007J000000008J000000009J000000010J000000011J000000012J000000013J000000014J000000015J000000016J000000017J000000018J000000019
13K000000014K000000015K000000016K000000017K000000018K000000019K000000020
```

### NodeMCU startup banner
```
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
	commit: e9a12991f7179ee8eb80b2fef27e2e7dafbc00fc
	release: 3.0-master_20200910 +22
	release DTS: 202009271933
	SSL: false
	build type: float
	LFS: 0xc000 bytes total capacity
	modules: bme280_math,enduser_setup,file,gpio,i2c,net,node,rtcmem,rtctime,sjson,sntp,struct,tmr,uart,wifi
 build 2020-09-27 19:43 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
```

### Hardware
No hardware dependent bug.

### Solution
The patch is very easy as we have to rewind the position only within current read chunk (it is incorrect when `j>0`).

```patch
diff --git a/app/modules/file.c b/app/modules/file.c
index e3157465e..4d6a6f98c 100644
--- a/app/modules/file.c
+++ b/app/modules/file.c
@@ -452,7 +452,7 @@ static int file_g_read( lua_State* L, int n, int16_t end_char, int fd )
     for (i = 0; i < nread; ++i) {
       luaL_addchar(&b, p[i]);
       if (p[i] == end_char) {
-        vfs_lseek(fd, -nread + j + i + 1, VFS_SEEK_CUR); //reposition after end char found
+        vfs_lseek(fd, -nread + i + 1, VFS_SEEK_CUR); //reposition after end char found
         nread = 0;   // force break on outer loop
         break;
       }
```

I am very surprised that such a bug could have survive so long so I wonder if the fix does not have any unintentional impacts (I don't expect any). I'm ready to submit a PR.

NicolSpies, 2020-09-28:
@vsky279 , I came across the same and equated it to `#define FILE_READ_CHUNK 1024` in `file.c` where a character is 8 bits resulting in 128 chars as per the documentation for `file.readline()`:

> Read the next line from the open file. Lines are defined as zero or more bytes ending with a EOL ('\n') byte. If the next line is longer than 1024, this function only returns the first 1024 bytes.

My fix was to read less data at a time in a loop and appending the returned data to build the read string, breaking the loop when `nil` is returned indicating the EOF.

vsky279, 2020-09-28:
@NicolSpies 1024 and I'm fine with that as it is well documented. However what happens between lengths 128 and 1024 is not ok for me. Your fix is a workaround that helps but is not a real fix.
In my previous application I came to this issue already but I was attributing to wrongly uploaded file. Than I was lucky with the file structure (line lengths) as the bug did not manifest in a disruptive way.

nwf, 2020-09-29:
Closed by #3302.

PR #3302 (2020-09-29): Fixing bug in `file.readline` when line length exceeds 128 chars
Fixes #3301.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Fixes bug in  `file.readline` (in fact in `file.read`) when line length exceeds 128 chars.
closed 2020-09-29 by nwf