nodemcu/nodemcu-firmware #3304: tls connection not working in latest release firmware

pfurbish, 2020-09-30:
I have multiple ESP8266 devkits running a firmware build from 2018-11-26.  They grab sensor data then post it to an Azure logic app using the net and tls modules. I updated one of the devkits with firmware built on 2020-09-18 using NodeMCU-PyFlasher-4.0-x64.  There was no change to the lua scripts.  The same certificate was stored.  Now the connection to Azure fails.

Below is my module to post data to an Azure logic app.  It builds some json with the sensor data then performs a post.  

-- Send to Azure
function sendMsg(t, h, status)
	local data = vars..'.h='..node.heap()
	local jsonMsg = '{\r\n"val1"\:\"'..t..'",\r\n"val2"\:\"'..h..'",\r\n"val3"\:\"'..status..'",\r\n"val4"\:\"'..locn..'\",\r\n"val5"\:\"'..data..'\"\r\n}'
	local mLen = string.len(jsonMsg)
	local request = httpHdr..mLen..'\r\n\r\n'..jsonMsg
	local sk = tls.createConnection()
	-- on connection
	sk:on('connection', function(conn, skdata)
		print('Send to Azure:',t, h, status)
		conn:send(request)
		conn, skdata = nil
	end)
	-- on receive
	sk:on('receive', function(conn, skdata)
		local respStat = string.sub(skdata, 10, 11)
		if (respStat ~= '20') then  -- 20x, Okay
			print('Send error:', skdata)
		end
		conn:close()
		sk, conn, skdata = nil
	end)
	print('Connect Azure:',host)
	sk:connect(443, host)   -- FAILS on this statement
end

With the latest firmware on V2 hardware, the connection fails with this panic...
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 27448, room 16 
tail 8
chksum 0x14
load 0x3ffe8000, len 2676, room 0 
tail 4
chksum 0x43
load 0x3ffe8a74, len 8, room 4 
tail 4
chksum 0x5e
csum 0x5e

With the latest firmware on V4 hardware, the connection fails with this panic...
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 27448, room 16 
tail 8
chksum 0x14
load 0x3ffe8000, len 2676, room 0 
tail 4
chksum 0x43
load 0x3ffe8a74, len 8, room 4 
tail 4
chksum 0x5e
csum 0x5e

### NodeMCU startup banner - latest release firmware
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: 64bbf006898109b936fcc09478cbae9d099885a8
	release: 3.0-master_20200910
	release DTS: 202009090323
	SSL: true
	build type: float
	LFS: 0x0 bytes total capacity
	modules: adc,dht,file,gpio,net,node,ow,tmr,uart,wifi,tls
 build 2020-09-30 17:18 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### NodeMCU startup banner - before upgrading firmware
NodeMCU custom build by frightanic.com
	branch: master
	commit: 4095c408e6a8cc9718cb06007b408d0aad15d9cd
	SSL: true
	modules: adc,dht,file,gpio,net,node,ow,tmr,uart,wifi,tls
 build created on 2018-11-26 01:16
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)

### Hardware
My first try was with an ESP8266 - ESP12E devkit V2, 4MB but I also tried a newer devkit, ESP-F V4, 4MB and had the same issue with the newer firmware.

I hope I provided what is needed to troubleshoot but let me know if more information is needed.

Thanks!

nwf, 2020-10-01:
Those are not panic messages; those are simply the bootloader doing its thing.  I'm afraid you'll have to tell us more.

marcelstoer, 2020-10-01:
Maybe run `print(node.heap())` before `sk:connect()` and compare the numbers across firmware releases?

pfurbish, 2020-10-02:
I pull my firmware builds from Marcel's service so I'm not familiar with building my own.  Is there an archive of builds that I could try?  If not, I might try the Docker firmware build option.  Otherwise, is there a way to log data from the tls handshake?  

Regarding the heap size, it's something I already have it in my send message module.  The heap size is pushed to Azure so I can watch for memory leaks.  The V2 devkit on the older, working firmware is a steady at 32,656.  It posts dht22 data once an hour.  For a V4 devkit with the new firmware, I added print(node.heap()) before sk:connect() since posting to Azure is not working and it reports 31.448 when running from ESPlorer.  Also, no E:M message, it just reboot after 60 second.

Thanks for the feedback.

marcelstoer, 2020-10-02:
>  Is there an archive of builds that I could try?

No

> is there a way to log data from the tls handshake?

Yes, you need to turn on the debug flag while building the firmware (translates to [`#define DEVELOP_VERSION`](https://github.com/nodemcu/nodemcu-firmware/blob/release/app/include/user_config.h#L221))

Also, please provide the [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) the issue template asks for. And...are TLS connections to other sites affected too or is this specific to Azure?

FrankX0, 2020-10-02:
And don't forget to turn on the TLS debug output:
https://nodemcu.readthedocs.io/en/dev/modules/tls/#tlssetdebug-function

pfurbish, 2020-10-11:
I was able to get Azure logic app posts to work on the current release firmware by setting tls verity = false, which is acceptable to me.  Thanks for everybody's help.

closed 2020-10-11 by pfurbish