nodemcu/nodemcu-firmware #3309: branch: 1.5.4.1-final - wifi does not connect

KomarRafal, 2020-10-10:
### Expected behavior
I have an esp8266 board with 512KB of flash.
When I use nodeMCU which is provided with ESP8266Flasher.exe (aka. INTERNAL://NODEMCU) WiFi works fine: (I use "live" uart console as well as lua script).
```Lua
> wifi.setmode(wifi.STATION)
> wifi.sta.config("MySSID","MyPassword")
> = wifi.sta.status()
1 CONNECTING
> = wifi.sta.status()
1 CONNECTING
> = wifi.sta.status()
5 GOT IP
> = wifi.sta.getip()
192.168.0.76    255.255.255.0   192.168.0.1
```
Pings from other computer work as well.

### Actual behavior
But for some reason, I have to use a new Lua API, so I want to have a newer version of nodeMCU. I built nodeMCU using web page https://nodemcu-build.com/, I left default options plus *one-wire* API. I selected branch: *1.5.4.1-final* (frozen, for 512KB flash). I tested float as well as the integer version. Unfortunately, WiFi does not work on both. I removed my wifi password and nothing changed.
```Lua
> wifi.setmode(wifi.STATION)
> wifi.sta.config {ssid="MySSID", pwd="MyPassword"}
> = wifi.sta.status()
1 CONNECTING
> = wifi.sta.status()
1 CONNECTING
> = wifi.sta.status()
4 CONNECT FAILED
```
On the other hand finding available ssids works fine:
```Lua
> = wifi.setmode(wifi.STATION
> = wifi.sta.getap(function(t) if t then print("\n\nVisible Access Points:\n") for k,v in pairs(t) do l = string.format("%-10s",k) print(l.."  "..v) end else print("Try again") end  end)

Visible Access Points:

MySSID   4,-43,ac:22:05:b1:2a:5b,8
OTHER_SSID  3,-47,8c:5b:f0:a1:5f:9f,6
```
### Test code
```Lua
wifi.setmode(wifi.STATION)
wifi.sta.config {ssid="MySSID", pwd="MyPassword"}
while (true) do
    status = wifi.sta.status()
    print(wifi.sta.status())
    if status ~= 1 then break end
end
```

### NodeMCU startup banner
```
NodeMCU custom build by frightanic.com
	branch: 1.5.4.1-final
	commit: 9ad37255717e02120f43f213a892a845dbe40615
	SSL: false
	modules: file,gpio,net,node,ow,tmr,uart,wifi
 build created on 2020-10-06 19:50
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
```

### Hardware
```
>esptool.py --port=COM6 flash_id
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 18:fe:34:99:9f:6b
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4013
Detected flash size: 512KB
```
```
> = node.chipid()
10067819
```

marcelstoer, 2020-10-11:
Dup of the support request at https://stackoverflow.com/q/64285480/131929 -> closing here

closed 2020-10-11 by marcelstoer

marcelstoer, 2020-10-11:
Addendum: I can't reproduce this issue (1.5.4.1-final, same modules)