nodemcu/nodemcu-firmware #3311: tmr:alarm callback has nil for timer object

kit-klein, 2020-10-15:
### Expected behavior
We have an auto timer that checks for content in two queues (`actuatorGet` and `sensorPut`) every 200 ms and then attempts to update remote endpoints as needed.  When the alarm callback occurs I expect the timer object argument to be a valid timer object.

### Actual behavior
We will rarely see the timer object in the alarm callback is `nil`.  When this happens we trigger a panic followed by a watchdog reset.

```
PANIC: unprotected error in call to Lua API (/lfs/rest_endpoint.lua:68: attempt to index local 't' (a nil value))

ESC[0;31mE (332569) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:ESC[0m

ESC[0;31mE (332569) task_wdt:  - IDLE0 (CPU 0)ESC[0m

ESC[0;31mE (332569) task_wdt: Tasks currently running:ESC[0m

ESC[0;31mE (332569) task_wdt: CPU 0: mainESC[0m

ESC[0;31mE (332569) task_wdt: CPU 1: IDLE1ESC[0m

ESC[0;31mE (332569) task_wdt: Aborting.ESC[0m

abort() was called at PC 0x400d3610 on core 0
```

### Test code
I'm unable to reproduce the problem but this is the relevant code segment.

```Lua
  local timeout = tmr.create()
  timeout:register(10000, tmr.ALARM_SEMI, node.restart)

  local sendTimer = tmr.create()
  sendTimer:alarm(200, tmr.ALARM_AUTO, function(t)

    -- gets state of actuators
    if actuatorGet[1] then
      t:stop()
      local actuator = actuatorGet[1]
      timeout:start()

      http.get(table.concat({ settings.endpoint, "/device/", dni, '?pin=', actuator.pin }),
        table.concat({ "Authorization: Bearer ", settings.token, "\r\nAccept: application/json\r\n" }),
        function(code, response)
          timeout:stop()
          local pin, state, json_response, status
          if response and code >= 200 and code < 300 then
            status, json_response = pcall(function() return sjson.decode(response) end)
            if status then
              pin = tonumber(json_response.pin)
              state = tonumber(json_response.state)
            end
          end
          printHttpResponse(code, {pin = pin, state = state})

          gpio.mode(actuator.pin, gpio.OUTPUT)
          if pin == actuator.pin and code >= 200 and code < 300 and state then
            gpio.write(actuator.pin, state)
          else
            state = actuator.trigger == gpio.LOW and gpio.HIGH or gpio.LOW
            gpio.write(actuator.pin, state)
          end
          print("Heap:", node.heap(), "Initialized actuator Pin:", actuator.pin, "Trigger:", actuator.trigger, "Initial state:", state)

          table.remove(actuatorGet, 1)
          t:start()
        end)

      -- update state of sensors when needed
    elseif sensorPut[1] then
      t:stop()
      local sensor = sensorPut[1]
      printHttpResponse(0, sensor)
      timeout:start()
      http.put(table.concat({ settings.endpoint, "/device/", dni }),
        table.concat({ "Authorization: Bearer ", settings.token, "\r\nAccept: application/json\r\nContent-Type: application/json\r\n" }),
        sjson.encode(sensor),
        function(code)
          timeout:stop()
          printHttpResponse(code, sensor)

          -- check for success and retry if necessary
          if code >= 200 and code < 300 then
            table.remove(sensorPut, 1)
          else
            -- retry up to 10 times then reboot as a failsafe
            local retry = sensor.retry or 0
            if retry == 10 then
              print("Heap:", node.heap(), "Retried 10 times and failed. Rebooting in 30 seconds.")
              for k, v in pairs(sensorPut) do sensorPut[k] = nil end -- remove all pending sensor updates
              tmr.create():alarm(30000, tmr.ALARM_SINGLE, function() node.restart() end) -- reboot in 30 sec
            else
              sensor.retry = retry + 1
              sensorPut[1] = sensor
            end
          end

          blinktimer:start()
          t:start()
        end)
    end

    collectgarbage()
  end)
```

### NodeMCU startup banner
NodeMCU ESP32 189fe96 - Konnected Pro build 20200822 powered by Lua 5.1.4 on ESP-IDF v3.3.1 on SDK IDF

I'm going to attempt to mitigate the problem by guarding for this condition but it seems indicative of a potential bigger problem under the covers. I wanted to check if anyone has encountered a similar problem or is aware of some limitation around timer usage that could cause this.

nwf, 2020-10-15:
I wonder if this is #2553 showing up in a different place.

kit-klein, 2021-01-05:
Some additional context based on some logs surrounding the incident.

The timer object in the callback only seems to be `nil` if the timer was stopped inside a previous callback.  Every data point I have reflects this but it certainly doesn't always occur.

After inspecting the tmr module I noticed the timer task doesn't check if `tmr->self_ref` is populated. 
https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/components/modules/tmr.c#L46

I also noticed `tmr->self_ref` is set to `LUA_NOREF` when we call stop on a timer.
https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/components/modules/tmr.c#L149

This made me start thinking we could have a race condition where timer callbacks are queued up faster than they can be processed.  Times of intense cpu usage could cause this (which makes sense based on when the issue occurs in our observations).  I wrote some test code to attempt this,

```
local sendTimer = tmr.create()
local blockTimer = tmr.create()

sendTimer:register(20, tmr.ALARM_AUTO, function(t)

  -- guard against nil t
  if t == nil then
    print("timer object is nil")
    return
  end

  t:stop()
  print("stopped timer")

  tmr.create():alarm(1000, tmr.ALARM_SINGLE, function()
    print("starting timer")
    t:start()
  end)
end)

blockTimer:register(2000, tmr.ALARM_AUTO, function(t)

  sendTimer:start()
  for i=1,100 do
    print("wasting time")
  end
  -- consume a lot of cpu cycles
end)

blockTimer:start()
```

This creates the issue - resulting in recurring output like
```
wasting time
wasting time
wasting time
...
wasting time
stopped timer
timer object is nil
timer object is nil
timer object is nil
timer object is nil
starting timer
stopped timer
starting timer
stopped timer
```

Based on this I think we should check `tmr->self_ref` [similar to how we check ](https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/components/modules/tmr.c#L46)`tmr->cb_ref` or somehow drain the task queue for tasks corresponding to a tmr when stopped.

PR #3366 (2021-01-05): Ensure tmr ref is valid before calling tmr callback
Fixes #3311 

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Rapid auto timers can resul [... cut]
closed 2021-01-05 by marcelstoer