nodemcu/nodemcu-firmware #3312: boot loop

Na2l, 2020-10-21:
Let me start by saying, I am not t sure what I am doing wrong :)
here is the problem:
after erasing and flashing the latest firmware from https://nodemcu-build.com/ (release branch), the node keeps rebooting, to see the boot message I had to change the baud-rate to 74480.   I tried several flash sizes all with the same result/ here is the boot message:

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 27444, room 16 
tail 4
chksum 0xc9
load 0x3ffe8000, len 2568, room 4 
tail 4
chksum 0x61
load 0x3ffe8a08, len 8, room 4 
tail 4
chksum 0x14
csum 0x14
system SPI FI size:2, Flash size: 1048576
boot not set
ota1 not set
ota2 not set

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 27444, room 16 
tail 4
chksum 0xc9
load 0x3ffe8000, len 2568, room 4 
tail 4
chksum 0x61
load 0x3ffe8a08, len 8, room 4 
tail 4
chksum 0x14
csum 0x14
system SPI FI size:2, Flash size: 1048576
boot not set
ota1 not set
ota2 not set

I hope someone can help me resolving the issue.

marcelstoer, 2020-10-21:
-> #1010

closed 2020-10-21 by marcelstoer

TerryE, 2020-10-26:
@Na2l only select the modules you need and also try doing a full erase of your module before flashing the firmware.