nodemcu/nodemcu-firmware #3318: LUA53 goes very low in heap with compared to LUA51

chathurangawijetunge, 2020-10-26:
When running lua script in LUA51 it performs very well with heap about 37000 
But in LUA53 it goes down lower as 18000 and in some situations LUA53 runs out of heap and panics.
(All codes in LFS)

galjonsfigur, 2020-10-26:
Have you tried using `collectgarbage` function in your code? The way GC runs changed in Lua 5.3 builds and can give an illusion of lower heap when using Lua 5.3 - see #3175.

chathurangawijetunge, 2020-10-26:
Yes I tried 'collectgarbage' but no luck.
Runs out of heap and reboots...

TerryE, 2020-10-26:
@chathurangawijetunge, you need to distill this down to a repeatable test case.

marcelstoer, 2020-10-27:
As this doesn't adhere to our issue template it's got no [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem -> closing

closed 2020-10-27 by marcelstoer