nodemcu/nodemcu-firmware #3319: PWM Driver that can do reasonable resolution at 25kHz

pjsg, 2020-10-27:
### Missing feature
I'd like to have a PWM driver that can manage a 25kHz repetition rate with a reasonable resolution -- e.g. at least 100 different levels. It appears that neither of the existing `pwm` or `pwm2` can manage that.

### Justification
This will allow me to drive a 4-wire FAN.

### Workarounds
I can use the `pwm2` module with the cpu at 160MHz but it doesn't really provide enough resolution.

### Possible approach

I'm thinking of trying to use the SPI hardware to output (on a single pin) an arbitrary waveform at up to the top speed of the SPI. The DMA can be used to cycle a single buffer of bits out of the port. It isn't good for long period waveforms, but fast repetition should be the sweet spot with maybe up to 8k bits before repeating.

fikin, 2020-10-27:
+1
sounds good
@pjsg : do you have an idea of the lua api already?

KT819GM, 2020-10-27:
For the exact same reason, when I found that I can't use pwm2 at ~30 kHz with more than 10 steps I went with `sigma-delta` module. Maybe we could use that hardware component (if I understood correctly) to use for plain PWM only?

pjsg, 2020-10-27:
The problem with the sigma-delta component is that the frequency varies based on the duty cycle.

athompson673, 2021-01-18:
can I suggest a hardware solution to a hardware problem (there is no hardware pwm on esp8266)?

https://www.mouser.com/ProductDetail/Maxim-Integrated/MAX31790ATI%2b/?qs=4Rc5iGDDRGgYTNUV2AhNCw%3D%3D

closed 2022-04-30 by stale[bot]