nodemcu/nodemcu-firmware #3326: tmr.softwd(-1) returns error "invalid time"

Nerdvana2, 2020-11-09:
### Expected behavior
    disable watchdog, as in documentation

### Actual behavior
   returns error "invalid time"

### Test code
> tmr.softwd(-1)
Lua error: 	stdin:1: bad argument #2 to 'softwd' (invalid time)
stack traceback:
	[C]: in function 'softwd'
	stdin:1: in main chunk
	[C]: ?
	[C]: ?

### NodeMCU startup banner
NodeMCU 3.0.0.0 
	branch: release
	commit: 4f6792773f93f36a7255cfd28dca7aa6c4aa9552
	release: 3.0-release_20201107
	release DTS: 202011071523
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: adc,bit,dht,encoder,file,gpio,gpio_pulse,hx711,i2c,mqtt,net,node,ow,pipe,rtcmem,rtctime,sjson,sntp,spi,tmr,uart,wifi
 build 2020-11-08 22:39 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware

####### I think this was introduced as part of:
commit d72ea91ed01518ef0779d7aa998c5c3a9c269d71
Author: Lukáš Voborský <vsky279@users.noreply.github.com>
Date:   Sat Aug 29 21:43:50 2020 +0200

Cheers
Sean

HHHartmann, 2020-11-09:
From the code it also seems that setting it to zero would disable it. But that's not working atm. either.

vsky279, 2020-11-09:
No, no, it was not PR #3111 as there was only formatting change.
PR #3163 implemented the check that the parameter is positive:
```patch
commit 1f2e5bba4ab38c499d41c57a1e74e5c405fd3868
Author: Terry Ellison <Terry@ellisons.org.uk>
Date:   Tue Jun 16 08:19:55 2020 +0100

    Implement panic call handling for all modules (#3163)

diff --git a/app/modules/tmr.c b/app/modules/tmr.c
--- a/app/modules/tmr.c
+++ b/app/modules/tmr.c
@@ -358,3 +265,5 @@
 static int tmr_softwd( lua_State* L ){
-       soft_watchdog = luaL_checkinteger(L, 1);
+       int t = luaL_checkinteger(L, 1);
+       luaL_argcheck(L, t>0 , 2, "invalid time");
+  soft_watchdog = t;
        return 0;
```

The patch we need is:
```patch
-	luaL_argcheck(L, t>0 , 2, "invalid time");
+	luaL_argcheck(L, t>0 || t==-1, 2, "invalid time");
```

Should I post a PR?

HHHartmann, 2020-11-09:
@vsky279 See https://github.com/nodemcu/nodemcu-firmware/pull/3327

HHHartmann, 2020-11-14:
Has been fixed in current dev branch

PR #3327 (2020-11-13): Allow turning off softwd again as documented
Fixes #3326.
I also added a Testcase for NTest to not have this happen again

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Disabling softwd now is possible again
closed 2020-11-14 by marcelstoer