nodemcu/nodemcu-firmware #3328: dht module returns checksum errors/weird values

wilhelmy, 2020-11-12:
I recently updated, now I'm getting checksum errors with the dht module. Is this normal?

On the old firmware, the readings are way off, more than the ±2°C tolerance which are specified for the sensor iirc.

In case it matters, this is a WeMos D1 mini.

### Expected behavior

NodeMCU 3.0.0.0 
	branch: master
	commit: 3d917850180f67adc7f2c6b5d00f27c152e7194c
	release: 3.0-master_20190907 +2
	release DTS: 202002231746
	SSL: false
	build type: float
	LFS: 0x0
	modules: adc,bit,dht,file,gpio,i2c,mqtt,net,node,ow,spi,tmr,uart,wifi
 build 2020-03-17 02:44 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
lua: cannot open init.lua
   
> 
> for i=5,7 do print(dht.read11(i)) end
0	24	34	0	0
0	25	33	0	0
0	23	38	0	0

### Actual behavior
NodeMCU 3.0.0.0 
	branch: master
	commit: 4f6792773f93f36a7255cfd28dca7aa6c4aa9552
	release: 3.0-release_20201107
	release DTS: 202011071523
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: adc,bit,dht,file,gpio,i2c,mqtt,net,node,ow,spi,tmr,uart,wifi
 build 2020-11-12 01:20 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
cannot open init.lua: 
> for i=5,7 do print(dht.read11(i)) end
0	614.4	870.4	400	400
-1	614.4	896	400	0
-1	742.4	1203.2	400	200
> for i=5,7 do print(dht.read11(i)) end
-1	793.6	1203.2	600	200
-1	793.6	1203.2	600	200
-1	793.6	1203.2	600	200
> for i=5,7 do print(dht.read11(i)) end
-2	-999	-999	0	0
-2	-999	-999	0	0
-2	-999	-999	0	0
> for i=5,7 do print(dht.read11(i)) end
-1	793.6	1203.2	600	200
-1	793.6	1203.2	600	200
-1	793.6	1203.2	600	200
> for i=5,7 do print(dht.read11(i)) end
-2	-999	-999	0	0
-2	-999	-999	0	0
-2	-999	-999	0	0
> for i=5,7 do print(dht.read11(i)) end
-1	793.6	1203.2	600	200
-1	793.6	1203.2	600	200
-1	793.6	1203.2	600	200

marcelstoer, 2020-11-14:
Related to @emdioh's #3210?

ezbella, 2020-11-16:
I'm facing the same issue with the latest version of dht module. More details here: [https://github.com/nodemcu/nodemcu-firmware/pull/3210#issuecomment-727264396](url)
Which is the recommended way of building the firmware but using the previous working version of dht module?
Thanks in advance for your wisdom! =)

vsky279, 2020-11-21:
I confirm the issue is due to #3210. I'll try to fix it in coming days.

wilhelmy, 2020-12-14:
Would it be possible to merge this into master?

marcelstoer, 2020-12-14:
#3334 will soon be merged - to `dev`. Once we release it, planned at the end of this month, it'll be available on `release` and `master`. See https://github.com/nodemcu/nodemcu-firmware/tree/dev#releases for details.

jmiskovic, 2020-12-16:
I also ran against this issue with DHT22 and #3334 solved it! Thanks for the fix.

PR #3334 (2020-12-16): DHT module bug fix
Fixes #3328.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

After #3210 DHT module started to return checksum errors aft [... cut]
closed 2020-12-16 by marcelstoer