nodemcu/nodemcu-firmware #3338: http client crashes on answer with no body

quackmore, 2020-11-26:
### Expected behavior
http to accept answers with empty body
e.g. 'HTTP/1.1 200 OK\r\n\'
like for instance curl do

### Actual behavior
http crashes executing the callback function after the response has been received when the answer has no body
e.g. when receiving 'HTTP/1.1 200 OK\r\n'
possibly because of how 'body_size' is calculated on line 415 of httpclient.c

workaround: 
pass nil instead of a callback
e.g. http.post(cmd, nil, nil, **nil**)
or place extra \r\n at the end of the answer
e.g. 'HTTP/1.1 200 OK\r\n\r\n\r\n'

### Test code
full code and setup here: https://www.esp8266.com/viewtopic.php?f=11&t=22152

testing the server (returning 'HTTP/1.1 200 OK\r\n') with curl shows no error
curl -v --request POST --location 'http://192.168.10.1/led/turn/off'
*   Trying 192.168.10.1...
* TCP_NODELAY set
* Connected to 192.168.10.1 (192.168.10.1) port 80 (#0)
> POST /led/turn/off HTTP/1.1
> Host: 192.168.10.1
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 200 OK
* Curl_http_done: called premature == 0
* Connection #0 to host 192.168.10.1 left intact

### NodeMCU startup banner
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
branch: release
commit: 4f6792773f93f36a7255cfd28dca7aa6c4aa9552
release: 3.0-release_20201107
release DTS: 202011071523
SSL: false
build type: integer
LFS: 0x0 bytes total capacity
modules: file,gpio,http,net,node,tmr,uart,wifi
build 2020-11-25 12:40 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
WEMOS D1 and NODEMCU ESP8266

### Extra
Love your job!

HHHartmann, 2020-11-27:
Please give a minimal and complete example of at least the client and the exact error message. 

In the code on the linked page there isn't even a post used.

quackmore, 2020-11-27:
SETUP
two esp8266 devices, one client and one server
client (wifi station) connects to server (wifi AP)
client waits 10s then send a POST (http.post) to server

################## CLIENT
`-- init.lua

print("Running")

-- wifi_station.lua
SSID="NODEMCU-1"
PASSWORD="node123456"

connected_to_AP = false

-- Define WiFi station event callbacks
wifi_connect_event = function(T)
  print("Connection to AP("..T.SSID..") established!")
  print("Waiting for IP address...")
  if disconnect_ct ~= nil then 
    disconnect_ct = nil 
  end
end

wifi_got_ip_event = function(T)
  print("Wifi connection is ready! IP address is: "..T.IP)
  connected_to_AP = true
end

wifi_disconnect_event = function(T)
  if T.reason == wifi.eventmon.reason.ASSOC_LEAVE then
    --the station has disassociated from a previously connected AP
    return
  end
  connected_to_AP = false
  print("\nWiFi connection to AP("..T.SSID..") has failed!")
  --There are many possible disconnect reasons, the following iterates through
  --the list and returns the string corresponding to the disconnect reason.
  for key,val in pairs(wifi.eventmon.reason) do
    if val == T.reason then
      print("Disconnect reason: "..val.."("..key..")")
      break
    end
  end

  if disconnect_ct == nil then
    disconnect_ct = 1
  else
    disconnect_ct = disconnect_ct + 1
  end
  print("Retrying connection...(attempt "..(disconnect_ct+1)..")")
end

-- Register WiFi Station event callbacks
wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, wifi_connect_event)
wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, wifi_got_ip_event)
wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, wifi_disconnect_event)

print("Connecting to WiFi access point...")
wifi.setmode(wifi.STATION)
wifi.sta.config({ssid=SSID, pwd=PASSWORD})
-- wifi.sta.connect() not necessary because config() uses auto-connect=true by default

-- application.lua
cmd = "http://192.168.10.1/led/turn/on"
-- cmd = "http://192.168.10.1/led/turn/off"

periodic_task = function()
    print("sending POST message to the server ...")
[... 144 lines cut ...]
client trace:
`Wifi connection is ready! IP address is: 192.168.10.11
sending POST message to the server ...
HTTP client: Body shouldn't be NULL
�
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 27444, room 16 
tail 4
...
`
same with curl
`curl -v --request POST --location 'http://192.168.10.1/led/turn/on'
*   Trying 192.168.10.1...
* TCP_NODELAY set
* Connected to 192.168.10.1 (192.168.10.1) port 80 (#0)
> POST /led/turn/on HTTP/1.1
> Host: 192.168.10.1
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 200 OK
* Curl_http_done: called premature == 0
* Connection #0 to host 192.168.10.1 left intact
`

HHHartmann, 2020-11-27:
Ah, now I get it.
The readon is because the reply you send is not confotm to the HTTP protocol. See here: https://www.w3.org/Protocols/rfc2616/rfc2616-sec4.html

It states that a message Head has to be terminated by an empty line.

So we could fix this by assuming that the (unterminated) header at the end of the package is the end of the header. As it is in the disconnect  handler this seems to be safe.

@nwf do you think this would be safe?

nwf, 2020-11-28:
As all processing happens in the disconnect handler, I suppose that would be fine, but *yikes*, our `http` client is doing all of its work in **the disconnect handler**.  WTAF.  That probably explains quite a lot about its historical bug record.

closed 2022-04-30 by stale[bot]