nodemcu/nodemcu-firmware #3343: LUA module bme280 not work.

Plazmin, 2020-12-06:
### Expected behavior
Operation of the bme280 sensor using the "LUA module bme280". If I use "C module bme280" the sensor is working fine. However, according to the information "Warning, deprecated API! bme280. It will be removed soon. Use bme280math and bme280 Lua module instead" this version of the module is deprecated  and will be retired soon. That's why I want to use the new LUA module in my application. Of course, in the case of "C module" I use firmware that has the bme280 module and the bme280_math module built in. In the case of "LUA module bme280", only the bme280_math module is compiled in the firmware and the device has the bme280.lua file downloaded from https://github.com/nodemcu/nodemcu-firmware/tree/dev/lua_modules/bme280/bme280.lua. Both firmware are built using the website https://nodemcu-build.com/

### Actual behavior
The program reports errors and the NodeMcu device resets.

> dofile("test280.lua")
> test280.lua:8: attempt to index global 's' (a nil value)
stack traceback:
	test280.lua:8: in function <test280.lua:7>
test280.lua:8: attempt to index global 's' (a nil value)
stack traceback:
	test280.lua:8: in function <test280.lua:7>
test280.lua:8: attempt to index global 's' (a nil value)
stack traceback:
	test280.lua:8: in function <test280.lua:7>
test280.lua:8: attempt to index global 's' (a nil value)
stack traceback:
	test280.lua:8: in function <test280.lua:7>
test280.lua:8: attempt to index global 's' (a nil value)
stack traceback:
	test280.lua:8: in function <test280.lua:7>

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 27444, room 16 
tail 4
chksum 0x9e
load 0x3ffe8000, len 2576, room 4 
tail 12
chksum 0x2a
ho 0 tail 12 room 4
load 0x3ffe8a10, len 8, room 12 
tail 8
chksum 0x75
csum 0x75
�2r���o���

### Test code

alt=36 -- altitude of the measurement place

sda, scl = 2, 1
i2c.setup(0, sda, scl, i2c.SLOW)
s = require('bme280').setup(0)

tmr.create():alarm(500, tmr.ALARM_AUTO, function()
    local T, P, H, QNH = s:read(alt)
    local D = s:dewpoint(H, T)
    print(("T=%.2f, QFE=%.3f, QNH=%.3f, humidity=%.3f, dewpoint=%.2f"):format(T, P, QNH, H, D))
end)
```
The above code is an example taken from the LUA module bme280 NodeMCU documentation. In my example it is called "test.280.lua"

### NodeMCU startup banner
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: 4f6792773f93f36a7255cfd28dca7aa6c4aa9552
	release: 3.0-release_20201107
	release DTS: 202011071523
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: bme280_math,file,gpio,i2c,net,node,tmr,uart,wifi
 build 2020-12-05 11:51 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
Amica NodeMcu v.2 and CJMCU BME280 sensor.

vsky279, 2020-12-07:
Hi. I'm sorry for this issue. It's wise to use the new `bme280_math` module together with the `bme280` Lua module.

It is useless to have both `bme280` module and the `bme280_math` module together. The `bme280` module includes all the math that is in the `bme280_math` module. If you use `bme280` Lua module (to be used with `bme280_math`) then you can have conflicts between the Lua and the C module. But apparently this is not the issue here.

When `setup` returns `nil` it means that it could not initialize the BME280 sensor. So before calling s' methods you should test it for `nil`. This is omitted in the simple example. Is the sensor connected to GPIO5 (sda) and GPIO4 (scl)?

You can also try to enable debug in `bme280` module. You would need to uncomment this part:
```Lua
local print, node_heap = print, node.heap
local function debug (fmt, ...) -- upval: cnt (, print, node_heap)
  if not bme280.debug then return end
  if (...) then fmt = fmt:format(...) end
  print("[bme280]", node_heap(), fmt)
end
```
and then uncomment every occurrence of `debug`.

Your test code will then need to look like something like this:
```Lua
alt=36 -- altitude of the measurement place

sda, scl = 2, 1
i2c.setup(0, sda, scl, i2c.SLOW)
sr = require('bme280')
sr.debug = true
s = sr.setup(0)

tmr.create():alarm(500, tmr.ALARM_AUTO, function()
    local T, P, H, QNH = s:read(alt)
    local D = s:dewpoint(H, T)
    print(("T=%.2f, QFE=%.3f, QNH=%.3f, humidity=%.3f, dewpoint=%.2f"):format(T, P, QNH, H, D))
end)
```

Plazmin, 2020-12-09:
The documentation for the BME280 LUA module says: "Note: the module works only with the bme280_math module.".  That's why I compiled the module bme280_math into the firmware. If, as you write, it is not needed and it can cause problems, the documentation should be corrected.

>  Is the sensor connected to GPIO5 (sda) and GPIO4 (scl)?

Yes. The same connection worked without problems with the C modules. From the point of view of hardware and connections, everything is rather correct.

I will build the firmware without the bme_math module and try if it works properly. I will also modify the lua code, according to your directions, do some tests and let you know the result.

Plazmin, 2020-12-09:
**I loaded the firmware:**

NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: 4f6792773f93f36a7255cfd28dca7aa6c4aa9552
	release: 3.0-release_20201107
	release DTS: 202011071523
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: file,gpio,i2c,mdns,mqtt,net,node,tmr,uart,wifi
 build 2020-12-09 08:18 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

**I uncomment this part in bme280.lua file:**

local print, node_heap = print, node.heap
local function debug (fmt, ...) -- upval: cnt (, print, node_heap)
  if not bme280.debug then return end
  if (...) then fmt = fmt:format(...) end
  print("[bme280]", node_heap(), fmt)
end

**The bme280.lua file was uploaded to the nodemcu device. 
And if I run code (test280-2.lua):**

alt=36 -- altitude of the measurement place

sda, scl = 2, 1
i2c.setup(0, sda, scl, i2c.SLOW)
sr = require('bme280')
sr.debug = true
s = sr.setup(0)

tmr.create():alarm(500, tmr.ALARM_AUTO, function()
    local T, P, H, QNH = s:read(alt)
    local D = s:dewpoint(H, T)
    print(("T=%.2f, QFE=%.3f, QNH=%.3f, humidity=%.3f, dewpoint=%.2f"):format(T, P, QNH, H, D))
end)

**I get this message:**
> dofile('test280-2.lua')
Lua error: 	bme280.lua:22: attempt to index global 'bme280_math' (a nil value)
stack traceback:
	bme280.lua:22: in main chunk
	[C]: in function 'require'
	test280-2.lua:5: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk
	[C]: ?
	[C]: ?
> 
I don't understand whether the bme280_math C module has to be compiled into the firmware or not. However, in both cases, the code does not work :-(

vsky279, 2020-12-09:
Sorry if I was not clear. For the new bme280 Lua module you need to include `bme280_math`.

You can see it from this error message:
```
Lua error: bme280.lua:22: attempt to index global 'bme280_math' (a nil value)
```

Plazmin, 2020-12-09:
**Firmware:**                                                                                                                                                                                         
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: 4f6792773f93f36a7255cfd28dca7aa6c4aa9552
	release: 3.0-release_20201107
	release DTS: 202011071523
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: bme280_math,file,gpio,i2c,net,node,tmr,uart,wifi
 build 2020-12-05 11:51 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

**It still does not work:**
> dofile("test280-2.lua")
> test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 27444, room 16 
tail 4
chksum 0x9e
load 0x3ffe8000, len 2576, room 4 
tail 12
chksum 0x2a
ho 0 tail 12 room 4
load 0x3ffe8a10, len 8, room 12 
tail 8
chksum 0x75
csum 0x75
�2r���N��$���

vsky279, 2020-12-10:
I don't see any debug output if the debug output was enabled. Could you try again with enabled debug output?

Plazmin, 2020-12-10:
**bme280.lua is uploaded to nodemcu and it looks like this:**
```Lua
--[[
 BME280 Lua module
 Requires i2c and bme280_math module
 Written by Lukas Voborsky, @voborsky
]]

local bme280 = {}

-- bme280.setup()
-- bme280:setup()
-- bme280:read()
-- bme280:altitude()
-- bme280:dewpoint()
-- bme280:qfe2qnh()
-- bme280:startreadout()

local type, assert = type, assert
local table_concat, math_floor = table.concat, math.floor
local i2c_start, i2c_stop, i2c_address, i2c_read, i2c_write, i2c_TRANSMITTER, i2c_RECEIVER =
  i2c.start, i2c.stop, i2c.address, i2c.read, i2c.write, i2c.TRANSMITTER, i2c.RECEIVER
local bme280_math_setup, bme280_math_read, bme280_math_qfe2qnh, bme280_math_altitude, bme280_math_dewpoint =
  bme280_math.setup, bme280_math.read, bme280_math.qfe2qnh, bme280_math.altitude, bme280_math.dewpoint
local tmr_create, tmr_ALARM_SINGLE = tmr.create, tmr.ALARM_SINGLE

local BME280_I2C_ADDRESS1 = 0x76
local BME280_I2C_ADDRESS2 = 0x77

local BME280_REGISTER_CONTROL = 0xF4
local BME280_REGISTER_CONTROL_HUM = 0xF2
local BME280_REGISTER_CONFIG= 0xF5
local BME280_REGISTER_CHIPID = 0xD0

local BME280_REGISTER_DIG_T = 0x88  -- 0x88-0x8D ( 6)
local BME280_REGISTER_DIG_P = 0x8E  -- 0x8E-0x9F (18)
local BME280_REGISTER_DIG_H1 = 0xA1  -- 0xA1      ( 1)
local BME280_REGISTER_DIG_H2 = 0xE1  -- 0xE1-0xE7 ( 7)
local BME280_REGISTER_PRESS = 0xF7	 -- 0xF7-0xF9

-- local BME280_FORCED_MODE = 0x01

-- maximum measurement time in ms for maximum oversampling for all measures
-- 113 > 1.25 + 2.3*16 + 2.3*16 + 0.575 + 2.3*16 + 0.575 ms
local BME280_SAMPLING_DELAY =113

-- Local functions
local read_reg
local write_reg
local bme280_setup
local bme280_read
local bme280_startreadout

-- -- Note that the space between debug and the arglist is there for a reason
-- -- so that a simple global edit "   debug(" -> "-- debug(" or v.v. to
-- -- toggle debug compiled into the module.
local print, node_heap = print, node.heap
local function debug (fmt, ...) -- upval: cnt (, print, node_heap)
  if not bme280.debug then return end
  if (...) then fmt = fmt:format(...) end
  print("[bme280]", node_heap(), fmt)
end

--------------------------- Set up the bme280 object ----------------------------
--       bme280 has method setup to create the sensor object and setup the sensor
--       object created by bme280.setup() has methods: read, qfe2qnh, altitude, dewpoint
---------------------------------------------------------------------------------

function bme280.setup(id, addr, temp_oss, press_oss, humi_oss, power_mode, inactive_duration, IIR_filter, full_init)
  return bme280_setup(nil, id,
[... 149 lines cut ...]
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 27444, room 16 
tail 4
chksum 0x9e
load 0x3ffe8000, len 2576, room 4 
tail 12
chksum 0x2a
ho 0 tail 12 room 4
load 0x3ffe8a10, len 8, room 12 
tail 8
chksum 0x75
csum 0x75
�r�ےN��l���
```

vsky279, 2020-12-10:
Yes, you still need to uncomment the lines `-- debug(...` and add `sr.debug = true` line into your test file.

Plazmin, 2020-12-10:
The line `sr.debug = true` was added all the time, but the `-- debug(...`  lines were commented out.
**The result:**
```
> dofile("test280-2.lua")
[bme280]	33432	0 76 208
[bme280]	33424	No ACK on address: 76
> test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>
test280-2.lua:10: attempt to index global 's' (a nil value)
stack traceback:
	test280-2.lua:10: in function <test280-2.lua:9>

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 27444, room 16 
tail 4
chksum 0x9e
load 0x3ffe8000, len 2576, room 4 
tail 12
chksum 0x2a
ho 0 tail 12 room 4
load 0x3ffe8a10, len 8, room 12 
tail 8
chksum 0x75
csum 0x75
�2r���o��l���
```
The error occurs only when I have the sensor address set to 0x77, after changing to 0x76 the sensor works fine!
Hope we're getting closer to solving the problem :)

vsky279, 2020-12-10:
Ok, so when your sensor is sitting on the addres 0x77 you need to initialize with:
```
s = sr.setup(0, 2)
```

I quote part of the documentation:

#### Syntax

`bme280.setup(id, [address, temp_oss, press_oss, humi_oss, power_mode, inactive_duration, IIR_filter])`

#### Parameters
- `id` - I2C bus number
- **(optional)`address` - BME280 sensor address. `1` for `BME280_I2C_ADDRESS1 = 0x76`, `2` for  `BME280_I2C_ADDRESS2 = 0x77`. Default sensor address is `BME280_I2C_ADDRESS1`.**

Plazmin, 2020-12-11:
Wow, you are right i got lost with it. Now it works very well. It seemed to me that I was doing everything right, I checked several times. I need to read the documentation more carefully. Thank you for your help and your time.
Thanks!

closed 2020-12-11 by Plazmin