nodemcu/nodemcu-firmware #3355: Does not complie: CONFIG_CONSOLE_UART_NUM deprecated (fix included)

chribo, 2020-12-25:
### Expected behavior
make flash compiles
### Actual behavior
compilation stops with

```
In file included from /home/pi/esp/nodemcu-firmware-esp32/components/driver_console/include/driver/console.h:34,
                 from /home/pi/esp/nodemcu-firmware-esp32/components/base_nodemcu/uart.c:6:
/home/pi/esp/esp-idf-v4.1/components/esp32/include/rom/uart.h:1:2: warning: #warning rom/uart.h is deprecated, please use esp32/rom/uart.h instead [-Wcpp]
 #warning rom/uart.h is deprecated, please use esp32/rom/uart.h instead
  ^~~~~~~
In file included from /home/pi/esp/nodemcu-firmware-esp32/components/base_nodemcu/uart.c:6:
/home/pi/esp/nodemcu-firmware-esp32/components/base_nodemcu/uart.c: In function 'uart_on':
/home/pi/esp/nodemcu-firmware-esp32/components/driver_console/include/driver/console.h:39:22: error: 'CONFIG_CONSOLE_UART_NUM' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
 #define CONSOLE_UART CONFIG_CONSOLE_UART_NUM
                      ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/esp/nodemcu-firmware-esp32/components/base_nodemcu/uart.c:50:17: note: in expansion of macro 'CONSOLE_UART'
   unsigned id = CONSOLE_UART;
                 ^~~~~~~~~~~~
/home/pi/esp/nodemcu-firmware-esp32/components/driver_console/include/driver/console.h:39:22: note: each undeclared identifier is reported only once for each function it appears in
 #define CONSOLE_UART CONFIG_CONSOLE_UART_NUM
                      ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/esp/nodemcu-firmware-esp32/components/base_nodemcu/uart.c:50:17: note: in expansion of macro 'CONSOLE_UART'
   unsigned id = CONSOLE_UART;
                 ^~~~~~~~~~~~
/home/pi/esp/nodemcu-firmware-esp32/components/base_nodemcu/uart.c: In function 'uart_setup':
/home/pi/esp/nodemcu-firmware-esp32/components/driver_console/include/driver/console.h:39:22: error: 'CONFIG_CONSOLE_UART_NUM' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
 #define CONSOLE_UART CONFIG_CONSOLE_UART_NUM
                      ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/esp/nodemcu-firmware-esp32/components/base_nodemcu/uart.c:145:12: note: in expansion of macro 'CONSOLE_UART'
   if(id == CONSOLE_UART && lua_isnumber(L,6)){
            ^~~~~~~~~~~~
make[1]: *** [/home/pi/esp/esp-idf-v4.1/make/component_wrapper.mk:292: uart.o] Error 1
make: *** [/home/pi/esp/esp-idf-v4.1/make/project.mk:635: component-base_nodemcu-build] Error 2
```
### Test code

Not a lua code problem 
in components/driver_console/include/driver/console.h
```C
#include "rom/uart.h"
#include "task/task.h"
#include <stdint.h>
#include <stdbool.h>

#define CONSOLE_UART CONFIG_CONSOLE_UART_NUM
```

### NodeMCU startup banner
You MUST include the firmware startup banner to describe the version you are using. We reserve the right to immediately close any bug that doesn't.

None before fix (below), since it didn't compile.

### Hardware
Describe which ESP8266/ESP32 device you use and document any special hardware setup
required to reproduce the problem.

joy-it: SBC-NodeMCU-ESP32
```
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

### Fix
```
s/CONSOLE_UART CONFIG_CONSOLE_UART_NUM/CONSOLE_UART CONFIG_ESP_CONSOLE_UART_NUM/
```
```
diff --git a/components/driver_console/include/driver/console.h b/components/driver_console/include/driver/console.h
index 592c92ab..c2cb7fb9 100644
--- a/components/driver_console/include/driver/console.h
+++ b/components/driver_console/include/driver/console.h
@@ -36,7 +36,7 @@
 #include <stdint.h>
 #include <stdbool.h>
 
-#define CONSOLE_UART CONFIG_CONSOLE_UART_NUM
+#define CONSOLE_UART CONFIG_ESP_CONSOLE_UART_NUM
 
 typedef enum
 {
```

marcelstoer, 2020-12-27:
> /home/pi/esp/esp-idf-v4.1

Hhmm, did you manually upgrade the ESP-IDF to 4.1? We use 3.3.2 LTS (but maybe should bump to 3.3.4).

chribo, 2020-12-27:
I used the standard ESP-IDF (which is 4.1) on a raspberrypi - bad idea. 
You can close the ticket, on x64_86  it works straight through.

closed 2020-12-27 by marcelstoer