nodemcu/nodemcu-firmware #3375: Problem with u8g2.ssd1327_i2c_ea_w128128 (and all other ssd1327_i2c driver)

webstatic, 2021-01-18:
### Expected behavior
Do not reboot itself 
### Actual behavior
It causes the device to reboot.

Many times it reboots after print('init ssd1327_i2c_ea_w128128').

[image]
[image]

Sometimes it reboots after print('sendBuffer'), and show "hello" on display.
About one to five chance that it will display "hello" and reboot.

But It never pass disp:sendBuffer() line.

### Test code
    local scl = 1
    local sda = 2

    local sla = 0x3C
    i2c.setup(0, sda, scl, i2c.SLOW)
    print('init ssd1327_i2c_ea_w128128')
    disp = u8g2.ssd1327_i2c_ea_w128128(0, sla)
    print('setFontRefHeightExtendedText')
    disp:setFontRefHeightExtendedText()
    print('setDrawColor')
    disp:setDrawColor(1)
    print('setFontPosTop')
    disp:setFontPosTop()
    print('setFontDirection')
    disp:setFontDirection(0)

    print('clearBuffer')
    disp:clearBuffer()
    print('setFont')
    disp:setFont(u8g2.font_6x10_tf)
    print('drawStr')
    disp:drawStr(0, 0, "hello");
    print('sendBuffer')
    disp:sendBuffer()

### NodeMCU startup banner
branch: dev
	commit: 6316b33296e180de4dbcfedecb8fe5c18383d11d
	release: 
	release DTS: 202101162126
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: bit,file,gpio,i2c,net,node,spi,tmr,u8g2,uart,wifi
 build 2021-01-18 08:47 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
• 3V3 -> VCC
• GND -> GND
• D2 -> SDA
 • D1 -> SCL
[image]

webstatic, 2021-01-18:
And when use disp:drawXBM, it displays about haft of the image before the reboot, with random progress.

[image]

sonaux, 2021-01-20:
It can be: 
power issue (onboard 3.3V regulator is not powerful enough)  - use external voltage regulator 
watchdog timed out - increase I2C speed; [ssd1327 datasheet](https://www.generationrobots.com/media/Ecran_OLED_0_96/SSD1327_datasheet.pdf) says on page 62 that it supports clock up to 400 kHz

Check "rst cause" after reboot:
Reset Cause | Description
0 | Unidentified
1 | Power reboot or normal boot
2 | External reset using reset pin or wake up from deep sleep
3 | Software reset
4 | Hardware watchdog (WDT) reset

webstatic, 2021-01-22:
Thank you for your reply.

Yes, I think the reason is watchdog timed out.
When I change some code to 

node.setcpufreq(node.CPU160MHZ)
i2c.setup(0, sda, scl, i2c.FASTPLUS)

the chance that it will not reboot is higher.

reset_reason is 3

Can I do something about it?

HHHartmann, 2021-01-22:
You could try and feed the software watchdog. https://nodemcu.readthedocs.io/en/release/modules/tmr/#tmrwdclr
You could also split the code in several methods which you separate by node.task.post()

You could also post (some of) the output of the reboot.

node.bootreason() returns 2 Values, the second contains better diagnostics. see https://nodemcu.readthedocs.io/en/release/modules/node/#nodebootreason

webstatic, 2021-01-24:
Thank you for your reply.

The watchdog reset comes inside single C function 
u8g2.ssd1327_i2c_ea_w128128(0, sla)
or 
disp:sendBuffer()
so tmr.wdclr() in lua won't work.

I not sure about  node.task.post(), sometimes it works, sometimes don't

node.bootreason() is  2	3, reset (software?) and software watchdog reset.

closed 2022-04-30 by stale[bot]