nodemcu/nodemcu-firmware #3379: bme280 LUA module does not load

Fivefold, 2021-01-21:
### Expected behavior
Load the bme280 LUA module. ``` require('bme280')``` should return an object with the bme280 functions.

### Actual behavior
``` require('bme280')``` returns ```true```. When using the test code (see below) ``` s = require('bme280').setup(0,2) ``` gives the following error:

```
init.lua:5: attempt to index a boolean value
stack traceback:
	init.lua:5: in main chunk
	[C]: ?
```

I hope ``` require('bme280')``` is the correct way to embed/include the Lua module?

### Test code
init.lua:
```Lua
alt=320 -- altitude of the measurement place

sda, scl = 6, 7
i2c.setup(0, sda, scl, i2c.SLOW)
s = require('bme280').setup(0,2)

tmr.create():alarm(500, tmr.ALARM_AUTO, function()
    local T, P, H, QNH = s:read(alt)
    local D = s:dewpoint(H, T)
    print(("T=%.2f, QFE=%.3f, QNH=%.3f, humidity=%.3f, dewpoint=%.2f"):format(T, P, QNH, H, D))
end)
```

This is the first example from the [documentation](https://nodemcu.readthedocs.io/en/release/lua-modules/bme280/), only adjusted to my I2C GPIO pins and the bme280 address.

### NodeMCU startup banner
```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: 4f6792773f93f36a7255cfd28dca7aa6c4aa9552
	release: 3.0-release_20201107
	release DTS: 202011071523
	SSL: true
	build type: float
	LFS: 0x0 bytes total capacity
	modules: adc,bme280_math,file,gpio,http,i2c,mqtt,net,node,rtctime,sjson,tmr,uart,wifi,tls
 build 2021-01-21 16:48 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```
### Hardware
[ESP12F](https://cdn.tindiemedia.com/images/resize/a_CjNvRLuZNRDh0g9uMSBBvYbCs=/p/full-fit-in/2400x1600/i/52445/products/2016-08-01T15%3A56%3A14.863Z-esp-12f-front-500px.jpg) on a custom PCB. I successfully used it for other projects
 and sensors in the past, so the board itself should not be the issue.

HHHartmann, 2021-01-21:
There currently is a bug which prints wrong type information. It is fixed in dev, however please use 
`print( require('bme280') )` as a workaround to see what is returned in the meantime.

Fivefold, 2021-01-22:
``` print( require('bme280') ) ``` prints ```true```

Is there a known workaround to use the lua module in the release build? Or do I need to switch to dev?

Fivefold, 2021-01-22:
After noticing that I still had some remnants of old code on the device I wiped/reflashed everything to get a clean slate: firmware, filesystem, lua files. Before, I only reflashed the firmware and init.lua. 

It works now, so I must have made a mistake before that I can't reproduce anymore. Sorry for the confusion!

closed 2021-01-22 by Fivefold