nodemcu/nodemcu-firmware #3383: Suggestion: Add short info on how to differentiate bme280/bmp280 sensors to readthedocs

Fivefold, 2021-01-23:
### Missing feature
Add a small paragraph to the [bme280 lua module documentation](https://nodemcu.readthedocs.io/en/release/lua-modules/bme280/) on how to differentiate between BME280 and BMP280 sensors.

The BME280 has a square shape, whereas the BMP280 has a rectangular one. [Image comparison](https://imgur.com/a/cXlS7L0)

I can make the edit and create a PR if this suggestion is accepted.

### Justification
The BMP280 sensor only supports temperature and pressure measurements.
The BME280 sensor supports temperature, pressure and humidity measurements.

The bme280 Lua module works with both of them and will happily give humidity measurements that are usually either 0 or 100 %, sometimes they are within a reasonable range (40-60) but always static.

Sellers of breakout modules sometimes confuse the two and sell one as the other. This happened to me and I spent a day debugging and trying to find out what was going on with my humidity measurements.

I suppose a check in the code that gives a warning if one tries to measure humidity with a BMP280 would work too. One possible way to find that out is to check the humidity oversampling setting/register. It can be written to in the BMP280 but won't change it's content which stays at 000. The check would unnecessarily hurt efficiency though (most of the time one will have the correct sensor) and be quite a bit of work for little payoff.

vsky279, 2021-01-24:
Hi @Fivefold, you are more than welcome to submit a PR precising the documentation. :thumbsup:

jmdasnoy, 2021-01-24:
@Fivefold @vsky279 Yes, the difference might not be clearly marked nor visible especially on cheaper breakout boards.
You can distinguish them by software.
Both chips have an id register at 0xd0.
Reading this register gives a chip signature of 0x60 for a BME280 but a signature of 0x58 for the BMP280.
This is used in the code  in the `setup()` function. but does not seem to report an error in case of mismatch.
You might consider adapting that.
Sorry I can't offer to do it myself as I have no ESP8266.
Regards.

marcelstoer, 2021-01-24:
@jmdasnoy thanks for the valuable information.

PR #3384 (2021-01-24): Doc update for bme280.lua
Fixes #3383.

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

1. Changed the note about bme280_math dependency into a blue note box, like in the [bh1750 documentation](https://nodemcu.readthedocs.io/en/release/lua-modules/bh1750/)
2. Added a pagragr [... cut]
closed 2021-01-24 by marcelstoer