nodemcu/nodemcu-firmware #3391: nodemcu-build doesn't work

azureblue, 2021-01-30:
Not getting any emails from https://nodemcu-build.com after triggering builds.

Tried different emails, days.

Yes, I checked spam folder.

marcelstoer, 2021-01-30:
Sorry for the trouble. 

This came completely out of the blue. The build sends me an email if things go south but it's been very quiet lately. I checked the logs, all good as well. Then I checked the build pipeline on travis-ci.org because Travis does the heavy lifting...not a pretty sight. For whatever reason their NodeMCU build queue is full and the backlog huge. No idea why. 

As an immediate measure I switched from their .org to .com as the former is soon shutting down soon. Hence, I would have had to transition at some point anyway.

They transitioned some of the build queue with it but unfortunately they're processing it the LIFO-way rather than FIFO. Again, no idea why; doesn't make sense to me.

If Travis continues to cause trouble I might have to bite the bullet and move everything to GitHub actions. @HHHartmann I feel you're quite happy with their offering?

closed 2021-01-30 by marcelstoer

azureblue, 2021-01-30:
Ok :). Thanks for the answer. :+1:

marcelstoer, 2021-01-30:
Oh, and in case it wasn't obvious: since they use LIFO you might be able to jump the queue if you just trigger another build 😜

azureblue, 2021-01-30:
Ok, let's test that statement then! :)

azureblue, 2021-01-30:
The build is indeed running :).

PS but why would they use LIFO? 😲