nodemcu/nodemcu-firmware #3397: ESP32: Trying to build with the current IDF

pjsg, 2021-02-11:
I just received one of the ESP32-C3 demo boards and so I thought that I'd get my feet wet with the nodemcu/esp32 code. However, it looks as though the version of the IDF that we are using is quite old, and when I upgrade to the current master branch, nothing really works.

The base makefile in nodemcu-firmware doesn't actually build anything (once you have the idf installed and configured). Worse, I don't seem able to get the nodemcu components included. If I do `make menuconfig` then nodemcu shows up, but it doesn't appear to put anything useful into sdkconfig. If I do `idf.py menuconfig` then the nodemcu config doesn't even show up.

Is there anybody who understands the nodemcu/idf build system? Can it be made to work with the current idf master branch (or even the release/v4.3 branch)?

marcelstoer, 2021-02-11:
As we're on 3.x LTS (see #3196) I expect upgrading to 4.x will be quite a bit of work. In theory it should "just" be the list of breaking changes that came with https://github.com/espressif/esp-idf/releases/tag/v4.0 but surprises wouldn't surprise me.

marcelstoer, 2021-02-11:
> Is there anybody who understands the nodemcu/idf build system?

At least @jmattsson should be familiar with it.

jmattsson, 2021-02-12:
I recall Angus saying something about them switching from GNU make to cmake with the 4.0 release, so probably no compatibility whatsoever between those two releases. I'm woefully out of date on my knowledge and experience here though :/

pjsg, 2021-02-12:
I'm making progress by making random changes to things and have been having
some success. I'm able to get some of nodemcu to build and it is mostly
around weird include path problems that is hampering me at the moment.

pjsg, 2021-02-12:
I have a branch `pjsg:idf4` that is my work in progress if anyone is
interested....

pjsg, 2021-02-14:
I can get an image to build and I can flash it, but it doesn't work. Any
ideas on how to debug this?

load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:15000
entry 0x40078630
[0;32mI (30) boot: ESP-IDF v3.1-dev-1381-g9a55b42f 2nd stage bootloader [0m
[0;32mI (30) boot: compile time 18:01:16 [0m
[0;32mI (30) boot: Enabling RNG early entropy source... [0m
[0;32mI (36) boot: SPI Speed      : 40MHz [0m
[0;32mI (40) boot: SPI Mode       : DIO [0m
[0;32mI (44) boot: SPI Flash Size : 4MB [0m
[0;32mI (48) boot: Partition Table: [0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset
Length [0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000
00006000 [0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000
00001000 [0m
[0;32mI (74) boot:  2 factory          factory app      00 00 00010000
00100000 [0m
[0;32mI (81) boot: End of partition table [0m
[0;32mI (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020
size=0x109c8 ( 68040) map [0m
[0;32mI (118) esp_image: segment 1: paddr=0x000209f0 vaddr=0x3ffb0000
size=0x02d34 ( 11572) load [0m
[0;32mI (123) esp_image: segment 2: paddr=0x0002372c vaddr=0x40080000
size=0x0c8ec ( 51436) load [0m
[0;32mI (146) esp_image: segment 3: paddr=0x00030020 vaddr=0x400d0020
size=0x41760 (268128) map [0m
[0;32mI (241) esp_image: segment 4: paddr=0x00071788 vaddr=0x4008c8ec
size=0x01040 (  4160) load [0m
[0;32mI (251) boot: Loaded app from partition at offset 0x10000 [0m
[0;32mI (251) boot: Disabling RNG early entropy source... [0m

then nothing..... The link map shows that the app_main is being included....

Philip

jmattsson, 2021-02-14:
See if the gdbstub is alive. If you can hook in with gdb you should be able
to make short work of the debugging. Docs e.g.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html

Failing that, try printf'ing your way through the start-up (that's what I
normally end up doing).

If you're still not getting somewhere, modifying the bootloader to actually
show the address it's jumping to when it thinks it's starting the app might
be useful / give you a clue.

pjsg, 2021-02-18:
I now have a version that boots on the esp32 with v4 of the IDF. I have a
file system, and I have wifi (though not the net module).

```I (144629) phy_init: phy_version 4660,0162888,Dec 23 2020
W (144629) phy_init: failed to load RF calibration data (0x1102), falling
back to full calibration
I (144789) wifi:mode : sta (24:0a:c4:06:f4:f4)
I (144789) wifi:enable tsf
> I (144799) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (145469) wifi:state: init -> auth (b0)
I (145479) wifi:state: auth -> assoc (0)
I (145479) wifi:state: assoc -> run (10)
I (145509) wifi:connected with Chateau Gladstone Guest, aid = 4, channel 1,
BW20, bssid = e2:63:da:7d:38:79
I (145509) wifi:security: WPA2-PSK, phy: bgn, rssi: -77
I (145509) wifi:pm start, type: 1

I (145589) wifi:AP's beacon interval = 102400 us, DTIM period = 3
W (145779) wifi:<ba-add>idx:0 (ifx:0, e2:63:da:7d:38:79), tid:0, ssn:1,
winSize:64
I (146779) esp_netif_handlers: sta ip: 192.168.103.140, mask:
255.255.255.0, gw: 192.168.103.1
```
However, the dev-esp32 branch is really old and I suspect that we *really*
need to move to a much more common codebase.

Another issue is that the code size seems huge -- the nodemcu wifi module
brings in maybe 400kb of code space and blew the 1MB code segment size. I
actually switched to using -O2 and that brought it down somewhat. I can
increase the partition size, but that seems wrong as well.

Anyway, my goal in all of this was to try and do the port to the esp32c3
which is (apparently) intended to be the new ESP8266 and come out at a
similar price point. That task is up next, but at least I know that I have
a roughly working codebase!

For those of you that haven't seen the new part, this is a side-by-side
photo with a standard ESP32 dev board. It certainly could fit into a D1
Mini sized PCB. There is nothing on the underside of the new board and only
silicon (as far as I can see) is the CP2104 USB/Serial interface, and some
voltage regulators and what looks like a very small WS2812 -- but I have no
idea what that part is.

[image: PXL_20210218_005422607.jpg]
Philip

pjsg, 2021-02-18:
After ignoring all the warnings in the esp32-c3 build and disabling a bunch
more modules, it boots:

```
 (60) boot_comm: chip revision: 2, min. bootloader chip revision: 0
I (67) boot.esp32c3: SPI Speed      : 80MHz
I (72) boot.esp32c3: SPI Mode       : DIO
I (77) boot.esp32c3: SPI Flash Size : 4MB
I (82) boot: Enabling RNG early entropy source...
I (87) boot: Partition Table:
I (91) boot: ## Label            Usage          Type ST Offset   Length
I (98) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (105) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (113) boot:  2 factory          factory app      00 00 00010000 00177000
I (120) boot:  3 node_spiffs      unknown          c2 00 00187000 00100000
I (128) boot: End of partition table
I (132) boot_comm: chip revision: 2, min. application chip revision: 0
I (139) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=252bch
(152252) map
I (172) esp_image: segment 1: paddr=000352e4 vaddr=3fc8f400 size=033c8h (
13256) load
I (175) esp_image: segment 2: paddr=000386b4 vaddr=40380000 size=07964h (
31076) load
I (183) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=d1fe4h
(860132) map
I (320) esp_image: segment 4: paddr=0011200c vaddr=40387964 size=07994h (
31124) load
I (326) esp_image: segment 5: paddr=001199a8 vaddr=50000000 size=00004h (
  4) load
I (332) boot: Loaded app from partition at offset 0x10000
I (332) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (406) cpu_start: Pro cpu start user code
I (406) cpu_start: cpu freq: 160000000
I (406) cpu_start: Application information:
I (409) cpu_start: Project name:     nodemcu
I (414) cpu_start: App version:      1.4.0-master_20151229-779-g1459
I (421) cpu_start: Compile time:     Feb 17 2021 21:17:50
I (427) cpu_start: ELF file SHA256:  6450465877ed63d3...
I (433) cpu_start: ESP-IDF:          v4.4-dev-4-g73db14240-dirty
I (439) heap_init: Initializing. RAM available for dynamic allocation:
I (447) heap_init: At 3FC981E0 len 00027E20 (159 KiB): DRAM
I (453) heap_init: At 3FCC0000 len 0001F260 (124 KiB): STACK/DRAM
I (460) heap_init: At 50000004 len 00001FFC (7 KiB): FAKEDRAM
I (466) spi_flash: detected chip: generic
I (471) spi_flash: flash io: dio
I (475) cpu_start: Starting scheduler.
怘��������fx~x�������f��fx�fx�~fx����x��f��~����~������怘�怘����~�����~����f���fx��~f�xxff�������fx����fx�fx~f�~f��xff��~怘��ff�~�x��f�����f~�x��f�xf��f~�xx��f��fx�`f~�xx��x���`ff~�xx���f��ff~�x~�xx�xf�x`f~���xf~�x�x�x�x���f��xf~�xx��f��fxx����f~�xx��f��fxx���~�xx��f��fxx���f~�xx��f��fx`������x~��x~�xx��f��fx����~���f~�xx��f����~������~��~�x�f�`����x~�xx��f��fxf���x~���f~��f~���~���~���f~��f~��f~��f��~���`fx~�f�`��f`~����~ff~�xf`fx~��x��x�����fffff�~�����fxf�~�x�����ffff~��fxf�~���昀���f��`~���怘�f~������fx���f�~怘�xf��fx~`�x�x~�����xx���xx��f��fxx���x�x`f���xx��f�f�������x��x���xf��fxf~�x�x�x�x`�x�f`�xx��f�����x��xx��f��fxf�xx��f����`�xx��f��fx`���x����xf��fx~~�xx�x��`�x��~����x`f��x�`�x`���xx��x�x�x�x�x�f`���xf��fx�~�xf�x�xf�x�xx���xxf��f���xf��fx~x�xx��f��x�x���xx�xx��f��x����xff���xf��fx`~x`��xx��f�f�x��x~���xf��fxx~���xx��f������xx��f���������x�xx�����xf���x~x��fx�x�����xf���x~fffffffffffffff怘�xf���xf~�x~��x��xx���������xx���xf���x~~�f�����f�x~�����xx�x�x�x���xx���xf���x�~��xx�x~����~�x���x`�x`�xff`��fx��xxxff��������xxxf�������xf���x~�xxfx�fxxxfx�xx�x~����~fffxx�xx��f�f�x����xf���x`~f�xfx�f�xx��x�`��ff��x`�����xxx�`fx`x�xxxf�����x�xf���xx�xxx�`���怘�xf���xx~�x�x�x�x�xxfx�x�xfx�xxfxf�����x�x�xxx�xxfxf��������xf���~ff���xf��fx���xxfx�fxxxfx�xf���~���xxx�������fx��fx�xxfxxxf���xf��x~�xf�x�x����x�xxfxxxf��f�x���fx������fx`�xf�x�x~���f`�xx��x�xf������x`�����xf���~f�x��xf�x�xxf����x��ffxx��x�xx�x��fff�����xf�xx��ff�������xf��~�x�x����f�����xf�x~�������xxx�������fx��fx怘�xf���~�xf�x�x�xx�xxf�xf�����xxxx�xx�xx�xf���xff���xxfx���xf��~f�xf�����xx���xx�xx�x�xx�xx�xx�xx�xfxx�xx�xx�xx���x���xf�x~�xx�x��x�xxx��������x�xx�������x�f��x~��x�xxx`���x怘�xf�x~x`�������x���
```

But doesn't work.....😣

KT819GM, 2021-05-25:
Any success on this? or you have dropped this idea? Mainly thinking about ESP32-S2 but have no idea how and if at all ...

pjsg, 2021-05-25:
I stopped at this stage. I need to see if Espressif added the support into
the IDF that we needed to get the linking to work. If so, then it ought to
be possible to build a very limited subset of modules with the current IDF.
I never tried building for the 8266, so I don't know if that would be
viable.

jmattsson, 2021-07-08:
I'm *hoping* I'll be able to spend some time bringing the ESP32 branch up to the latest IDF in the near future. I'm guessing the smartest thing would be to start from your idf4 branch, @pjsg?

pjsg, 2021-07-08:
I would certainly take a look there. It is fairly hacked up and doesn't
build most of nodemcu, but it built enough to get to a prompt. I don't know
whether espressif ever fixed the linker issue...

jmattsson, 2021-07-08:
I've just been going through the latest IDF documentation, and it looks like there's enough stuff in the linker fragment support to make things work now. I'll have to switch from our current null-terminated lists over to known-size arrays, but that should be easy enough.

I've also had a look through your branch, and if you're okay with it I think it'll be easier for me to start clean off dev-esp32 and then borrow heavily on a file-by-file or snippet-by-snippet basis so I can dodge the C3 stuff for now. I just don't want you to think I'd be doing it to steal your thunder - it certainly looks like you've done a great chunk of the heavy lifting already! :)

As long as $work doesn't give me new priorities I'm hoping to get onto this next week.

pjsg, 2021-07-08:
I'm fine with you starting again. I didn't really know what I was doing when I started. It didn't take me long to get to the stage that I got to. I'd start again as well!

jmattsson, 2021-07-12:
Okay, so I've started this, but have now hit a bit of a snag. The new linker fragments support appears to only be available on the `master` branch and not `v4.4-dev` or earlier, and said `master` branch is not currently in a buildable state it seems (issue in `newlib/locks.c`). I'll see if I can find a workaround and/or work my way backwards to hopefully find a sweet spot where both things work.

Say @igrr, I don't suppose you have an estimate on when the nice `KEEP()/ALIGN()/SURROUND()` support is going to arrive on a named branch?

igrr, 2021-07-12:
@jmattsson for the master branch you probably need to run `install.sh`, `. ./export.sh` and then `idf.py fullclean` — this is because the toolchain has recently been updated from esp-2020r3 to esp-2021r1. The build errors in locks.c you observe will disappear then.

We will create release/v4.4 around the end of August, and aim to release v4.4 around November.

jmattsson, 2021-07-12:
@igrr Ah, thank you for those pointers, that saved me some hours of troubleshooting! I'm now back to dealing with my own issues instead 🤣
And much obliged with the release estimates! The new ld fragments support should make my life considerably easier going forward, once I've tamed the transition.

jmattsson, 2021-07-22:
Okay, I believe I have this almost fully in the bag now over on [this branch](https://github.com/DiUS/nodemcu-firmware/tree/dev-esp32-idf4). What do people think of me pushing this to a (temporary) `dev-esp32-idf4` branch under the `nodemcu` org for better exposure/testing? I'd be reluctant to PR this right away, since it's a significant change and there are a few changes which break backwards compatibility. Do we have any highly visible release notes somewhere that should also be updated? (Sorry, I've been out of the loop for too long)

@jpeletier I haven't touched the "extmod" stuff yet, since I've never used it and am thus not quite sure of usage / best way forward to migrate from the 3.x IDF build structure to the 4.x IDF for those. I'd welcome your (and others') input here.

@igrr In the end it wasn't possible to move everything over to `_seg_start`/`_seg_end` symbol usage, so I've done "something clever" to get the zero-termination happening, which I'm hoping won't turn out to be fragile, but if you'd consider a feature request to add a linker fragment option of e.g. `APPEND_LONGS(n, val)` to facilitate easier placement of zeroes after a `SURROUND()` block that would be welcome :)

marcelstoer, 2021-07-22:
> What do people think of me pushing this to a (temporary) dev-esp32-idf4 branch under the nodemcu org for better exposure/testing?

By all means, go ahead!

> Do we have any highly visible release notes somewhere

We only write release notes for master (erg.. release) drops. Hence, we've never had ESP32 release. However, I'm sure people would appreciate up-to-date docs (what works, what doesn't) in the IDF 4 branch. If it won't mean too much work for me I could even make that branch available in the cloud builder. Maybe my [existing scripts](https://github.com/marcelstoer/nodemcu-custom-build/tree/master/ESP32) work out of the box?

jmattsson, 2021-07-23:
Thanks @marcelstoer.

Okay, there's now a `dev-esp32-idf4` branch here too :)  The docs in this branch have been updated, so where there was once a function there is now a note saying it has been removed due to underlying functionality disappearing upstream. It could be nice to those items listed somewhere explicitly so people can come forewarned rather than find out once their code throws an error. I don't know where might be a good place for that. Suggestions welcome.

Your cloud builder scripts would need some tweaking to cope with the new things. `make defconfig` was removed, but you can probably replace it with `cp sdkconfig.defaults sdkconfig`. There's a new `./install.sh` script which has to be run first, which chains through to the IDF's `install.sh` script automatically. Oh, and `cmake` needs to be available on the build host. If you're building in a directory where you previously had IDF 3.x, you'll also need to `find . -name *.pyc | xargs rm` to clean up interfering precompiled python things. And `MORE_CFLAGS` is no longer a thing. The integer-only build is an option in the sdkconfig now instead.

Oh, and I'd carefully updated the travis build script, only to discover that TravisCI.org is now fully gone. Do we have plans to migrate to something else, and if so what? (Somewhat tangential to the rest of the discussion, but...)

jpeletier, 2021-07-23:
> extmod

@jmattsson, drop this for now, I need to see the community demand to port this forward before dedicating effort, if it is only me I see no point yet.
Thank you for your work on this, looks great btw

jmattsson, 2021-07-23:
@pjsg I've also ordered myself a couple of C3 dev boards so I can start looking at that in a few weeks hopefully. We have some Xtensa-only assembler over in the lua component that we'll need to #ifdef properly now that we get RISC-V in the mix.

marcelstoer, 2021-07-23:
> It could be nice to those items listed somewhere explicitly so people can come forewarned rather than find out once their code throws an error.

Absolutely. Add a chapter/paragraph to docs entry page?

> Your cloud builder scripts would need some tweaking to cope with the new things. 

Thanks for the analysis. I don't see me having the capacity to look after that for the moment. However, by the time we want to merge your work back to the `dev-esp32` branch I would have to have it ready.

Btw, the cloud builder runs in a [GitHub CI pipeline](https://github.com/marcelstoer/nodemcu-custom-build/blob/master/.github/workflows/build.yml).

> TravisCI.org is now fully gone. Do we have plans to migrate to something else

@HHHartmann migrated the non-ESP32 braches from Travis to GitHub CI half a year ago: https://github.com/nodemcu/nodemcu-firmware/commits/release/.github/workflows. However, at the time he left the `.travis.yml` untouched as the service was still available.

jmattsson, 2021-07-23:
Thanks, I'll have a look at the github workflows then!
I'm also thinking that it might be nice to branch off dev-esp32 to a dev-esp32-idf3-legacy branch to give people a better chance to migrate as we hop onto idf4 for main dev. Thoughts?

KT819GM, 2021-07-23:
Any chance of using LUA 5.3 or later like ```https://github.com/UncleRus/esp-idf-lua```? Just a stupid question, have no idea how complicated it could be. For me ESP32-S2 with SPIRAM looks ideal for LUA, but after getting hands dirty with ESP8266 and LUA 5.3 I would like to go with ESP32-S2 and LUA 5.3 or 5.4.x

Thank you

jmattsson, 2021-07-24:
@KT819GM That is definitely on the cards. We're wanting to get to the point where we have a shared branch for both the ESP8226 and ESP32 so we can have as many features shared across both platforms as possible, including the new Lua version.

The path to get there is a bit challenging, seeing as we're still using the "non-OS" SDK on the ESP8266 which is very different from the RTOS & IDF on the ESP32 side. I'm hoping that now that $work has a desire for some of the features only available on the RTOS version in the 8266 SDK that I'll get some time to dedicate to this again. I had hoped to have had the branches merged well over a year ago, but then life & work & world circumstances changed 😩

marcelstoer, 2021-07-24:
> I'm also thinking that it might be nice to branch off dev-esp32 to a dev-esp32-idf3-legacy

That might sense once your IDF 4 branch is sufficiently stable. Just let me know when you're ready so I can adjust the cloud builder such that at least that legacy branch is still supported.

> Any chance of using LUA 5.3 or later like https://github.com/UncleRus/esp-idf-lua

OT: Lua is not an acronym 😉 
I can't assess the quality of that project but if someone is familiar with IDF and Lua then maybe she/he could be convinced to join forces with us?

jmattsson, 2021-07-29:
Okay I've switched this branch over to github actions. The workflow is not as full-featured as on the esp8266 branch, but it's matching what the old travis build was doing at least. I've also fixed a couple of issues I found, and done a code-cleanup to sort out most of the warnings (the luac.cross build is still giving off a bunch, I may or may not investigate those).

We're slowly starting to test the idf4 stuff at $work, but that will only cover the parts we actually use, so everyone, please do dig in and tell me where I b0rked the upgrade :)

jmattsson, 2021-08-04:
Okay, I took care of the luac.cross warnings too.

jmattsson, 2021-08-10:
My ESP32C3 boards have arrived. I'm hoping to be able to push coexisting C3 & regular ESP32 support within a day or two (much thanks to @pjsg's existing work on getting the C3 up and running).

jmattsson, 2021-08-11:
Alright, I've pushed the C3 stuff. I don't _think_ I've broken anything... :D

marcelstoer, 2021-08-13:
@jmattsson should we lock the `dev-esp32` branch to avoid nasty merge conflicts?

jmattsson, 2021-08-13:
I've heard worse ideas. If others are cool with it, that would certainly make life easier.

I'm currently working on bringing over Lua 5.3 from the 8266 branch, fwiw.

jmattsson, 2021-08-21:
@KT819GM Well, it turned out that getting the 5.1+5.3 work from the ESP8266 branch over to the ESP32 branch was a lot more work than I had expected, to the tune of over a week's worth of evening effort. That said, I've got it all going in a branch [over here](https://github.com/jmattsson/nodemcu-firmware/tree/dev-esp32-idf4-lua53) if you want to try it out. I need to go over the docs and see what needs updating there before I push to the main repo, but that's a job for another night.

KT819GM, 2021-08-21:
Hi,
I would happily test it, but at home somehow I have only ESP32-S2 available and:
When trying to build it fails on the CAN module which is disabled on a `menuconfig`. if I set target for ESP32 build is successful, but I have no live unit to try. Checked `sdkconfig` to make sure that CAN is disabled. 
In any case t.y. for your hard work, if you could look at the build issue and could tell me how to avoid CAN module I would try to build and flash it onto device.

About build process itself - not sure who done this, but it's straight forward on Ubuntu 20.04 in three steps (four if need to `apt install cmake`) in total, so perfectly doable on WSL on windows (just did that). 

error itself with `idf.py set-target esp32-s2`:
```C
In file included from /home/mode/esp/nodemcu-firmware/sdk/esp32-esp-idf/components/soc/esp32s2/include/soc/dport_reg.h:33,
                 from /home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:36:
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c: In function 'CAN_init':
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:163:53: error: 'DPORT_CAN_CLK_EN' undeclared (first use in this function); did you mean 'DPORT_RMT_CLK_EN'?
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                                                     ^~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/sdk/esp32-esp-idf/components/soc/esp32s2/include/soc/dport_access.h:76:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
/home/mode/esp/nodemcu-firmware/sdk/esp32-esp-idf/components/soc/esp32s2/include/soc/dport_access.h:89:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:163:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:163:53: note: each undeclared identifier is reported only once for each function it appears in
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                                                     ^~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/sdk/esp32-esp-idf/components/soc/esp32s2/include/soc/dport_access.h:76:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
/home/mode/esp/nodemcu-firmware/sdk/esp32-esp-idf/components/soc/esp32s2/include/soc/dport_access.h:89:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:163:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:164:55: error: 'DPORT_CAN_RST' undeclared (first use in this function); did you mean 'DPORT_WDG_RST'?
     DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
                                                       ^~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/sdk/esp32-esp-idf/components/soc/esp32s2/include/soc/dport_access.h:76:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
/home/mode/esp/nodemcu-firmware/sdk/esp32-esp-idf/components/soc/esp32s2/include/soc/dport_access.h:86:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
                                              ^~~~~~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:164:5: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
     DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:168:39: error: 'CAN_TX_IDX' undeclared (first use in this function); did you mean 'TWAI_TX_IDX'?
     gpio_matrix_out(CAN_cfg.tx_pin_id,CAN_TX_IDX,0,0);
                                       ^~~~~~~~~~
                                       TWAI_TX_IDX
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:173:35: error: 'CAN_RX_IDX' undeclared (first use in this function); did you mean 'TWAI_RX_IDX'?
  gpio_matrix_in(CAN_cfg.rx_pin_id,CAN_RX_IDX,0);
                                   ^~~~~~~~~~
                                   TWAI_RX_IDX
/home/mode/esp/nodemcu-firmware/components/driver_can/CAN.c:228:20: error: 'ETS_CAN_INTR_SOURCE' undeclared (first use in this function)
     esp_intr_alloc(ETS_CAN_INTR_SOURCE,0,CAN_isr,NULL,NULL);
                    ^~~~~~~~~~~~~~~~~~~
make[3]: *** [esp-idf/driver_can/CMakeFiles/__idf_driver_can.dir/build.make:63: esp-idf/driver_can/CMakeFiles/__idf_driver_can.dir/CAN.c.obj] Error 1
make[3]: Leaving directory '/home/mode/esp/nodemcu-firmware/build'
make[2]: *** [CMakeFiles/Makefile2:5840: esp-idf/driver_can/CMakeFiles/__idf_driver_can.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
```

marcelstoer, 2021-08-21:
@jmattsson thanks a lot for your hard work! As this thread is quite long already I was wondering whether it's more effective to discuss Lua 5.3 on ESP32 in a separate issue or whether that'd be more distracting as IDF 4 and Lua 5.3 are going to land in the same PR (towards `dev-esp32`) anyway.

jmattsson, 2021-08-22:
@KT819GM Hey thanks for pointing that out! I had not realised the S2/S3 had different peripherals. I've reworked the modules/driver handling, so it should build on my branch now.

@marcelstoer Yeah I think this is all sufficiently big that a "big bang" approach is the only sensible way to do it now. I think I'm still of the view that a rename of our existing `dev-esp32` to `dev-esp32-legacy` and swap the `dev-esp32-idf4` into its place is going to be the least insane thing. I've done the docs, and I've sorted out the S2/S3 issues, so I'm tempted to merge my `dev-esp32-idf4-lua53` branch onto `dev-esp32-idf4` here again. I'm not aware of any regressions from my porting of the Lua 5.1+5.3 stuff, so I think it's ready. A second view/opinion certainly appreciated.

I still want to have a look at the VFS stuff before we do any cut-over though. By now it feels we might be suffering from using our own VFS implementation over the Espressif/IDF provided one. Switching over would allow for more regular POSIX type code, which wouldn't be a bad thing. Plus any IDF components which use the filesystem would actually find files where they expect them...

KT819GM, 2021-08-22:
@jmattsson Hi, I've tried with updated branch, now I'm up to this point:
```
NodeMCU ESP32 build unspecified powered by Lua 5.3.5 on IDF v4.4-dev-1849-g8e3e65a47b
cannot open init.lua: 
> print("hello")
Guru Meditation Error: Core  0 panic'ed (Memory protection fault). 
```

With debug enabled could get bit more info, sadly not usable for me:
```
NodeMCU ESP32 build unspecified powered by Lua 5.3.5 on IDF v4.4-dev-1849-g8e3e65a47b
cannot open init.lua: 
> print("hello")
Guru Meditation Error: Core  0 panic'ed (Memory protection fault). 
abort() was called at PC 0x40026de3 on core 0
abort() was called at PC 0x40026de3 on core 0
abort() was called at PC 0x40026de3 on core 0
Backtrace:0x40026b0d:0x3fd90560 |<-CORRUPTED
ELF file SHA256: 5c9f801ce5737c8b
```
I've disabled SPI RAM, not used LFS and in general any activity in the terminal (pressing enter on a empty field) will force panic. 
Full startup header:
```
Mounting flash filesystem...
D (729) partition: Loading the partition table[1B][0m
D (729) partition: Partition table MD5 verified[1B][0m
D (739) esp_netif_lwip: LwIP stack has been initialized[1B][0m
D (739) esp_netif_lwip: esp-netif has been successfully initialized[1B][0m
D (739) nvs: nvs_open_from_partition lfsload 0[1B][0m
D (739) nvs: nvs_open_from_partition lfsload 0[1B][0m
D (739) intr_alloc: Connected src 23 to int 12 (cpu 0)[1B][0m
D (739) esp_netif_objects: esp_netif_add_to_list 0x3ffd95e4[1B][0m
D (739) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)[1B][0m
D (749) esp_netif_objects: esp_netif_add_to_list 0x3ffd9b4c[1B][0m
D (749) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 2)[1B][0m
D (749) nvs: nvs_open_from_partition misc 1[1B][0m
D (749) nvs: nvs_get_str_or_blob log[1B][0m
I (759) wifi:wifi driver task: 3ffda568, prio:23, stack:6656, core=0
[1B][0;32mI (759) system_api: Base MAC address is not set[1B][0m
[1B][0;32mI (759) system_api: read default base MAC address from EFUSE[1B][0m
D (759) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 8 bit[1B][0m
D (759) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 0 bit[1B][0m
D (759) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 24 bit[1B][0m
D (759) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 16 bit[1B][0m
D (759) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 8 bit[1B][0m
D (769) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 0 bit[1B][0m
D (769) nvs: nvs_open_from_partition nvs.net80211 1[1B][0m
D (769) nvs: nvs_get opmode 1[1B][0m
D (769) nvs: nvs_get_str_or_blob sta.ssid[1B][0m
D (769) nvs: nvs_get sta.authmode 1[1B][0m
D (769) nvs: nvs_get_str_or_blob sta.pswd[1B][0m
D (769) nvs: nvs_get_str_or_blob sta.pmk[1B][0m
D (769) nvs: nvs_get sta.chan 1[1B][0m
D (769) nvs: nvs_get auto.conn 1[1B][0m
D (779) nvs: nvs_get bssid.set 1[1B][0m
D (779) nvs: nvs_get_str_or_blob sta.bssid[1B][0m
D (779) nvs: nvs_get sta.lis_intval 2[1B][0m
D (779) nvs: nvs_get sta.phym 1[1B][0m
D (779) nvs: nvs_get sta.phybw 1[1B][0m
D (779) nvs: nvs_get_str_or_blob sta.apsw[1B][0m
D (779) nvs: nvs_get_str_or_blob sta.apinfo[1B][0m
D (779) nvs: nvs_get sta.scan_method 1[1B][0m
D (779) nvs: nvs_get sta.sort_method 1[1B][0m
D (779) nvs: nvs_get sta.minrssi 1[1B][0m
D (779) nvs: nvs_get sta.minauth 1[1B][0m
D (789) nvs: nvs_get sta.pmf_e 1[1B][0m
D (789) nvs: nvs_get sta.pmf_r 1[1B][0m
D (789) nvs: nvs_get sta.btm_e 1[1B][0m
D (789) nvs: nvs_get sta.rrm_e 1[1B][0m
D (789) nvs: nvs_get_str_or_blob ap.ssid[1B][0m
D (789) nvs: nvs_get_str_or_blob ap.passwd[1B][0m
D (789) nvs: nvs_get_str_or_blob ap.pmk[1B][0m
D (789) nvs: nvs_get ap.chan 1[1B][0m
D (789) nvs: nvs_get ap.authmode 1[1B][0m
D (789) nvs: nvs_get ap.hidden 1[1B][0m
D (789) nvs: nvs_get ap.max.conn 1[1B][0m
D (789) nvs: nvs_get bcn.interval 2[1B][0m
D (799) nvs: nvs_get ap.phym 1[1B][0m
D (799) nvs: nvs_get ap.phybw 1[1B][0m
D (799) nvs: nvs_get ap.sndchan 1[1B][0m
D (799) nvs: nvs_get ap.pmf_e 1[1B][0m
D (799) nvs: nvs_get ap.pmf_r 1[1B][0m
D (799) nvs: nvs_get ap.p_cipher 1[1B][0m
D (799) nvs: nvs_get lorate 1[1B][0m
D (799) nvs: nvs_get_str_or_blob country[1B][0m
D (799) nvs: nvs_get ap.ftm_r 1[1B][0m
D (799) nvs: nvs_set ap.sndchan 1 1[1B][0m
I (799) wifi:wifi firmware version: ff5f4ea
I (799) wifi:wifi certification version: v7.0
I (809) wifi:config NVS flash: enabled
I (809) wifi:config nano formating: disabled
I (809) wifi:Init data frame dynamic rx buffer num: 32
I (809) wifi:Init management frame dynamic rx buffer num: 32
I (809) wifi:Init management short buffer num: 32
I (809) wifi:Init static tx buffer num: 16
I (809) wifi:Init static rx buffer size: 1600
I (809) wifi:Init static rx buffer num: 10
I (809) wifi:Init dynamic rx buffer num: 32
[1B][0;32mI (819) wifi_init: rx ba win: 6[1B][0m
[1B][0;32mI (819) wifi_init: tcpip mbox: 32[1B][0m
[1B][0;32mI (819) wifi_init: udp mbox: 6[1B][0m
[1B][0;32mI (819) wifi_init: tcp mbox: 6[1B][0m
[1B][0;32mI (819) wifi_init: tcp tx win: 5744[1B][0m
[1B][0;32mI (819) wifi_init: tcp rx win: 5744[1B][0m
[1B][0;32mI (819) wifi_init: tcp mss: 1440[1B][0m
[1B][0;32mI (819) wifi_init: WiFi IRAM OP enabled[1B][0m
[1B][0;32mI (819) wifi_init: WiFi RX IRAM OP enabled[1B][0m

NodeMCU ESP32 build unspecified powered by Lua 5.3.5 on IDF v4.4-dev-1849-g8e3e65a47b
cannot open init.lua: 
```

Thank you

marcelstoer, 2021-08-22:
> I'm still of the view that a rename of our existing `dev-esp32` to `dev-esp32-legacy` and swap the `dev-esp32-idf4` into its place is going to be the least insane thing. 

All fine for me. In accordance with our `1.5.4.1-final` we could also call it `dev-esp32-idf3-final` to express that it's effectively frozen/unmaintained/etc. I need to make a few small adjustments to the Docker image in the cloud builder to support building from that IDF3 branch. So, I appreciate it if we can coordinate this to avoid disruptions.

> I've done the docs, and I've sorted out the S2/S3 issues, so I'm tempted to merge my `dev-esp32-idf4-lua53` branch onto `dev-esp32-idf4` here again.

Fine for me.

jmattsson, 2021-08-23:
@KT819GM Hmm, that's weird. I am unable to reproduce that here on my ESP32 and ESP32C3 boards. Looking at that startup log though, it looks like the external SPI RAM is still active? `spiram: Adding pool of 2048K of external SPI memory to heap allocator`. What board are you using? I might need to get myself an ESP32S2 so I can track this down...

igrr, 2021-08-23:
> Memory protection fault

On ESP32-S2 and later, this means that the program has either tried to write to memory which is used for code storage, or tried to execute from memory used for data storage. I'm not sure if this is something NodeMCU might be doing on purpose, or is a simple out-of-bounds memory access.

KT819GM, 2021-08-23:
I will take ESP32 home today to try on the same dev. environment and will come back here with results. Log which I've posted with bootloader messages was with psram enabled (I've noted that before log 🙃) but in general I have tested with and without psram with same results.

jmattsson, 2021-08-23:
@KT819GM I may have gotten part-way to something that can help work out what's going on with your S2. I discovered that Lua panic handling on the console was broken, and have now pushed a fix for that. I've also swapped from our old console UART driver to the regular IDF UART driver. I'm hoping at least one of these will mean we get something more informative when it crashes...

@igrr Any chance of not having the IDF UART driver refuse to install when the panic-to-gdb-stub option is enabled? Removing that check made my `make monitor` -> automatic gdb session work just fine.

KT819GM, 2021-08-23:
Ok, so I'm not too fast as @jmattsson  so did not even tried ESP32. Pulled latest commit and result is:
```C
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x212c
load:0x4004c000,len:0xb88
load:0x40050000,len:0x2fec
entry 0x4004c234
[1B][0;32mI (21) boot: ESP-IDF v4.4-dev-1849-g8e3e65a47b 2nd stage bootloader[1B][0m
[1B][0;32mI (21) boot: compile time 19:46:59[1B][0m
D (21) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)[1B][0m
[1B][0;32mI (29) boot: chip revision: 0[1B][0m
D (32) boot.esp32s2: magic e9[1B][0m
D (35) boot.esp32s2: segments 03[1B][0m
D (39) boot.esp32s2: spi_mode 02[1B][0m
D (42) boot.esp32s2: spi_speed 0f[1B][0m
D (45) boot.esp32s2: spi_size 02[1B][0m
[1B][0;32mI (49) boot.esp32s2: SPI Speed      : 80MHz[1B][0m
[1B][0;32mI (53) boot.esp32s2: SPI Mode       : DIO[1B][0m
[1B][0;32mI (58) boot.esp32s2: SPI Flash Size : 4MB[1B][0m
D (63) boot: Enabling RTCWDT(9000 ms)[1B][0m
[1B][0;32mI (66) boot: Enabling RNG early entropy source...[1B][0m
D (72) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0[1B][0m
D (81) boot: mapped partition table 0x8000 at 0x3f008000[1B][0m
D (86) flash_parts: partition table verified, 6 entries[1B][0m
[1B][0;32mI (91) boot: Partition Table:[1B][0m
[1B][0;32mI (95) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
D (102) boot: load partition table entry 0x3f008000[1B][0m
D (107) boot: type=1 subtype=2[1B][0m
[1B][0;32mI (110) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
D (118) boot: load partition table entry 0x3f008020[1B][0m
D (123) boot: type=1 subtype=1[1B][0m
[1B][0;32mI (126) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
D (134) boot: load partition table entry 0x3f008040[1B][0m
D (139) boot: type=0 subtype=0[1B][0m
[1B][0;32mI (142) boot:  2 factory          factory app      00 00 00010000 00180000[1B][0m
D (149) boot: load partition table entry 0x3f008060[1B][0m
D (154) boot: type=c2 subtype=1[1B][0m
[1B][0;32mI (157) boot:  3 lfs              unknown          c2 01 00190000 00010000[1B][0m
D (165) boot: load partition table entry 0x3f008080[1B][0m
D (170) boot: type=c2 subtype=0[1B][0m
[1B][0;32mI (173) boot:  4 nodemcuspiffs    unknown          c2 00 001a0000 00070000[1B][0m
[1B][0;32mI (181) boot: End of partition table[1B][0m
D (185) boot: Trying partition index -1 offs 0x10000 size 0x180000[1B][0m
D (191) esp_image: reading image header @ 0x10000[1B][0m
D (196) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)[1B][0m
D (203) esp_image: image header: 0xe9 0x07 0x02 0x02 400254a8[1B][0m
V (209) esp_image: loading segment header 0 at offset 0x10018[1B][0m
V (214) esp_image: segment data length 0x2f820 data starts 0x10020[1B][0m
V (221) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f000020[1B][0m
[1B][0;32mI (229) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=2f820h (194592) map[1B][0m
D (237) esp_image: free data page_count 0x0000003f[1B][0m
D (242) bootloader_flash: mmu set paddr=00010000 count=3 size=2f820 src_addr=10020 src_addr_aligned=10000[1B][0m
V (291) esp_image: loading segment header 1 at offset 0x3f840[1B][0m
D (291) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)[1B][0m
V (292) esp_image: segment data length 0x7d0 data starts 0x3f848[1B][0m
V (298) esp_image: segment 1 map_segment 0 segment_data_offs 0x3f848 load_addr 0x3ffca0e0[1B][0m
[1B][0;32mI (307) esp_image: segment 1: paddr=0003f848 vaddr=3ffca0e0 size=007d0h (  2000) load[1B][0m
D (315) esp_image: free data page_count 0x0000003f[1B][0m
D (320) bootloader_flash: mmu set paddr=00030000 count=2 size=7d0 src_addr=3f848 src_addr_aligned=30000[1B][0m
V (330) esp_image: loading segment header 2 at offset 0x40018[1B][0m
D (335) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)[1B][0m
V (342) esp_image: segment data length 0xbd828 data starts 0x40020[1B][0m
V (348) esp_image: segment 2 map_segment 1 segment_data_offs 0x40020 load_addr 0x40080020[1B][0m
[1B][0;32mI (357) esp_image: segment 2: paddr=00040020 vaddr=40080020 size=bd828h (776232) map[1B][0m
D (365) esp_image: free data page_count 0x0000003f[1B][0m
D (370) bootloader_flash: mmu set paddr=00040000 count=12 size=bd828 src_addr=40020 src_addr_aligned=40000[1B][0m
[... 199 lines cut ...]
[1B][0;32mI (1976) wifi_init: tcp mbox: 6[1B][0m
[1B][0;32mI (1976) wifi_init: tcp tx win: 5744[1B][0m
[1B][0;32mI (1986) wifi_init: tcp rx win: 5744[1B][0m
[1B][0;32mI (1986) wifi_init: tcp mss: 1440[1B][0m
[1B][0;32mI (1986) wifi_init: WiFi IRAM OP enabled[1B][0m
[1B][0;32mI (1996) wifi_init: WiFi RX IRAM OP enabled[1B][0m

NodeMCU ESP32 build unspecified powered by Lua 5.3.5 on IDF v4.4-dev-1849-g8e3e65a47b
cannot open init.lua: 
```
Following:
```Lua
> =node.heap()
2123699
> print("now it's working, with PSRAM enabled :)")
now it's working, with PSRAM enabled :)
```
That heap size guys ... even my worst code will fit easily. 

Now for the serious questions:
1. As I'm not familiar yet with ESP32 series on nodemcu, asking if we can get into the approach to use ESP-IDF modules for WiFi and Mqtt handling (if not using yet) by making some kind of a wrapper or whatever, if it's not like that yet. Simply on ESP8266 there are some mysterious behavior with, in my view critical components for "internet connected" units, and not enough people to try to fix them. 
2. Maybe we (I mean devs, with rights to post for the users like myself) should start using github discussions, as that would allow to keep issues more clean from the features development.

At the moment all from me, will try to run some usable code tonight. 
Thank you

igrr, 2021-08-23:
@jmattsson thanks for the heads up regarding the UART & GDB Stub. There is a mistake in the code, it was supposed to check CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME instead. Will submit a fix ASAP.

KT819GM, 2021-08-23:
Have tried to play a little bit more, WiFi is working. ADC is working. So tried mqtt module from an example in a esp32-dev. It not worked with an error:
```Lua
dofile("mqtt.lua")
D (10505551) MQTT: MQTT client id clientid[0m
Lua error: 	mqtt.lua:7: attempt to index a userdata value (field 'm')
stack traceback:
	mqtt.lua:7: in main chunk
	[C]: in function 'ROM.dofile'
	stdin:1: in main chunk
	[C]: in ?
	[C]: in ?
```
In the coming days I will try to test more peripherals but as a Marcel already mentioned this thread is getting long and also especially with my questions is getting out of scope. Same could be applied to the issues with this at the moment special branch which has nothing to do with the current main branches. So I will wait for you people to tell us how you will move forward with the IDF4 and where should we start posting issues.

Thank you

jmattsson, 2021-08-23:
@igrr Ah, lovely, thanks Ivan! Dare I also mention that panic-to-gdb-stub is broken on the C3? ("`undefined reference to esp_gdbstub_writemem`"; looks like that's not implemented on riscv yet)

jmattsson, 2021-08-23:
@KT819GM Thanks for reminding me. I needed to update the modules to comply with the new ordering restrictions in the rom tables. I've pushed a commit which should deal with the "attempt to index a userdata value" issue.

jmattsson, 2021-09-21:
@marcelstoer I think this branch is just about ready. At $work we're already using parts of it in production (pre-5.1+5.3 introduction), and we're now testing with 5.3 and it's looking solid so far.

jmattsson, 2021-10-17:
Alright, our old VFS stuff has annoyed me enough that I'm doing something about it. Substantial overhaul in progress to switch things over to the RTOS safe VFS that's provided in the IDF, plus swapping `file` for standard `io` (hopefully). This should be the last breaking change for now I hope.

pjsg, 2021-10-21:
> If someone is interested in creating a Lua shim for forwarding old file
> module functions over to io, I would not object.
>
> I feel an old->new table with function names in the docs would be enough.

I just updated to the new branch that eliminates the file functions, and
esplorer (my ide) now doesn't work as it relies on the file.* methods to
operate.

What do other people use as an IDE these days? Even, how do you download
source onto an ESP32 board?

Philip

jmattsson, 2021-10-21:
Hmm, breaking Esplorer is indeed not ideal. I could certainly be convinced that we should perhaps then have a compatibility option in Kconfig that brings in whatever Esplorer needs then.

I do use Esplorer, but only for interactive testing. For putting code onto the system it's LFS all the way.

pjsg, 2021-10-22:
> Would this be a good opportunity to remove all that code?
>
> There is certainly something to be said for getting all the breakage done
> in one go!
>
> Slippery slope then has me mentioning our inconsistent function naming
> with somethings using underscores and others not...

I put up a PR for ESPLorer -- I don't think that I caught all the issues.
Did the `wdclr` method go somewhere in the esp32 branch? I think that we
still need it.....

jmattsson, 2021-10-22:
I don't think it's implemented. FreeRTOS does its own hung task detection, so on the esp32 you'd be more likely to want/need to use coroutines and/or node.task.post() to break things up.

Thanks for preparing and submitting that Esplorer PR!

pjsg, 2021-10-22:
> Okay, I've provided a tmr.wdclr() function. I did not see a better way of
> implementing it, at least not while sticking to the public FreeRTOS API.

While testing the explorer stuff, i did get a wdt expiry, so something can
go wrong!

kxygk, 2022-05-18:
I just wanted to tack on here, in case someone else comes across this, that the current instructions on the `dev-esp32` and `dev-esp32-idf4` branch will not work for the C3 variant

https://nodemcu.readthedocs.io/en/dev-esp32/
https://nodemcu.readthedocs.io/en/dev-esp32-idf4/build/

If someone wants to try it, you need to make some minor changes to force the tools to build for this target (it defaults to the xtensa ESP).

Some of the dependencies are also a bit different on Ubuntu 22.04. For instance instead of `python-pip` it was `python3-pip`

1) you do the usual recursive clone as before

2) you need to call `./install.sh esp32c3`

3) the `make` method won't work and you need to call the python script directly. First you need to specify the target
`/sdk/esp32-esp-idf/tools/idf.py set-target esp32c3`

4) I think at this point you can configure what modules you want with `make menuconfig` - however do note that some of the Build Options have changed. I think the UART baud rate is no longer where it used to be

5) Then you can can build the IDF with `./sdk/esp32-esp-idf/tools/idf.py build`

6) Then you can flash it with `./sdk/esp32-esp-idf/tools/idf.py -p /dev/ttyACM0 flash` (modify the port to match your own)

I think I must have some minor error though. B/c while at this point I can connect with `nodemcu-tool -p /dev/ttyACM0 terminal` for instance, I can't actually upload code. If I try to upload code I get an error

```
> nodemcu-tool -p /dev/ttyACM0 upload lua_examples/webap_toggle_pin.lua 
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Invalid node.chipid() Response: Lua error:      stdin:1: attempt to call field 'chipid' (a nil value)
stack traceback:
        stdin:1: in main chunk
        [C]: ?
        [C]: ? 

```

If anyone sees the problem, please do let me know :)

pjsg, 2022-05-19:
Thank you for doing this investigation -- I just got a couple of C3 boards and it was on my list of things to try.

I'll try your steps and see if it works any better for me!

kxygk, 2022-05-19:
Oh I thought you guys had already been running on C3's :)
I didn't realize i was going into terra icognita here

I didn't actually run `pip install -r sdk/esp32-esp-idf/requirements.txt` (where did you find this?). I think the build steps pulled in the python dependencies on their own. Though I could be wrong.. I did do some fiddling around before I got things working

I think when you run `/.install.sh esp32c3` it should pull in all these toolchains. You can verify if the toolchain is present by going to `~/.espressif/` and searching for `riscv32`

jmattsson, 2022-05-19:
Things were working fine with my ESP32C3 back when I got it supported, but I haven't tinkered with it (or any ESP32) for some time now. The `install.sh` in the nodemcu-firmware directory should pull in everything needed, and the Makefile sets up the necessary paths.

[19 more replies]

closed 2023-04-30 by jmattsson