nodemcu/nodemcu-firmware #3402: wifi.sta.setip() validation

chathurangawijetunge, 2021-02-22:
wifi.sta.setip({ ip = "192.168.0.111", netmask = "255.255.255.0",gateway = "192.168.0.1"})
with above we can set static IP,
but by mistake if someone enters the netmask/gateway/ip a digit or to wrong (eg:netmask = "255.255.2.0" ) we have no way of detracting it. or correcting it 
1) by re entering correctly
2) rebooting

so it would be nice if we can detect the status somehow like with wifi.sta.status() do.

chathurangawijetunge, 2021-05-05:
No way to clean static ip flag and switch to DHCP unless reboot...!

chathurangawijetunge, 2021-06-08:
we must add 'wifi.sta.dhcp()'

Solved in wifi.c

static int wifi_station__dhcp( lua_State* L )
{
 lua_pushboolean(L, wifi_station_dhcpc_start());
 return 1; 
}

TerryE, 2021-09-26:
@chathurangawijetunge,  The workaround is to restart your app to do this switch.   if you want to raise PR then reopen the issue and do so.

closed 2021-09-26 by TerryE