nodemcu/nodemcu-firmware #3405: ESP32 WS2812 module not working as documented with M5Stack Hex neopixel board

jmdasnoy, 2021-03-02:
The ws2812 module is meant for driving addressable leds a.k.a. "Neo Pixels".
The documented examples don't work as described on an ESP-32 driving a recent M5Stack hex neopixel board (37 SK6812 GRB leds)
Symptoms include flickering during writes, bad pixel addressing or colours, incomplete fills.

One workaround is to repeat each write instruction
  ``ws2812.write( {pin = hexpin, data = buffer} ) ; 
  ws2812.write( {pin = hexpin, data = buffer} )``

or to issue an empty write before each effective write
  ``ws2812.write( {pin = hexpin, data = ""} ) ; 
  ws2812.write( {pin = hexpin, data = buffer} )``

The ws2812 module issues a reset signal to the leds before writing out the effective data.
This signal currently has a length of 50 µS. It may have been long enough for older neopixels, but the SK6812 pixels on the M5Stack board seem to expect a reset pulse longer than 80 µS. Available datasheets are quite laconic or cryptic...

The workaround of an empty write before a full data write seems to work by issuing two back to back reset signals before sending out data.

I have modified the timing definitions in components/platform/ws2812.c to provide a 90 µS reset pulse.
I have also adapted the bit on/off timings to add up to 1.2 µS.

With these changes, the documented examples (and many others) work reliably and as expected.
A longer reset pulse should not cause any problems to older leds, so this proposed change should not break any existing code or applications.

Existing code also assumes an 80 MHz APB clock. This could also be rewritten to use the effective APB clock frequency.

If this is of interest or use to the community, I will provide a pull request with these changes and updated documentation.
Let me know your thoughts.

f4grx, 2021-09-30:
hello yes, I am interested, I had to do the same kind of code changes for my own m5stack.

the timing should be configurable by indicating the neopixel model to use.

jmdasnoy, 2021-10-03:
@f4grx thanks for your interest and comments. It's nice to know I wasn't the only one to struggle with this !
What code changes did you make to get this working ?
Are you planning to make a pull request ?
Your suggestion of configurable timing is interesting but it should not break existing code, so this needs some more thought...
Regards.

f4grx, 2021-10-04:
I just changed the timings in the cod to match the specs of these particular leds, because I have a specific source tree for the m5 stack.

Unfortunately I did not make this code configurable in any way :(

marcelstoer, 2021-10-04:
I am no expert on the subject but after watching https://youtu.be/rHoFqKGOPRI?t=560 I at least understand the basic principle of these pixels work. 

So, yes, the timing for the SK6812 does appear to be incompatible with the ws2812.

f4grx, 2021-10-06:
It is great to learn that the changes are so simple. Also appreciated, thanks.