nodemcu/nodemcu-firmware #3406: Loss of accuracy with math.floor

vidalouiswang, 2021-03-10:
If do 
`print(math.floor(1615391446930/36))`
The value is 2^32 -1, this is incorrent.

And the offical doc says support Lua 5.3, but why can't use "//" ?

Thank you.

pjsg, 2021-03-10:
> And value incorrect with mod:
>
> print(1615391446930 % 36);
> 1538082035638
> print(1538082035638%36);
> 1460772624346
> print(1460772624346%36);
> 1383463213054

I suspect that you wan to use a 64bit build. By default, the firmware only
supports 32 bit integers and then 32 bit floats. This means that there is
no exact representation for integers over 2^32. If you build for 64 bits,
then you get integers up to 64 bits. The downside is that numbers take more
storage....

pjsg, 2021-03-10:
> So I should follow
> `Double build (Lua 5.3 only)
> By default a build will be generated supporting floating point variables
> (floats) and integers. To increase the precision of the floating point
> variables, a double build can be created. This is also the default in the
> Lua 5.1 builds. The downside is that more memory is consumed when storing
> variables. You can change this either by uncommenting LUA_NUMBER_64BITS in
> app/include/user_config.h:
> //#define LUA_NUMBER_64BITS`
> to build a 64bit firmware right? Thank you.

Yes. That is exactly right.

vidalouiswang, 2021-03-10:
> Yes. That is exactly right.
> […](#)

Sir, I tried to build a 64bit firmware, but there is an error:
./lua/luaconf.h:185:2: error: #error Lua 5.1 does not support 64 bit inetegers.
 #error Lua 5.1 does not support 64 bit inetegers.
  ^
make[2]: *** [../../Makefile:424: .output/eagle/debug/obj/user_main.o] Error 1
make[2]: Leaving directory '/root/nodemcu-firmware/app/user'
make[1]: *** [../Makefile:380: .subdirs] Error 2
make[1]: Leaving directory '/root/nodemcu-firmware/app'
make: *** [Makefile:380: .subdirs] Error 2
make: Leaving directory '/root/nodemcu-firmware'

I didn't find method how to change to Lua 5.3, please tell me how to do, thanks.

vidalouiswang, 2021-03-10:
> Yes. That is exactly right.
> […](#)

To rename folder "lua53" to "lua"?

vidalouiswang, 2021-03-10:
> Yes. That is exactly right.
> […](#)

I renamed the folder and this error show up:
CC app/libc/stdlib.c
CC app/libc/snprintf.c
CC app/libc/math.c
AR app/libc/
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/liblibc.a
make[2]: Leaving directory '/root/nodemcu-firmware/app/libc'
make[2]: Entering directory '/root/nodemcu-firmware/app/lua'
Makefile:29: *** Expected environment variable "LUA" to be "53", not "".  Stop.
make[2]: Leaving directory '/root/nodemcu-firmware/app/lua'
make[1]: *** [../Makefile:380: .subdirs] Error 2
make[1]: Leaving directory '/root/nodemcu-firmware/app'
make: *** [Makefile:380: .subdirs] Error 2
make: Leaving directory '/root/nodemcu-firmware'

Please teach me, thank you.

vidalouiswang, 2021-03-10:
> Yes. That is exactly right.
> […](#)

I restored those two folders and use "make LUA=53" to build, some errors in the coap module:
DEPEND: CC app/coap/coap_client.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from coap_client.c:4:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
DEPEND: CC app/coap/coap_io.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from pdu.h:8,
                 from coap_io.h:10,
                 from coap_io.c:2:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
DEPEND: CC app/coap/endpoints.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from endpoints.c:4:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
DEPEND: CC app/coap/coap.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from coap.c:4:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
DEPEND: CC app/coap/str.c
DEPEND: CC app/coap/uri.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from uri.c:9:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
DEPEND: CC app/coap/hash.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from hash.h:8,
                 from hash.c:1:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
DEPEND: CC app/coap/coap_timer.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from hash.h:8,
                 from node.h:8,
                 from coap_timer.c:1:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
DEPEND: CC app/coap/node.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
[... 111 lines cut ...]
CC app/coap/pdu.c
In file included from ../lua53/lua.h:16:0,
                 from ../lua53/lauxlib.h:15,
                 from coap.h:10,
                 from pdu.h:8,
                 from pdu.c:2:
../lua53/luaconf.h:388:2: error: #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS'   or '-DLUA_C89_NUMBERS' (see file 'luaconf.h' for details)"
 #error "Compiler does not support 'long long'. Use option '-DLUA_32BITS' \
  ^
In file included from ../lua53/lauxlib.h:15:0,
                 from coap.h:10,
                 from pdu.h:8,
                 from pdu.c:2:
../lua53/lua.h:94:1: error: unknown type name 'LUA_INTEGER'
 typedef LUA_INTEGER lua_Integer;
 ^
../lua53/lua.h:97:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'lua_Unsigned'
 typedef LUA_UNSIGNED lua_Unsigned;
                      ^
make[2]: *** [../../Makefile:424: .output/eagle/debug/obj/pdu.o] Error 1
make[2]: Leaving directory '/root/nodemcu-firmware/app/coap'
make[1]: *** [../Makefile:380: .subdirs] Error 2
make[1]: Leaving directory '/root/nodemcu-firmware/app'
make: *** [Makefile:380: .subdirs] Error 2
make: Leaving directory '/root/nodemcu-firmware'

vidalouiswang, 2021-03-10:
> Yes. That is exactly right.
> […](#)

After removed "coap" module, the build process is succeeded. Thank you very much.

closed 2021-03-10 by vidalouiswang

pjsg, 2021-03-10:
You really want to be using the lua53 version -- that is also controlled by
a line in the user_config.h