nodemcu/nodemcu-firmware #3409: https server support on esp32

kit-klein, 2021-03-16:
### Missing feature
I'm looking to add https server side support to a project utilizing the esp32 build of nodemcu.  

A moderate amount of google research hasn't identified in an existing implementation or http overlay that works with nodemcu (if anyone is aware of one - please let me know). At this point it seems like the only option is to build one.

I'm considering a few paths here and would appreciate any thoughts/insight before starting.

1. Create a C based http module that leverages esp-idfs [http and https implementation ](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/api-reference/protocols/esp_https_server.html). I like this approach as creating c lua modules the wrap esp-idf functionality is well established and we can piggyback off fixes/testing/stability of the esp implementation.
2. Adapt something like [LuaSec](https://github.com/brunoos/luasec) to overlay the existing Lua HTTP module. This seems like a less straightforward path to me.

### Justification
- Security is important
- ESP32 has the power to handle a secure server
- Other frameworks support https servers

### Workarounds
None

nwf, 2021-03-18:
Does the ESP32 really have the power to be a TLS server?  The demands are quite high.

marcelstoer, 2021-03-19:
> Does the ESP32 really have the power

As the IDF provides the foundation for that and since there's an [established lib for the Arduino](https://github.com/fhessel/esp32_https_server) the answer is likely a "yes", no? Haven't tried either of them, though.

heythisisnate, 2021-04-05:
my company (Konnected) is willing to sponsor a bounty or fee to any contributor who wants to accomplish this
@tomsci @thornley-david @jpeletier any interest?

closed 2022-04-30 by stale[bot]