nodemcu/nodemcu-firmware #3410: UART undefined constant  uart.parity_NONE

ProtoSD, 2021-03-22:
```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
    branch: release
    commit: 136e09739b835d6dcdf04034141d70ab755468c6
    release: 3.0.0-release_20210201
    release DTS: 202102010145
    SSL: true
    build type: float
    LFS: 0x0 bytes total capacity
    modules: adc,bit,file,gpio,net,node,tmr,uart,wifi,tls
 build 2021-03-17 08:20 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```

ESP-01 (8266)

```
>
uart=require("uart")
>
=uart
table: 0x4028d530
>
=print(uart.parity_NONE)
nil
>
=uart.STOPBITS_1
1
> 
```
```
uart=require("uart")
uart.setup(0, 19200, 8, uart.parity_NONE, uart.STOPBITS_1, 0)

Error: stdin:1: bad argument #4 to 'setup' (number expected, got nil)
```

HHHartmann, 2021-03-22:
Try `uart.PARITY_NONE`, Lua is case sensitive.

closed 2021-03-22 by HHHartmann

ProtoSD, 2021-03-22:
Thanks, thought I'd copied and pasted from documentation, but I checked and that's correct. Sorry for the false alarm!
I just tested and it works, thank you.

HHHartmann, 2021-03-22:
Happy to help.
From which Documentation did you copy it. That should be fixed if the program is not working.

ProtoSD, 2021-03-22:
I'm not certain, too many tabs opened! Yes, if I find it I will report the problem to the poster.

I'm experiencing another unusual problem with the uart, but I'm still troubleshooting. Out of memory when I change baud from 115200 to 19200, but works perfect at 115200. About to go to sleep so I'll look into that tomorrow.

Thanks again!